Pulse-Eight/libcec #165: Sometimes AVR powers off instead of being selected

elouet, 2015-12-02:
I understand that libcec tries hard to keep track of power status of each device around in the bus, but this seems to fail every now and then. 

I have had occurrences of this on two distinct setups : 
- a Pioneer AVR hooked onto a Sony TV + a RPi as test device
- an Onkyo AVR hooked onto a dumb monitor, and a RPi2 running Kodi
although I am not 100% shure it is the same issue. 

I managed to capture a log of the problem happening during a test session (manually, using cec-client). I would set the AVR to an alternate source, then type in "as" in the client to switch back 
to my RPi. In this case, (and this is after switching the AVR on and off a couple of times) the libcec thinks it needs to power on the device, and sends a power key sequence, which in effect turns the receiver off. I am linking this behaviour with CECCommandHandler.cpp function PowerOn, but again since I did not debug this, I am not entirely sure. 

I wonder is a

opdenkamp, 2016-01-19:
the problem with that key is that it's not supported by a lot of devices, even though it's included in the specification. can you give me a debug log of when it doesn't work.

elouet, 2016-01-21:
Yesterday I tried to make a log in kodi using cec-client -m but I think this won't work since invoking cec-client seems to break CEC in kodi. I might have to take a closer look at the kodi internal logs then.

opdenkamp, 2016-01-21:
You can only use one of the two: either Kodi or cec-client. You can get all the logging from Kodi, by going to system -> debugging -> enable debug logging and enabling the component below that.

elouet, 2016-01-29:
I tried this, but to no avail, the kodi log did not show anything. Now that I have the PulseEight USB adapter I will be able to spy on the link and confirm whether it is the same bug as #186.

opdenkamp, 2016-01-29:
I'm going to need Kodi's log to see what's going on there exactly. If Kodi's log didn't show anything, then you need to enable the CEC debugging component where you enable debugging. Then restart Kodi and try to reproduce it. Thanks.

Theetjuh, 2016-01-29:
Hi, since I got the same issue, maybe my Kodi logs help as well?
Please see https://github.com/Pulse-Eight/libcec/issues/186

elouet, 2016-01-30:
I just made a log with the CEC USB adapter. Things look a bit different : in my case the TX-NR646 does respond in time and accurately to the <give device power status> requests. I can't understand why libcec should send an order to switch the amp on. Again, deterministic key codes would help, but if they're not widely supported, it's a shame.
`
opening a connection to the CEC adapter...
DEBUG:   [              11]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              24]     connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [              24]     communication thread started
DEBUG:   [              34]     turning controlled mode on
NOTICE:  [              54]     connection opened
waiting for input
DEBUG:   [              54]     processor thread started
TRAFFIC: [            8236]     >> 5f:80:11:00:13:00
DEBUG:   [            8236]     Audio (5): power status changed from 'unknown' to 'on'
DEBUG:   [            8236]     device Audio (5) status changed to present after command routing change
DEBUG:   [            8236]     >> Audio (5) -> Broadcast (F): routing change (80)
TRAFFIC: [            8301]     >> 15:8f
TRAFFIC: [            8675]     >> 51:90:00
TRAFFIC: [            9380]     >> 15:8f
TRAFFIC: [            9756]     >> 51:90:00
TRAFFIC: [           10441]     >> 15:8f
TRAFFIC: [           10816]     >> 51:90:00
TRAFFIC: [           11504]     >> 15:8f
TRAFFIC: [           11879]     >> 51:90:00
TRAFFIC: [           12590]     >> 15:44:40
TRAFFIC: [           12657]     >> 15:45
TRAFFIC: [           12698]     >> 10
DEBUG:   [           12698]     << POLL: Recorder 1 (1) -> TV (0)
TRAFFIC: [           12733]     >> 10
DEBUG:   [           12733]     << POLL: Recorder 1 (1) -> TV (0)
TRAFFIC: [           12780]     >> 10
DEBUG:   [           12780]     << POLL: Recorder 1 (1) -> TV (0)
TRAFFIC: [           12816]     >> 10
DEBUG:   [           12816]     << POLL: Recorder 1 (1) -> TV (0)
TRAFFIC: [           12943]     >> 1f:82:13:00
DEBUG:   [           12943]     Recorder 1 (1): physical address changed from ffff to 1300
DEBUG:   [           12943]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [           12943]     making Recorder 1 (1) the active source
DEBUG:   [           12943]     >> Recorder 1 (1) -> Broadcast (F): active source (82)
DEBUG:   [           12943]     TV (0): power status changed from 'unknown' to 'in transition from standby to on'
DEBUG:   [           12943]     device Recorder 1 (1) status changed to present after command active source
TRAFFIC: [           13033]     >> 5f:72:00
DEBUG:   [           13033]     >> Audio (5): system audio mode status changed from on to off
DEBUG:   [           13033]     >> Audio (5) -> Broadcast (F): set system audio mode (72)
`

opdenkamp, 2016-01-30:
I need Kodi's log please, in which you've reproduced the issue.

zagoruev, 2016-02-06:
Hi, i've got the same issue. Sometimes when i switch from TV to RPi (connected to HDMI-2), my home theater (which is connected to HDMI-1-ARC) goes standby.

Here is Kodi log.

Switching to TV, sound play thru ARC:
```
12:44:34 52601.222656 T:1866462272   DEBUG: CecLogMessage - >> 0f:82:00:00
12:44:34 52601.222656 T:1866462272   DEBUG: CecLogMessage - making TV (0) the active source
12:44:34 52601.222656 T:1866462272   DEBUG: CecLogMessage - marking Recorder 1 (1) as inactive source
12:44:34 52601.222656 T:1866462272   DEBUG: CecLogMessage - >> source deactivated: Recorder 1 (1)
12:44:34 52601.257812 T:1866462272   DEBUG: CecLogMessage - >> TV (0) -> Broadcast (F): active source (82)
```

Switching back to RPi, expecting AVR should stand back in «ARC-mode» but it goes to standby:
```
12:44:59 52626.683594 T:1866462272   DEBUG: CecLogMessage - >> 0f:80:00:00:20:00
12:44:59 52626.683594 T:1866462272   DEBUG: CecLogMessage - making Recorder 1 (1) the active source
12:44:59 52626.683594 T:1866462272   DEBUG: CecLogMessage - marking TV (0) as inactive source
12:44:59 52626.683594 T:1866462272   DEBUG: CecLogMessage - >> source activated: Recorder 1 (1)
12:44:59 52626.683594 T:1866462272   DEBUG: CecLogMessage - << requesting power status of 'Audio' (5)
12:44:59 52626.683594 T:1866462272   DEBUG: CecLogMessage - << 15:8f
12:44:59 52626.683594 T:1866462272   DEBUG: CecLogMessage - >> TV (0) -> Broadcast (F): routing change (80)
12:44:59 52626.773438 T:1866462272   DEBUG: CecLogMessage - expected response received (90: report power status)
12:44:59 52626.773438 T:1866462272   DEBUG: CecLogMessage - sending active source message for 'Recorder 1'
12:44:59 52626.773438 T:1866462272   DEBUG: CecLogMessage - << requesting vendor ID of 'TV' (0)
12:44:59 52626.773438 T:1866462272   DEBUG: CecLogMessage - 'give device vendor id' is marked as unsupported feature for device 'TV'
12:44:59 52626.773438 T:1866462272   DEBUG: CecLogMessage - << powering on 'TV' (0)
12:44:59 52626.773438 T:1866462272   DEBUG: CecLogMessage - << 10:04
12:44:59 52626.832031 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=8f initiator=1 destination=5 response=0
12:44:59 52626.984375 T:1866462272   DEBUG: CecLogMessage - << Recorder 1 (1) -> broadcast (F): active source (2000)
12:44:59 52626.984375 T:1866462272   DEBUG: CecLogMessage - << 1f:82:20:00
12:44:59 52627.101562 T:1866462272   DEBUG: CecLogMessage - << Recorder 1 (1) -> TV (0): menu state 'activated'
12:44:59 52627.105469 T:1866462272   DEBUG: CecLogMessage - 'menu status' is marked as unsupported feature for device 'TV'
12:44:59 52627.105469 T:1866462272   DEBUG: CecLogMessage - >> 51:90:00
12:44:59 52627.105469 T:1866462272   DEBUG: CecLogMessage - >> Audio (5) -> Recorder 1 (1): report power status (90)
12:45:00 52627.242188 T:1866462272   DEBUG: CecLogMessage - >> 51:90:00
12:45:00 52627.246094 T:1866462272   DEBUG: CecLogMessage - >> Audio (5) -> Recorder 1 (1): report power status (90)
12:45:00 52627.832031 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=8f initiator=1 destination=5 response=0
12:45:00 52627.941406 T:1866462272   DEBUG: CecLogMessage - >> 51:90:00
12:45:00 52627.941406 T:1866462272   DEBUG: CecLogMessage - >> Audio (5) -> Recorder 1 (1): report power status (90)
12:45:01 52628.894531 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=8f initiator=1 destination=5 response=0
12:45:01 52629.015625 T:1866462272   DEBUG: CecLogMessage - >> 51:90:00
12:45:01 52629.019531 T:1866462272   DEBUG: CecLogMessage - >> Audio (5) -> Recorder 1 (1): report power status (90)
12:45:02 52629.953125 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=8f initiator=1 destination=5 response=0
12:45:02 52630.066406 T:1866462272   DEBUG: CecLogMessage - >> 51:90:00
12:45:02 52630.070312 T:1866462272   DEBUG: CecLogMessage - >> Audio (5) -> Recorder 1 (1): report power status (90)
12:45:03 52631.046875 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=44 initiator=1 destination=5 response=0
12:45:03 52631.105469 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=45 initiator=1 destination=5 response=0
12:45:04 52631.257812 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=4 initiator=1 destination=0 response=0
12:45:04 52631.527344 T:1866462272 WARNING: CecLogMessage - unhandled response received: opcode=82 initiator=1 destination=f response=0
12:45:07 52634.367188 T:1866462272   DEBUG: CecLogMessage - >> 5f:72:00
12:45:07 52634.367188 T:1866462272   DEBUG: CecLogMessage - >> Audio (5): system audio mode status changed from on to off
12:45:07 52634.367188 T:1866462272   DEBUG: CecLogMessage - >> Audio (5) -> Broadcast (F): set system audio mode (72)
```

dsebastien, 2016-02-14:
I'm also having this issue.

nrushforth, 2016-03-07:
I am also having same issue. 

RPI connected to TV via HDMI. 

TV connected to Sony Soundbase via HDMI ARC. 

When switching TV source to RPI the Soundbase goes into standby.  

This is happening in OPENELEC 6.0.3. I was previously using 5.0.6 where this was not happening. 

Any ideas?

popcornmix, 2016-03-20:
Latest Milhouse nightly build on Pi includes https://github.com/Pulse-Eight/libcec/pull/208.
Would be interesting to hear reports if this improves behaviour.

delicious2k, 2016-03-31:
I have the same issue, and it drives me crazy :-)

I am using Kodi 16.0 on a Windows 7 Desktop PC with an Onkyo TX-NR616 AVR and a Philips PFL TV.

The interesting fact is, that it was working well with XBMC 13.0. After an update to Kodi 15.2 my AVR   turns off from time to time I switch the input to Kodi. As I read in XBMC they still used libcec2.2.0 and with the update to Kodi 15.2 they also included libcec3.0.0. 
=> So it was working in libcec2.2.0

Hope you can solve this issue!

Theetjuh, 2016-03-31:
Looks like the patch popcornmix is talking about fixed the issue!

dsebastien, 2016-03-31:
I've installed the latest OpenElec RC and the problem seems to be fixed for me. @popcornmix you are my hero! :)

delicious2k, 2016-03-31:
Is there an easy way to patch my Kodi 16.0... I dont want to "fix" it with an alpha build of Kodi 17

dsebastien, 2016-07-26:
I guess that a new version of OpenElec was released with a different version of libcec because this problem has resurfaced for me..

popcornmix, 2016-07-26:
OpenELEC official builds have never included #208. There hasn't been an OpenELEC build for a long time.
I'd suggest using LibreELEC which I belive does include #208.

Theetjuh, 2016-07-26:
I haven't had an issue since Sam included the patch in OSMC, so that is also an option.

opdenkamp, 2016-10-25:
Forgot to close this one. It'll be fixed in the next release

closed 2016-10-25 by opdenkamp