Pulse-Eight/libcec #166: Raspberry Pi support: no

G-IV, 2015-12-11:
Hi all,

I'm using a Raspberry Pi 2 Model B Rev 1.1 - For those who don't know how, I found it using the code below:

 `dmesg | grep 'Machine model'`

I have been trying to get the libcec working for the last two days, and I haven't had much luck.  I've tried following the instructions located in the [read me](https://github.com/Pulse-Eight/libcec/blob/master/README.md) file, but those didn't work.  I wound up being unable to connect to the TV. 

I have tried a couple of the ways in [Complete installation instructions on Raspbian #84](https://github.com/Pulse-Eight/libcec/issues/new), and that had the same issue I had in the first attempt.

I've tried to dig around to figure this out, but I'm not really sure what is happening.  I have a feeling that it is related to the line that says `Raspberry Pi support: no` that comes up after the command `cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..`.

Here is what I am doing:
```
pi@raspberrypi ~ $ sudo apt-get update
Get:1 http://archive.raspberrypi.org jessie InRelease [13.3 kB]
Get:2 http://mirrordirector.raspbian.org jessie InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org jessie/main Sources [31.5 kB]
Get:4 http://mirrordirector.raspbian.org jessie/main armhf Packages [8,962 kB]
Get:5 http://archive.raspberrypi.org jessie/ui Sources [5,197 B]
Get:6 http://archive.raspberrypi.org jessie/main armhf Packages [101 kB]
Get:7 http://archive.raspberrypi.org jessie/ui armhf Packages [7,639 B]
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Get:8 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [37.5 kB]
Get:9 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [70.2 kB]
Get:10 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9,244 kB in 36s (254 kB/s)
Reading package lists... Done
pi@raspberrypi ~ $ apt-get install cmake liblockdev1-dev libudev-dev libxrandr-dev python-dev swig
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?
pi@raspberrypi ~ $ sudo apt-get install cmake liblockdev1-dev libudev-dev libxrandr-dev python-dev swig
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  cmake-data liblockdev1 libpthread-stubs0-dev libpython-dev libpython2.7-dev libx11-dev libx11-doc libxau-dev libxcb1-dev libxdmcp-dev libxext-dev libxrender-dev
  python2.7-dev swig2.0 x11proto-core-dev x11proto-input-dev x11proto-kb-dev x11proto-randr-dev x11proto-render-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
Suggested packages:
  codeblocks eclipse ninja-build libxcb-doc libxext-doc swig-doc swig-examples swig2.0-examples swig2.0-doc
The following NEW packages will be installed:
  cmake cmake-data liblockdev1 liblockdev1-dev libpthread-stubs0-dev libpython-dev libpython2.7-dev libudev-dev libx11-dev libx11-doc libxau-dev libxcb1-dev libxdmcp-dev
  libxext-dev libxrandr-dev libxrender-dev python-dev python2.7-dev swig swig2.0 x11proto-core-dev x11proto-input-dev x11proto-kb-dev x11proto-randr-dev x11proto-render-dev
  x11proto-xext-dev xorg-sgml-doctools xtrans-dev
0 upgraded, 28 newly installed, 0 to remove and 70 not upgraded.
Need to get 27.1 MB of archives.
After this operation, 62.9 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-dev armhf 2.7.9-1 [1,188 B]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main cmake-data all 3.0.2-1 [929 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main cmake armhf 3.0.2-1 [1,870 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpython2.7-dev armhf 2.7.9-2 [17.9 MB]
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main liblockdev1 armhf 1.0.3-1.6+b1 [12.7 kB]
Get:6 http://mirrordirector.raspbian.org/raspbian/ jessie/main liblockdev1-dev armhf 1.0.3-1.6+b1 [7,804 B]
Get:7 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpthread-stubs0-dev armhf 0.3-4 [4,042 B]
Get:8 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpython-dev armhf 2.7.9-1 [19.6 kB]
Get:9 http://mirrordirector.raspbian.org/raspbian/ jessie/main libudev-dev armhf 215-17+deb8u2 [23.2 kB]
Get:10 http://mirrordirector.raspbian.org/raspbian/ jessie/main xorg-sgml-doctools all 1:1.11-1 [21.9 kB]
Get:11 http://mirrordirector.raspbian.org/raspbian/ jessie/main x11proto-core-dev all 7.0.26-1 [729 kB]
[... 258 lines cut ...]
-- Looking for include file sys/file.h - found
-- Looking for flock
-- Looking for flock - found
-- checking for module 'udev'
--   found udev, version 215
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xrandr.h - found
-- Looking for XRRGetScreenResources in Xrandr
-- Looking for XRRGetScreenResources in Xrandr - found
-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.12")
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   no
-- TDA995x support:                        no
-- Exynos support:                         no
-- DRM support:                            yes
-- Python support:                         version 3.4.2 (3.4)
-- lib info: git revision: libcec-2.1.4+223-2f62595, compiled on Fri 11 Dec 19:04:21 UTC 2015 by pi@raspberrypi on Linux 4.1.7-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/libcec/build
```
I realize the last `cmake ..` command I sent didn't manually set the directories - I did this because having the paths declared didn't make a difference on my last attempt.  I've been reinstalling from the SD after each failed attempt, just to be sure that I didn't mess things up on each attempt.  I didn't want to go through the correct installation process after a bad attempt on the off chance that I could have screwed something up.

philiprileyitu, 2015-12-15:
I have a similar situation. Did you fix your problem? I know you can just do
`sudo apt-get install libcec3 cec-utils`
this works fine but I need to use the python bindings for Python 2.7. Installing using the above method compiles for Python 3.4.

G-IV, 2015-12-15:
I haven't got it fixed yet, but I am going to try your suggestion!

Can I ask why you need Python 2.7 instead of 3.4?  Is there something stopping you from updating Python?

philiprileyitu, 2015-12-15:
I'm using a framework which was built using Python 2.7 and it doesn't work using Python 3.4. :-( Yes try using apt-get. I made sure I had all the dependencies installed first, but maybe that's unnecessary! :-/ I would be grateful if anyone else had an idea of how to solve the Found Devices: None issue on libcec 3. I have managed to install libcec 2 on my Pi and that works, so it's not a TV/Cable issue. I'm always installing on a clean installation of Jessie and I'm making sure to follow the instructions precisely.

G-IV, 2015-12-15:
Hi Phillip,

I appreciate your suggestion on installing the cec-utils, but that didn't resolve the issue.

opdenkamp, 2015-12-15:
The instructions in https://github.com/Pulse-Eight/libcec/blob/master/README.md have been updated recently and I've documented every step there on a clean Raspbian installation, which worked ~2 weeks ago.

If you need a specific Python version, then you must tell Cmake to use that one instead of the default one: https://cmake.org/cmake/help/v3.0/module/FindPythonLibs.html

closed 2015-12-15 by opdenkamp

G-IV, 2015-12-15:
Hi opdenkamp,

I literally just tried this again this morning, and it still isn't working.

opdenkamp, 2015-12-15:
I see that there's a "Raspbian Jessie Lite" on the download page https://www.raspberrypi.org/downloads/raspbian/
Are you perhaps using that one? Cause I can imagine that it won't come with the development headers that you need to build libCEC from source. You need the full version instead in that case.

G-IV, 2015-12-15:
I took a look at what I had, and everything says jessie (no mention of jessie lite).

However, I am just using whatever comes with NOOB - and from what I can tell, NOOB has jessie, not  jessie lite.

I'm trying to do this on another Raspi, hopefully it works this time.

opdenkamp, 2015-12-15:
The manual was written for the official builds, found on https://www.raspberrypi.org/downloads/raspbian/

philiprileyitu, 2015-12-15:
Hi opdenkamp
Thanks for replying!
I definitely used a clean build of Jessie (not Jessie Lite) as per that link you posted. I've actually tried it a few times to make sure that I didn't make a mistake and I ended up with Raspberry Pi Support: No and Found Devices: None. I'm sure I could edit the CMake files to ensure I get the version of Python I need, but the problem is getting Libcec to work. On the other hand, apt-get install libcec3 works, but I need Python 2.7 support!

@G-IV Did you only try to install cec-utils using apt-get? I installed both libcec3 and cec-utils in the same script and that worked. See below: 
`sudo apt-get install libcec3 cec-utils`
But as I mentioned, I had all the dependencies installed beforehand. I'm not sure if apt-get will pull those down for you.

kingosticks, 2015-12-15:
I've never used cmake before but having just tried this myself I also found it didn't work. I noticed that it wasn't reporting "Looking for bcm_host_init in bcm_host" like it does for `Xrandr` and the other libs making use of `check_library_exists`. So I made the check more like `Xrandr`'s (below) and then it worked. I guess it's something to do with trying to get cmake to set a variable that's already defined?

```
  # raspberry pi
  find_library(RPI_BCM_HOST bcm_host ${RPI_LIB_DIR})
  check_library_exists(bcm_host bcm_host_init ${RPI_LIB_DIR} HAVE_RPI_LIB)
  if (HAVE_RPI_LIB)
    set(HAVE_RPI_API 1)
```

G-IV, 2015-12-15:
That didn't do it either.  I feel like I have been slamming my head against a wall since I started trying to get this work last week.

kingosticks, 2015-12-15:
Did you make sure you clean our your build directory before running cmake again? Also, you must specify the paths to the rpi libs and includes just like the readme says to do.

G-IV, 2015-12-15:
Hi kingosticks,

I'm glad that I'm not the only one this isn't working for, hopefully that means that it isn't me being thick. :)

G-IV, 2015-12-15:
I'll go back and start from a clean build, let's see if that works.

philiprileyitu, 2015-12-15:
I got it working, well mostly! I tried using @kingosticks set(HAVE_RPI_API 1) but that didn't work first time around on a clean installation of Jessie. I still got Raspberry Pi Support: No. I tried again, but this time I commented out 
`if (HAVE_RPI_LIB)
ENDIF`
This worked and I was able to do cec-client -l successfully. yey for that, even if a bit hackish! :-)
Thanks @kingosticks 
I also set the cmake variables for changing the python library and after running cmake, it appeared that it had worked, as it showed:
`Python Support: Version 2.7.9 (2.7)`
However when I ran python and imported cec (import cec), I got a segmentation fault.
Has this been tested using Python 2.7?

philiprileyitu, 2015-12-16:
I got the python bindings working this morning so I'm good to go now. Here is the full installation recipe that worked for me.
**1)** Clean installation of the full desktop Jessie image. See here: [https://www.raspberrypi.org/downloads/raspbian/](https://www.raspberrypi.org/downloads/raspbian/)
**2)** Update Jessie
```
sudo apt-get update
sudo apt-get dist-upgrade
sudo rpi-update
sudo reboot
```
**3)** Follow normal procedure in the Libcec README
```
apt-get install cmake liblockdev1-dev libudev-dev libxrandr-dev python-dev swig
cd
git clone https://github.com/Pulse-Eight/platform.git
mkdir platform/build
cd platform/build
cmake ..
make
sudo make install
cd
git clone https://github.com/Pulse-Eight/libcec.git
mkdir libcec/build
cd libcec/build
```
**4)** Edit CheckPlatformSupport.cmake file. If you followed the steps above, this will be located here:
`~/libcec/src/libcec/cmake/`
 From around line 97, edit as follows:
 Comment out `if (HAVE_RPI_API)`
 Add a new line and enter `set(HAVE_RPI_API 1)`
 Comment out `endif()`
 You should end up with something like this:
```
  # raspberry pi 
  find_library(RPI_BCM_HOST bcm_host "${RPI_LIB_DIR}")
  check_library_exists(bcm_host bcm_host_init "${RPI_LIB_DIR}" HAVE_RPI_API)
  # if (HAVE_RPI_API) 
    set(HAVE_RPI_API 1)
    find_library(RPI_VCOS vcos "${RPI_LIB_DIR}")
    find_library(RPI_VCHIQ_ARM vchiq_arm "${RPI_LIB_DIR}")
    include_directories(${RPI_INCLUDE_DIR} ${RPI_INCLUDE_DIR}/interface/vcos/pthreads ${RPI_INCLUDE_DIR}/interface/vmcs_host/linux)

    set(LIB_INFO "${LIB_INFO}, RPi")
    set(CEC_SOURCES_ADAPTER_RPI adapter/RPi/RPiCECAdapterDetection.cpp
                                adapter/RPi/RPiCECAdapterCommunication.cpp
                                adapter/RPi/RPiCECAdapterMessageQueue.cpp)
    source_group("Source Files\\adapter\\RPi" FILES ${CEC_SOURCES_ADAPTER_RPI})
    list(APPEND CEC_SOURCES ${CEC_SOURCES_ADAPTER_RPI})
  # endif()
```
**5)** Follow the rest of the README to finish the build. **Note: if you need bindings for Python 2.7 see section 6 below, otherwise the bindings will be for Python 3.4.**
```
cmake -D RPI_INCLUDE_DIR=/opt/vc/include -D RPI_LIB_DIR=/opt/vc/lib ..
make -j4
sudo make install
sudo ldconfig
```
**6)** For Python 2.7 bindings, you need to add some additional variables to CMake so it can find the installation. There is probably a more correct way, but this worked for me. 
```
cmake -D RPI_INCLUDE_DIR=/opt/vc/include \ 
-D RPI_LIB_DIR=/opt/vc/lib \
-D PYTHON_LIBRARY=/usr/lib/arm-linux-gnueabihf/libpython2.7.so \
-D PYTHON_INCLUDE_DIR=/usr/include/python2.7 ..
make -j4
sudo make install
sudo ldconfig
```

Something is up with that `HAVE_RPI_API` variable, so maybe it needs to be looked at again. Thanks for all the help!

G-IV, 2015-12-16:
This still didn't work, and I followed your directions to the T.

The printout still showed that the Raspberry Pi didn't have support, and the cec-client command failed the auto-detect.

Out of curiosity, does this work without the usb-hdmi connector, or is that my issue?  I have been assuming that this ran directly on the HDMI port of the Raspi.

philiprileyitu, 2015-12-16:
I'm just using an HDMI cable directly to my TV, so it should work for you too. Dumb question, but did you remember to save the changes to the CheckPlatformSupport.cmake file? Just check it to be sure. ;-) I know you said you followed it to a T, but it was a clean installation of Jessie right?

kingosticks, 2015-12-16:
Are you sure you used exactly my code? I changed 3 lines in total. Did it print a message saying it was looking for the bcm_host library? I am not using a usb-hdmi cable.

philiprileyitu, 2015-12-16:
I just had another look at your changes in the commit you posted and I can see that I missed the change you made to this line:
`check_library_exists(bcm_host bcm_host_init "${RPI_LIB_DIR}" HAVE_RPI_API)`
which is now
`check_library_exists(bcm_host bcm_host_init "${RPI_LIB_DIR}" HAVE_RPI_LIB)`
Have you tried that @G-IV? Maybe then you won't have to comment out the if/endif() lines as per my instructions!

G-IV, 2015-12-16:
Step 1: 
'''
sudo apt-get update
sudo apt-get dist-upgrade
sudo rpi-update
sudo reboot
'''

G-IV, 2015-12-16:
Step 2:
```
sudo apt-get install cmake liblockdev1-dev libudev-dev libxrandr-dev python-dev swigReading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  cmake-data liblockdev1 libpthread-stubs0-dev libpython-dev libpython2.7-dev libx11-dev libx11-doc libxau-dev
  libxcb1-dev libxdmcp-dev libxext-dev libxrender-dev python2.7-dev swig2.0 x11proto-core-dev x11proto-input-dev
  x11proto-kb-dev x11proto-randr-dev x11proto-render-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
Suggested packages:
  codeblocks eclipse ninja-build libxcb-doc libxext-doc swig-doc swig-examples swig2.0-examples swig2.0-doc
The following NEW packages will be installed:
  cmake cmake-data liblockdev1 liblockdev1-dev libpthread-stubs0-dev libpython-dev libpython2.7-dev libudev-dev
  libx11-dev libx11-doc libxau-dev libxcb1-dev libxdmcp-dev libxext-dev libxrandr-dev libxrender-dev python-dev
  python2.7-dev swig swig2.0 x11proto-core-dev x11proto-input-dev x11proto-kb-dev x11proto-randr-dev
  x11proto-render-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
0 upgraded, 28 newly installed, 0 to remove and 0 not upgraded.
Need to get 27.1 MB of archives.
After this operation, 62.9 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main cmake-data all 3.0.2-1 [929 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-dev armhf 2.7.9-1 [1,188 B]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main cmake armhf 3.0.2-1 [1,870 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpython2.7-dev armhf 2.7.9-2 [17.9 MB]                 
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main liblockdev1 armhf 1.0.3-1.6+b1 [12.7 kB]                 
Get:6 http://mirrordirector.raspbian.org/raspbian/ jessie/main liblockdev1-dev armhf 1.0.3-1.6+b1 [7,804 B]             
Get:7 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpthread-stubs0-dev armhf 0.3-4 [4,042 B]              
Get:8 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpython-dev armhf 2.7.9-1 [19.6 kB]                    
Get:9 http://mirrordirector.raspbian.org/raspbian/ jessie/main libudev-dev armhf 215-17+deb8u2 [23.2 kB]                
Get:10 http://mirrordirector.raspbian.org/raspbian/ jessie/main xorg-sgml-doctools all 1:1.11-1 [21.9 kB]               
Get:11 http://mirrordirector.raspbian.org/raspbian/ jessie/main x11proto-core-dev all 7.0.26-1 [729 kB]                 
Get:12 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxau-dev armhf 1:1.0.8-1 [23.0 kB]                    
Get:13 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxdmcp-dev armhf 1:1.1.1-1 [40.8 kB]                  
Get:14 http://mirrordirector.raspbian.org/raspbian/ jessie/main x11proto-input-dev all 2.3.1-1 [157 kB]                 
Get:15 http://mirrordirector.raspbian.org/raspbian/ jessie/main x11proto-kb-dev all 1.0.6-2 [269 kB]                    
Get:16 http://mirrordirector.raspbian.org/raspbian/ jessie/main xtrans-dev all 1.3.4-1 [99.3 kB]                        
Get:17 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxcb1-dev armhf 1.10-3 [78.1 kB]                      
Get:18 http://mirrordirector.raspbian.org/raspbian/ jessie/main libx11-dev armhf 2:1.6.2-3 [726 kB]                     
Get:19 http://mirrordirector.raspbian.org/raspbian/ jessie/main libx11-doc all 2:1.6.2-3 [2,026 kB]                     
Get:20 http://mirrordirector.raspbian.org/raspbian/ jessie/main x11proto-xext-dev all 7.3.0-1 [212 kB]                  
Get:21 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxext-dev armhf 2:1.3.3-1 [102 kB]                    
Get:22 http://mirrordirector.raspbian.org/raspbian/ jessie/main x11proto-randr-dev all 1.4.0-2 [46.8 kB]                
Get:23 http://mirrordirector.raspbian.org/raspbian/ jessie/main x11proto-render-dev all 2:0.11.1-2 [20.8 kB]            
Get:24 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxrender-dev armhf 1:0.9.8-1 [38.8 kB]                
Get:25 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxrandr-dev armhf 2:1.4.2-1 [39.3 kB]                 
Get:26 http://mirrordirector.raspbian.org/raspbian/ jessie/main python2.7-dev armhf 2.7.9-2 [281 kB]                    
Get:27 http://mirrordirector.raspbian.org/raspbian/ jessie/main swig2.0 armhf 2.0.12-1 [1,071 kB]                       
Get:28 http://mirrordirector.raspbian.org/raspbian/ jessie/main swig armhf 2.0.12-1 [285 kB]                            
Fetched 27.1 MB in 24s (1,094 kB/s)                                                                                     
Selecting previously unselected package cmake-data.
(Reading database ... 116907 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.0.2-1_all.deb ...
Unpacking cmake-data (3.0.2-1) ...
Selecting previously unselected package cmake.
Preparing to unpack .../cmake_3.0.2-1_armhf.deb ...
Unpacking cmake (3.0.2-1) ...
Selecting previously unselected package libpython2.7-dev:armhf.
Preparing to unpack .../libpython2.7-dev_2.7.9-2_armhf.deb ...
Unpacking libpython2.7-dev:armhf (2.7.9-2) ...
Selecting previously unselected package liblockdev1.
Preparing to unpack .../liblockdev1_1.0.3-1.6+b1_armhf.deb ...
Unpacking liblockdev1 (1.0.3-1.6+b1) ...
[... cut ...]
Setting up libx11-doc (2:1.6.2-3) ...
Setting up x11proto-xext-dev (7.3.0-1) ...
Setting up libxext-dev:armhf (2:1.3.3-1) ...
Setting up x11proto-randr-dev (1.4.0-2) ...
Setting up x11proto-render-dev (2:0.11.1-2) ...
Setting up libxrender-dev:armhf (1:0.9.8-1) ...
Setting up libxrandr-dev:armhf (2:1.4.2-1) ...
Setting up python2.7-dev (2.7.9-2) ...
Setting up python-dev (2.7.9-1) ...
Setting up swig2.0 (2.0.12-1) ...
Setting up swig (2.0.12-1) ...
Processing triggers for libc-bin (2.19-18+deb8u1) ...
```

G-IV, 2015-12-16:
Step 3:
```
pi@raspberrypi ~ $ cd
pi@raspberrypi ~ $ git clone https://github.com/Pulse-Eight/platform.git
Cloning into 'platform'...
remote: Counting objects: 449, done.
remote: Total 449 (delta 0), reused 0 (delta 0), pack-reused 449
Receiving objects: 100% (449/449), 156.58 KiB | 0 bytes/s, done.
Resolving deltas: 100% (257/257), done.
Checking connectivity... done.
```

G-IV, 2015-12-16:
Step 4:
```
pi@raspberrypi ~ $ mkdir platform/build
pi@raspberrypi ~ $ cd platform/build
pi@raspberrypi ~/platform/build $ cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for __atomic_fetch_add_4 in atomic
-- Looking for __atomic_fetch_add_4 in atomic - found
-- Performing Test HAS_BUILTIN_SYNC_ADD_AND_FETCH
-- Performing Test HAS_BUILTIN_SYNC_ADD_AND_FETCH - Success
-- Performing Test HAS_BUILTIN_SYNC_SUB_AND_FETCH
-- Performing Test HAS_BUILTIN_SYNC_SUB_AND_FETCH - Success
-- Performing Test HAS_BUILTIN_SYNC_VAL_COMPARE_AND_SWAP
-- Performing Test HAS_BUILTIN_SYNC_VAL_COMPARE_AND_SWAP - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/platform/build
```

G-IV, 2015-12-16:
Step 5: 
```
pi@raspberrypi ~/platform/build $ sudo make
Scanning dependencies of target p8-platform
[100%] Building CXX object CMakeFiles/p8-platform.dir/src/util/StringUtils.cpp.o
Linking CXX static library libp8-platform.a
[100%] Built target p8-platform
pi@raspberrypi ~/platform/build $ sudo make install
[100%] Built target p8-platform
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/libp8-platform.a
-- Installing: /usr/local/include/p8-platform/os.h
-- Installing: /usr/local/include/p8-platform/posix/os-socket.h
-- Installing: /usr/local/include/p8-platform/posix/os-threads.h
-- Installing: /usr/local/include/p8-platform/posix/os-types.h
-- Installing: /usr/local/include/p8-platform/sockets/cdevsocket.h
-- Installing: /usr/local/include/p8-platform/sockets/socket.h
-- Installing: /usr/local/include/p8-platform/sockets/tcp.h
-- Installing: /usr/local/include/p8-platform/threads/atomics.h
-- Installing: /usr/local/include/p8-platform/threads/mutex.h
-- Installing: /usr/local/include/p8-platform/threads/threads.h
-- Installing: /usr/local/include/p8-platform/util/atomic.h
-- Installing: /usr/local/include/p8-platform/util/buffer.h
-- Installing: /usr/local/include/p8-platform/util/StringUtils.h
-- Installing: /usr/local/include/p8-platform/util/StdString.h
-- Installing: /usr/local/include/p8-platform/util/timeutils.h
-- Installing: /usr/local/include/p8-platform/util/util.h
-- Installing: /usr/local/lib/pkgconfig/p8-platform.pc
-- Installing: /usr/local/lib/p8-platform/p8-platform-config.cmake
pi@raspberrypi ~/platform/build $ cd
```

G-IV, 2015-12-16:
Step 6:
```
pi@raspberrypi ~ $ git clone https://github.com/Pulse-Eight/libcec.git
Cloning into 'libcec'...
remote: Counting objects: 13910, done.
remote: Total 13910 (delta 0), reused 0 (delta 0), pack-reused 13910
Receiving objects: 100% (13910/13910), 4.86 MiB | 1.18 MiB/s, done.
Resolving deltas: 100% (10604/10604), done.
Checking connectivity... done.
pi@raspberrypi ~ $ mkdir libcec/build
pi@raspberrypi ~ $ cd libcec/build
pi@raspberrypi ~/libcec/build $ cd
```

G-IV, 2015-12-16:
Step 7:
```
pi@raspberrypi ~ $ ls
Desktop  Documents  Downloads  libcec  Music  oldconffiles  Pictures  platform  Public  python_games  Templates  Videos
pi@raspberrypi ~ $ sudo nano libcec/src/libcec/cmake/
pi@raspberrypi ~ $ sudo nano libcec/src/libcec/cmake
pi@raspberrypi ~ $ cd libcec/src/libcec/cmake
pi@raspberrypi ~/libcec/src/libcec/cmake $ sudo nano CheckPlatformSupport.cmake
```

G-IV, 2015-12-16:
Let me try this again, I was going through everything, and saw that I did this: set(HAVE_RPI_API)... not this set(HAVE_RPI_API 1).

crap.  I'll get back to you, sorry I've been so difficult.  I really appreciate all the help!

kingosticks, 2015-12-16:
I thought as much. Maybe it would be simpler (just for testing here) to replace your step 4 with 
```
wget https://raw.githubusercontent.com/kingosticks/libcec/f3a09a1d6d468a67b95f6b106d7070f94663029f/src/libcec/cmake/CheckPlatformSupport.cmake -O ../src/libcec/cmake/CheckPlatformSupport.cmake
```

G-IV, 2015-12-16:
YOU GUYS ARE AWESOME!!!!

I really couldn't have done it without your help.  Thank you!!

philiprileyitu, 2015-12-16:
:+1: Did you do it with @kingosticks method, i.e., without commenting out the if/endif() lines? If so, I'll update step five.

G-IV, 2015-12-16:
I used @kingosticks method.  Now I am exploring how to use the library.  :)

kingosticks, 2015-12-17:
Good to hear. Hopefully @opdenkamp will merge the fix at some point, I think it is legit.

thundergreen, 2016-08-10:
hi guys.

i also followed your tutorial 1by1 but at the end i get this error:

```
pi@Home-Assistant:~/libcec/build $ cmake -D RPI_INCLUDE_DIR=/opt/vc/include \ -D RPI_LIB_DIR=/opt/vc/lib \ -D PYTHON_LIBRARY=/usr/lib/arm-linux-gnueabihf/libpython2.7.so \ -D PYTHON_INCLUDE_DIR=/usr/include/python2.7 ..
-- git found: libcec-3.1.0+25-62b8520~dirty
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   yes
-- TDA995x support:                        no
-- Exynos support:                         no
-- DRM support:                            yes
-- Python support:                         version 3.4.2 (3.4)
-- lib info: git revision: libcec-3.1.0+25-62b8520~dirty, compiled on Wed Aug 10 20:08:17 UTC 2016 by pi@Home-Assistant on Linux 4.4.13-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/libcec/build
pi@Home-Assistant:~/libcec/build $ make -j4
Scanning dependencies of target cec
[100%] [100%] [100%] [100%] Building CXX object src/libcec/CMakeFiles/cec.dir/CECClient.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/LibCEC.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/LibCECC.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/CECProcessor.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterCommunication.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterMessageQueue.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/AdapterFactory.cpp.o
In file included from /home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:37:0:
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h: In constructor ‘CEC::CRPiCECAdapterCommunication::CRPiCECAdapterCommunication(CEC::IAdapterCommunicationCallback*)’:
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:118:35: warning: ‘CEC::CRPiCECAdapterCommunication::m_bDisableCallbacks’ will be initialized after [-Wreorder]
     bool                          m_bDisableCallbacks;
                                   ^
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:105:10: warning:   ‘bool CEC::CRPiCECAdapterCommunication::m_bInitialised’ [-Wreorder]
     bool m_bInitialised;   /**< true when the connection is initialised, false otherwise */
          ^
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:68:1: warning:   when initialized here [-Wreorder]
 CRPiCECAdapterCommunication::CRPiCECAdapterCommunication(IAdapterCommunicationCallback *callback) :
 ^
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterMessage.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterCommands.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterCommunication.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterMessageQueue.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterDetection.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECAudioSystem.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECBusDevice.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECDeviceMap.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECPlaybackDevice.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECRecordingDevice.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECTuner.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/devices/CECTV.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/ANCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/CECCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/SLCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/VLCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/RLCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/PHCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/RHCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/implementations/AQCommandHandler.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/adl/adl-edid.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/nvidia/nv-edid.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/posix/os-edid.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/posix/serialport.cpp.o
[100%] Building CXX object src/libcec/CMakeFiles/cec.dir/platform/X11/randr-edid.cpp.o
Linking CXX shared library libcec.so
[100%] Built target cec
[100%] Built target cecc-client
[100%] [100%] Built target cec-client
Swig source
[100%] Building CXX object src/libcec/CMakeFiles/_cec.dir/libcecPYTHON_wrap.cxx.o
Linking CXX shared module _cec.so
[100%] Built target _cec
[... cut ...]
/usr/bin/ld: cannot find -lvchiq_arm
/usr/bin/ld: cannot find -lbcm_host
collect2: error: ld returned 1 exit status
src/libcec/CMakeFiles/_cec.dir/build.make:98: recipe for target 'src/libcec/_cec.so' failed
make[2]: *** [src/libcec/_cec.so] Error 1
CMakeFiles/Makefile2:192: recipe for target 'src/libcec/CMakeFiles/_cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/_cec.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
pi@Home-Assistant:~/libcec/build $
``````

opdenkamp, 2017-07-13:
Very late response, but in case someone bumps into this via google: the Raspberry Pi's firmware headers can't be found. This is described in the readme: https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md

uvwild, 2018-03-15:
since I am ending on this page with the same issue building a synergy 1.7.6 version on ubuntu 16.04 using cmake 3.5.2....
**The CMakeLists.txt dont work anymore with Cmake 3.5.2!!!**
I ended up changing the logic in the cmake modules as the quickest (and most dirty solution)

- CheckIncludeFiles.cmake and
- CheckLibraryExists.cmake

to not check the passed VARIABLE for being DEFINED.
I dont understand how this can happen. But skipping this test makes it work!
Some later cmake versions seem to have some efficiency included which breaks this.
```
macro(XXXX )
    if (not defined "${VARIABLE}")
```
```
macro(XXXX)
   if (true)
```

opdenkamp, 2018-03-16:
I'm not seeing any issues with cmake 3.7.2. Sounds like a bug in your cmake version.

commit c9423dda (2016-01-19): Merge pull request #168 from kingosticks/fix-rpi-check-lib-exists

Don't skip checking for Raspberry Pi library (Fixes #166).
commit f3a09a1d (2016-01-19): Don't skip checking for Rasperry Pi library (Fixes #166).