Pulse-Eight/libcec #169: Undefined symbol: _ZN11StringUtils6FormatEPKcz (Arch Linux)

kroegerama, 2015-12-16:
System: `Linux 3.12.26-1-ARCH #1 PREEMPT Fri Aug 8 17:40:14 MDT 2014 armv6l GNU/Linux`
libcec version: `libcec-rpi-3.0.1-2`

Running `cec-client` returns:
`cec-client: symbol lookup error: /usr/lib/libcec.so.3.0: undefined symbol: _ZN11StringUtils6FormatEPKcz`

iegubkin, 2015-12-17:
I rebuilt kodi-rbp after archlinuxarm's C++ ABI change and got this same libcec error when starting kodi. Maybe also related to this change? https://www.archlinux.org/news/c-abi-change/

/usr/lib/kodi/kodi.bin: symbol lookup error: /usr/lib/libcec.so.3.0: undefined symbol: _ZN11StringUtils6FormatEPKcz

Trying to rebuild libcec (libcec-rpi to 3.0.1-2) on archlinuxarm didn't work however. I received this error:
https://github.com/Pulse-Eight/libcec/issues/115
https://github.com/archlinuxarm/PKGBUILDs/blob/master/alarm/libcec-rpi/PKGBUILD

kroegerama, 2015-12-20:
Solved in `libcec-rpi-3.0.1-3`

closed 2015-12-20 by kroegerama