Pulse-Eight/libcec #172: Raspberry Pi support: no even when variables are set and files are available

voltagex, 2015-12-20:
```
voltagex@kiwi:~/src/libcec/build$ cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=1 -DCMAKE_INSTALL_PREFIX=/usr -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
-- git found: libcec-2.1.4+223-2f62595~dirty
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   no
-- TDA995x support:                        no
-- Exynos support:                         no
-- DRM support:                            yes
-- Python support:                         version 2.7.9 (2.7)
-- lib info: git revision: libcec-2.1.4+223-2f62595~dirty, compiled on Sunday 20 December  08:35:17 UTC 2015 by voltagex@kiwi.voltagex.org on Linux 3.19.3-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/voltagex/src/libcec
```

```
voltagex@kiwi:~/src/libcec/build$ strace -ff cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=1 -DCMAKE_INSTALL_PREFIX=/usr -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib .. 2>&1 | grep bcm_host
voltagex@kiwi:~/src/libcec/build$
```

So cmake isn't actually looking in /opt/vc?

kingosticks, 2015-12-20:
See https://github.com/Pulse-Eight/libcec/pull/168

voltagex, 2015-12-20:
Wish I'd seen that hours ago... thanks.

closed 2016-01-19 by opdenkamp