Pulse-Eight/libcec #174: Yamaha AVR didn´t go standby

vdrrossi, 2015-12-27:
Hi,

the Pulse Eight CEC USB Adapter is connected to HDMI1 of the Yamaha RX-V567 Receiver.
No cec aware TV is connected. I just want basic on/off and volume control from my linux htpc.

On 5 is working, but standby 5 didnt´t work out

root@vdr1:~# echo 'standby 5' | cec-client -s
opening a connection to the CEC adapter...
DEBUG: [ 34] Broadcast (F): osd name set to 'Broadcast'
DEBUG: [ 35] connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG: [ 35] communication thread started
DEBUG: [ 45] turning controlled mode on
NOTICE: [ 66] connection opened
DEBUG: [ 67] << Broadcast (F) -> TV (0): POLL
TRAFFIC: [ 67] << f0
DEBUG: [ 67] setting the line timeout to 3
DEBUG: [ 67] processor thread started
DEBUG: [ 103] CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [ 103] << f0
DEBUG: [ 140] CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: [ 140] >> POLL not sent
DEBUG: [ 140] TV (0): device status changed into 'not present'
NOTICE: [ 140] registering new CEC client - v3.0.1
DEBUG: [ 140] detecting logical address for type 'recording device'
DEBUG: [ 140] trying logical address 'Recorder 1'
DEBUG: [ 140] << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [ 140] << 11
DEBUG: [ 171] CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [ 171] << 11
DEBUG: [ 208] CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: [ 208] >> POLL not sent
DEBUG: [ 208] using logical address 'Recorder 1'
DEBUG: [ 208] Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG: [ 208] Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG: [ 208] Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG: [ 208] Recorder 1 (1): CEC version 1.4
DEBUG: [ 208] AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG: [ 208] setting ackmask to 2
DEBUG: [ 213] Recorder 1 (1): osd name set to 'CECTester'
DEBUG: [ 213] Recorder 1 (1): menu language set to 'eng'
DEBUG: [ 213] GetPhysicalAddress - trying to get the physical address via ADL
DEBUG: [ 214] GetPhysicalAddress - ADL returned physical address 0000
DEBUG: [ 214] GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG: [ 214] GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG: [ 214] GetPhysicalAddress - trying to get the physical address via drm files
DEBUG: [ 214] GetPhysicalAddress - drm files returned physical address 0000
DEBUG: [ 214] GetPhysicalAddress - trying to get the physical address from the OS
DEBUG: [ 214] GetPhysicalAddress - OS returned physical address 0000
NOTICE: [ 214] setting HDMI port to 1 on device TV (0)
DEBUG: [ 214] << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [ 214] << 10
DEBUG: [ 245] CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [ 245] << 10
DEBUG: [ 281] CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: [ 281] >> POLL not sent
DEBUG: [ 281] Recorder 1 (1): physical address changed from ffff to 1000
DEBUG: [ 281] << Recorder 1 (1) -> broadcast (F): physical adddress 1000
TRAFFIC: [ 282] << 1f:84:10:00:01
DEBUG: [ 418] using persisted autonomous mode setting: 'enabled'
DEBUG: [ 423] using persisted CEC version setting: '1.4'
DEBUG: [ 429] using persisted logical address setting: 'Recorder 1'
DEBUG: [ 434] using persisted device type setting: 'recording device'
DEBUG: [ 439] using persisted logical address mask setting: '206'
DEBUG: [ 455] using persisted device name setting: 'CECTester'
DEBUG: [ 460] using persisted physical address setting: '1000'
NOTICE: [ 460] CEC client registered: libCEC version = 3.0.1, client version = 3.0.1, firmware version = 4, firmware build date: Thu Dec 6 11:15:20 2012 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, compiled on Fri Jul 3 17:23:20 UTC 2015 by buildd@lgw01-07.buildd on Linux 2.6.59-21-generic (x86_64), features: P8_USB, P8_detect, randr
DEBUG: [ 460] << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
DEBUG: [ 460] << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [ 460] << 10
DEBUG: [ 491] CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [ 491] << 10
DEBUG: [ 528] CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: [ 528] >> POLL not sent
DEBUG: [ 528] not sending command 'set osd name': destination device 'TV' marked as not present
DEBUG: [ 528] << requesting power status of 'TV' (0)
DEBUG: [ 528] << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [ 528] << 10
DEBUG: [ 559] CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [ 559] << 10
DEBUG: [ 596] CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG: [ 596] >> POLL not sent
DEBUG: [ 596] not sending command 'give device power status': destination device 'TV' marked as not present
DEBUG: [ 596] << Recorder 1 (1) -> Audio (5): POLL
TRAFFIC: [ 596] << 15
DEBUG: [ 633] >> POLL sent
DEBUG: [ 633] Audio (5): device status changed into 'present'
DEBUG: [ 633] << requesting vendor ID of 'Audio' (5)
TRAFFIC: [ 633] << 15:8c
TRAFFIC: [ 832] >> 5f:87:00:a0:de
DEBUG: [ 832] Audio (5): vendor = Yamaha (00a0de)
DEBUG: [ 832] expected response received (87: device vendor id)
NOTICE: [ 832] << putting 'Audio' (5) in standby mode
DEBUG: [ 832] >> Audio (5) -> Broadcast (F): device vendor id (87)
TRAFFIC: [ 832] << 15:36
DEBUG: [ 890] unregistering all CEC clients
NOTICE: [ 890] unregistering client: libCEC version = 3.0.1, client version = 3.0.1, firmware version = 4, firmware build date: Thu Dec 6 11:15:20 2012 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, compiled on Fri Jul 3 17:23:20 UTC 2015 by buildd@lgw01-07.buildd on Linux 2.6.59-21-generic (x86_64), features: P8_USB, P8_detect, randr
DEBUG: [ 890] Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG: [ 890] Recorder 1 (1): vendor = Unknown (000000)
DEBUG: [ 890] Recorder 1 (1): CEC version unknown
DEBUG: [ 890] Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG: [ 890] Recorder 1 (1): device status changed into 'unknown'
DEBUG: [ 890] setting ackmask to 0
DEBUG: [ 895] turning controlled mode off
DEBUG: [ 900] unregistering all CEC clients
DEBUG: [ 1074] communication thread ended

Can you give me a hint ? In supported devices yamaha is marked as yes (for standby)

thanks in advance, Rossi

opdenkamp, 2016-01-19:
The protocol expects that a CEC capable TV is present, which may be causing issues with your AVR. libCEC is sending the correct command here.

can you try this:
`echo 'tx 0F:36' | cec-client -s`

if that doesn't work, try this:
`echo 'tx 15:44:40' | cec-client -s`

opdenkamp, 2016-10-18:
no response, and libCEC sends the correct commands, so closing this one

closed 2016-10-18 by opdenkamp