Pulse-Eight/libcec #175: libcec src build generates incorrect libdir entry in pkgconfig(libcec.pc)

pgnd, 2016-01-02:
Building libcec head on linux,

    cd libcec

    git log | head
        commit 2f625959a43ac68acce70f4e574b6357fe77a273
        Author: Lars Op den Kamp <lars@opdenkamp.eu>
        Date:   Wed Dec 2 19:47:51 2015 +0100

            changed/fixed: don't call PowerOnDevices() if no devices to wake have been configured. issue #97

        commit cc167e6075a6233b618622bbba4c7a3373436f55
        Author: Lars Op den Kamp <lars@opdenkamp.eu>
        Date:   Wed Dec 2 19:47:48 2015 +0100

    mkdir build
    cd build
    cmake .. \
     -DCMAKE_INSTALL_PREFIX=/usr/local \
     -DBUILD_SHARED_LIBS=1 \
     -DCMAKE_INSTALL_LIBDIR=/usr/local/lib64 \
     -DCMAKE_INSTALL_LIBDIR_NOARCH=/usr/local/lib64

    make -j$CORES
    rm -f /usr/local/lib/libcec*
    rm -f /usr/local/lib64/libcec*
    make install
        ...
        -- Installing: /usr/local/lib64/pkgconfig/libcec.pc
        -- Up-to-date: /usr/local/include/libcec/cec.h
        -- Up-to-date: /usr/local/include/libcec/cecc.h
        -- Up-to-date: /usr/local/include/libcec/cecloader.h
        -- Up-to-date: /usr/local/include/libcec/cectypes.h
        -- Installing: /usr/local/include/libcec/version.h
        -- Installing: /usr/local/lib64/libcec.so.3.0.1
        -- Installing: /usr/local/lib64/libcec.so.3
        -- Installing: /usr/local/lib64/libcec.so

note that the libs are where they're intended

    /usr/local/lib64/libcec*

but the pkconfig entry

    cat /usr/local/lib64/pkgconfig/libcec.pc
        prefix=/usr/local
        exec_prefix=
    ??? libdir=/usr/local//usr/local/lib64
        includedir=/usr/local/include

        Name: libcec
        Description: Pulse-Eight libCEC 3.0.1
        URL: http://www.pulse-eight.com/
        Version: 3.0.1
        Requires: p8-platform  
        Libs: -L${libdir} -lcec
        Cflags: -I${includedir} -I${includedir}/libcec

incorrectly sets

    libdir=/usr/local//usr/local/lib64

so

     pkg-config --libs libcec
        -L/usr/local//usr/local/lib64 -L -lcec -lp8-platform

pgnd, 2016-01-02:
similar issue for cmake-platform

so, for both

    git clone https://github.com/Pulse-Eight/platform cmake-platform
    git clone git://github.com/Pulse-Eight/libcec.git

    cd cmake-platform
    mkdir build
    cd build
    cmake .. \
     -DCMAKE_INSTALL_PREFIX=/usr/local \
     -DBUILD_SHARED_LIBS=1 \
     -DCMAKE_INSTALL_LIBDIR=/usr/local/lib64 \
     -DCMAKE_INSTALL_LIBDIR_NOARCH=/usr/local/lib64

    perl -pi -e \
     's|^(libdir)=.*|${1}=/usr/local/lib64|g; \
      s|^(Libs):.*|${1}: -L/usr/local/lib64 -lp8-platform|g' \
    ./p8-platform.pc

    make
    make install

    pkg-config --libs p8-platform
        -L/usr/local/lib64 -lp8-platform

    cd ../../libcec
    mkdir build
    cd build
    cmake .. \
     -DCMAKE_INSTALL_PREFIX=/usr/local \
     -DBUILD_SHARED_LIBS=1 \
     -DCMAKE_INSTALL_LIBDIR=/usr/local/lib64 \
     -DCMAKE_INSTALL_LIBDIR_NOARCH=/usr/local/lib64

    perl -pi -e \
     's|^(libdir)=.*|${1}=/usr/local/lib64|g' \
    ../src/libcec/libcec.pc

    make
    make install

    pkg-config --libs p8-platform
        -L/usr/local/lib64 -lp8-platform 
    pkg-config --libs libcec
        -L/usr/local/lib64 -lcec -lp8-platform 

    ls -al /usr/local/lib64/lib{cec,p8-platform}*
        lrwxrwxrwx  1 root root   11 Jan  1 18:15 /usr/local/lib64/libcec.so -> libcec.so.3*
        lrwxrwxrwx  1 root root   15 Jan  1 18:15 /usr/local/lib64/libcec.so.3 -> libcec.so.3.0.1*
        -rwxr-xr-x+ 1 root root 695K Jan  1 18:15 /usr/local/lib64/libcec.so.3.0.1*
        lrwxrwxrwx  1 root root   21 Jan  1 18:11 /usr/local/lib64/libp8-platform.so -> libp8-platform.so.2.0*
        lrwxrwxrwx  1 root root   23 Jan  1 18:11 /usr/local/lib64/libp8-platform.so.2.0 -> libp8-platform.so.2.0.0*
        -rwxr-xr-x+ 1 root root  57K Jan  1 18:11 /usr/local/lib64/libp8-platform.so.2.0.0*

opdenkamp, 2016-01-19:
it's the same bug as this one: https://github.com/Pulse-Eight/platform/commit/51e2b280b9704413050e01ce260dab1a1f8dc4cb

you need to update the platform lib to include this commit and I need to fix the same thing in libCEC

opdenkamp, 2016-01-19:
set correctly here already, upgrading the platform lib should fix it
https://github.com/Pulse-Eight/libcec/blob/master/cmake/PkgConfigHandler.cmake#L96

closed 2016-01-19 by opdenkamp