Pulse-Eight/libcec #176: CEC only partially working with Samsung UE48H6600

zaphod333, 2016-01-06:
It's been months since I started using an internal Pulse-Eight CEC adapter inside my Intel NUC (54250). The NUC is attached to HDMI Port 1 of my Denon AVR-X1000 receiver which connects to the ARC HDMI port of my Samsung UE48H6600 TV. I started with OpenELEC 14 and was hoping that CEC will eventually work in the future. Now, with release 16, which incorporates libcec-3.0.0 I am still facing my issues and want to address the developers.

The only feature which always works is powering on the NUC when switching the source on the AVR. Also powering on the TV and the AVR works, when selected in the config. After powering Kodi Kodi shows a CEC connection information which sometimes is successful (connected: TV - AVR-X1000) or not. Purely random beheaviour ! In very rare cases I am able to actually control Kodi with my TV remote when the connection successfully completed, but it stops after a certain time.

I checked the log messages (grep'ed all cec relevant log messages from kodi.log) and in the case the connection times out, there are some line errors (LOW_ERROR, HIGH_ERROR) but the communication continues afterwards. Somehow it looks like libcec is not error tolerant and gets stuck by those errors ?! Log files are attached.

I would be grateful if a developer could help me out with my issue.  I can control my AVR and my Bluray player without issues with my TV remote, but not the NUC. The CEC adapter is completely useless right now.

[ceclog_connected.txt](https://github.com/Pulse-Eight/libcec/files/79719/ceclog_connected.txt)
[ceclog_con_timeout.txt](https://github.com/Pulse-Eight/libcec/files/79718/ceclog_con_timeout.txt)

opdenkamp, 2016-01-19:
This is an issue with the cable that you're using most likely, or some other device that's connected to the bus is causing problems. Your TV doesn't ack some commands that are sent and libCEC retries them a bit later. The line errors are indicating the same thing.

Try swapping the HDMI cables and disconnect all other HDMI devices.

As for the device name, CECtester is used by cec-client, Kodi is used by Kodi. If you didn't start cec-client yourself, then something in OpenELEC is calling it (which it shouldn't be doing)

zaphod333, 2016-01-23:
Thanks for your response, I'm quite sure it is not a cabling issue but related to a buggy implementation of CEC in the TV firmware (Anynet+ in case of my Samsung device). When observing the CEC traffic with cec-client, I see line errors and receive failures in some cases when pressing keys on the TV remote, but it always recovers from the error and continues to log the key presses. Listening to the CEC traffic when controlling my Bluray player with the TV remote shows the line errors too, but also in this case it never stops working. Some key presses might get lost, but I can live with that and I would be really grateful if it eventually works out like this when controlling the NUC with my TV remote. 

Unfortunately besides the power on feature, I have no CEC control at all (only in very very rare cases for a short period of time). Is there some difference in the error handling between libcec and cec-client ?

This strange issue with the device name looks like a bug to me. In the log it shows "setting the OSD name to Kodi (previous: CECTester)". Fine, I wait for several minutes to compensate for delayed EEPROM writes (not sure if I really need to wait) and then restart the machine.
After restart it shows "using persisted device name setting: 'CECTester'" and "setting the OSD name to Kodi (previous: CECTester)" again. For some reason the device name doesn't get written in persistent manner.

opdenkamp, 2016-01-23:
"CECTester" is set by cec-client, which you were running. if that's happening without you running it too, so after a reboot, then some script is running before Kodi is started that calls cec-client. that's the only explanation for it to be reset like that, unless the eeprom is broken but I doubt that.

you can test this by unplugging the adapter after a minute and plugging it back in while Kodi is running. if it doesn't report the change then, then it's caused by some script. if you didn't add that yourself, then you'll have to ask openelec devs about it.

if it still says 'CECTester', then there's an issue with the eeprom. that wouldn't result in those line errors though.

as for running cec-client, are you running that without disabling the adapter in Kodi first? if that's the case and cec-client doesn't complain and exit, then it's been compiled incorrectly by openelec (without locking support). this will break cec with line errors and communication failures in the log.

zaphod333, 2016-01-23:
I always diabled the adapter, rebooted and then started cec-client. When enabled, cec-client correctly terminates because the adapter is already in use. The openelec guys obviously did their job correctly.

Regarding your recommendation of unplugging the adapter, booting and replugging, no success. It still shows CECTester at each reboot. Is the device name written at the time when the log message occurs ? If the EEPROM write fails, does libcec throw an error which occurs in the log ?

Back to my main issue, when you look into ceclog_connected.txt, you will only see the line errors as difference. After the line errors the same progress is shown with the only difference that in one log some messages report the detection of my AVR (in this case the volume control had influence on the AVR, in the other case not). I really don't see what is hindering in the case when nothing works out, but maybe some error happens which isn't logged.

opdenkamp, 2016-01-23:
sorry wasn't clear, boot Kodi, wait for a minute or so, so you're sure that the eeprom has been updated, then unplug the adapter, wait a minute, plug it back in and check the log. if it says that it was "CECtester" again, then it may be an issue with the eeprom. libCEC doesn't verify after writing.

as for the line errors, there are none in ceclog_connected.txt (failed ack is not an error in most situations). the odd thing with the other log is this part:
```
13:16:56 T:140300163745536   DEBUG: CecLogMessage - << f0
13:16:56 T:140300163745536   DEBUG: CecLogMessage - setting the line timeout to 3
13:16:56 T:140300163745536   DEBUG: CecLogMessage - >> POLL sent
13:16:56 T:140300163745536   DEBUG: CecLogMessage - TV (0): device status changed into 'present'
13:16:56 T:140300163745536   DEBUG: CecLogMessage - << requesting vendor ID of 'TV' (0)
13:16:56 T:140300163745536   DEBUG: CecLogMessage - << f0:8c
13:16:56 T:140300163745536   DEBUG: CecLogMessage - CEC transmission - received response - TRANSMIT_FAILED_ACK
13:16:56 T:140300163745536   DEBUG: CecLogMessage - << f0:8c
13:16:56 T:140300163745536 WARNING: CecLogMessage - LOW_ERROR line:261 time:1728
13:16:56 T:140300163745536 WARNING: CecLogMessage - RECEIVE_FAILED
13:16:56 T:140300163745536 WARNING: CecLogMessage - HIGH_ERROR line:189 time:1497
13:16:56 T:140300163745536 WARNING: CecLogMessage - HIGH_ERROR line:189 time:1531
13:16:56 T:140300163745536 WARNING: CecLogMessage - HIGH_ERROR line:189 time:1483
13:16:56 T:140300163745536 WARNING: CecLogMessage - HIGH_ERROR line:189 time:645
13:16:56 T:140300163745536 WARNING: CecLogMessage - HIGH_ERROR line:189 time:605
```
first a poll to address 0 is acked, then the next packet is not acked and when we retry, the line goes low when it shouldn't and the retried transmission fails too.

then, we try it again a bit later after we allocated a logical address and it works:
```
13:16:57 T:140300163745536   DEBUG: CecLogMessage - << requesting vendor ID of 'TV' (0)
13:16:57 T:140300163745536   DEBUG: CecLogMessage - << 10:8c
13:16:58 T:140300163745536   DEBUG: CecLogMessage - >> 0f:87:00:00:f0
13:16:58 T:140300163745536   DEBUG: CecLogMessage - TV (0): vendor = Samsung (0000f0)
```

it's possible that the TV doesn't like us sending things from the unregistered address, which matches the broadcast address, but that doesn't explain why you would have issues.

you cut the second log early, before it finished polling all addresses. the message about the AVR being detected pops up after that, after it send some more commands.

but before we start chasing rabbits, let's wait for openelec to pull libCEC 3.1.0 first and get that tested.

zaphod333, 2016-01-24:
I am quite sure now that all my problems are related to EMC problems. The CEC adapter is obvisously vulnerable to noise pickup and my other devices (AVR and Bluray) seem to be much more robust as they always worked fine. When only the power cord from the AC/DC adapter and the HDMI cable are attached to the NUC, everything works fine. Controlling Kodi via the TV remote is possible and also the strange device name issue disappeared. 

Plugging in the ethernet cable brings me back to my old issues. It looks like that it is working until traffic on the ethernet line occurs and I am quite sure the line errors are are caused by noise. Did you implement any filtering caps in the circuit (especially on the CEC line) ?

malard, 2016-01-27:
Your ethernet cable, is it shielded?

zaphod333, 2016-01-27:
I tried both shielded and unshielded ethernet cables, also put ferrite cores on the cables. No success.

malard, 2016-01-27:
It sounds to me like a ground loop if it only happens WHEN you connect an ethernet cable, somehow your computer has connected the ground of the ethernet cable to the USB port, which in turn is connected through our adapter to the HDMI port.

Just something else to test but put the CEC adapter on a USB hub

zaphod333, 2016-01-27:
You are right, but when using a cable without a shield the potential ground loop should be broken. All my devices (TV, AVR, NUC, Ethernet switch) have isolated power supplies. To be honest, I don't have the faintest clue where the noise pickup originates from and why it influences the CEC communication.

malard, 2016-01-27:
I have no idea either, but given we've sold around 30,000 of these little boxes and this isn't being reported widely it is either a faulty adapter or something unique to your setup

opdenkamp, 2016-01-28:
I have checked this with one of our hardware engineers. It could be related to the DP -> HDMI converter that's connected.

Could you try to disconnect the HDMI cable between the converter and USB-CEC adapter and check if you can still reproduce the problem please.

zaphod333, 2016-01-28:
DP = DisplayPort ? I am using a direct HDMI connection from NUC to AVR (Mini-HDMI -> HDMI).

opdenkamp, 2016-01-28:
In that case, can you try disconnecting that part

zaphod333, 2016-01-28:
But then there is no HDMI connection and therefore no CEC communication ?! How should I see any communication problems ?

opdenkamp, 2016-01-28:
Ah hold on, my colleague and I were thinking that you have an external (USB) adapter. I'll talk to him tomorrow.

opdenkamp, 2016-01-29:
Could you send an email to support@pulse-eight.com please and then we'll swap your adapter, to ensure that it's not an issue with the pcb or cable assembly.

zaphod333, 2016-02-07:
Today my OpenELEC OS upgraded to 6.0.1 (including libcec-3.0.1) and the issue is resolved ! After the first reboot with the new version it still showed the old behaviour (CEC works fine without Ethernet link and fails when link is active). I played around with some settings (e.g. activated the "wait until network interface is ready before starting Kodi") and with the next reboot it started to work out even with Ethernet link. It continues to do so when deactivating the wait feature. There must have been a very strange software issue when having a network connection after startup, which made it look like an EMC problem. To be honest, I don't understand it, but I am happy that the issue is resolved finally.

closed 2016-02-07 by zaphod333