Pulse-Eight/libcec #177: Broken on FreeBSD, version 2.x worked fine

uqs, 2016-01-09:
So I ported libplatform and libcec to FreeBSD (10.x in my case) and just running it without any adapter installed will cause a segfault:

no serial port given. trying autodetect: 
 path:     /dev/ttyU0
 com port: /dev/ttyU0

opening a connection to the CEC adapter...
DEBUG:   [               0]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [               1]     connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [               1]     communication thread started
DEBUG:   [             330]     turning controlled mode on
NOTICE:  [             354]     connection opened
DEBUG:   [             354]     << Broadcast (F) -> TV (0): POLL
TRAFFIC: [             354]     << f0
DEBUG:   [             354]     setting the line timeout to 3
DEBUG:   [             354]     processor thread started
DEBUG:   [            1362]     command 'CEC transmission' was not acked by the controller
TRAFFIC: [            1863]     << f0
DEBUG:   [            2867]     command 'CEC transmission' was not acked by the controller
DEBUG:   [            3372]     >> POLL not sent
DEBUG:   [            3372]     TV (0): device status changed into 'not present'
NOTICE:  [            3372]     registering new CEC client - v3.0.1
DEBUG:   [            3372]     detecting logical address for type 'recording device'
DEBUG:   [            3372]     trying logical address 'Recorder 1'
DEBUG:   [            3372]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [            3372]     << 11
DEBUG:   [            4376]     command 'CEC transmission' was not acked by the controller
TRAFFIC: [            4878]     << 11
DEBUG:   [            5882]     command 'CEC transmission' was not acked by the controller
DEBUG:   [            6382]     >> POLL not sent
DEBUG:   [            6382]     using logical address 'Recorder 1'
DEBUG:   [            6382]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            6382]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            6382]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            6382]     Recorder 1 (1): CEC version 1.4
DEBUG:   [            6382]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [            6382]     setting ackmask to  2
DEBUG:   [            7386]     command 'SET_ACK_MASK' was not acked by the controller
DEBUG:   [            7386]     couldn't change the ackmask: the connection is closed
DEBUG:   [            7386]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            7386]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            7386]     GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:   [            7389]     GetPhysicalAddress - ADL returned physical address 0000
DEBUG:   [            7389]     GetPhysicalAddress - trying to get the physical address via nvidia driver

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 802006400 (LWP 100536/cec-client-3.0.1)]
0x00000008015bf69a in pthread_mutex_lock () from /lib/libthr.so.3
(gdb) bt
#0  0x00000008015bf69a in pthread_mutex_lock () from /lib/libthr.so.3
#1  0x0000000801901302 in readdir () from /lib/libc.so.7
#2  0x00000008024702a0 in PLATFORM::CDRMEdidParser::GetPhysicalAddress (this=0x7fffffff7db8)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/platform/drm/drm-edid.cpp:54
#3  0x0000000802480372 in CEC::CUSBCECAdapterCommunication::GetPhysicalAddress (this=0x802085100)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/adapter/Pulse-Eight/USBCECAdapterCommunication.cpp:697
#4  0x00000008024551b6 in CEC::CCECProcessor::GetDetectedPhysicalAddress (this=0x80201b200)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/CECProcessor.cpp:570
#5  0x0000000802445ee4 in CEC::CCECClient::AutodetectPhysicalAddress (this=0x80202b800)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/CECClient.cpp:1224
#6  0x000000080243ed34 in CEC::CCECClient::SetPhysicalAddress (this=0x80202b800, configuration=@0x80202b840)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/CECClient.cpp:236
#7  0x000000080243e67c in CEC::CCECClient::OnRegister (this=0x80202b800)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/CECClient.cpp:138
#8  0x00000008024581cd in CEC::CCECProcessor::RegisterClient (this=0x80201b200, client=<value optimized out>)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/CECProcessor.cpp:899
#9  0x0000000802466b1e in CEC::CLibCEC::Open (this=0x802017100, strPort=0x62d911 "/dev/ttyU0", iTimeoutMs=10000)
    at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/libcec/LibCEC.cpp:91
#10 0x0000000000415c7d in main (argc=1, argv=0x7fffffffe5a8) at /home/uqs/ports/multimedia/libcec/work/libcec-libcec-3.0.1/src/cec-client/cec-client.cpp:1327
(gdb) 

Any idea what's going wrong here?

(gdb) l
49        // Fisrt we look for all DRM subfolder
50        std::string baseDir = "/sys/class/drm/";
51       
52        DIR *dir = opendir(baseDir.c_str());
53
54        struct dirent *entry = readdir(dir);
55        std::string enablededid;
56        std::string line;
57
58        while (entry != NULL)

There is no /sys under FreeBSD.

It looks like src/libcec/cmake/CheckPlatformSupport.cmake needs to make the -DHAS_DRM_EDID_PARSER conditional on whether it's actually Linux and not just !Win32

opdenkamp, 2016-01-19:
right, thanks for testing, will get that disabled on BSD

opdenkamp, 2016-01-20:
it should no longer have crashed with master, that should have been fixed already, but I've disabled it explicitly in the check: https://github.com/Pulse-Eight/libcec/commit/8d5379dce05f646975635d0f83dbe0d69a2e5d20

closed 2016-01-20 by opdenkamp