Pulse-Eight/libcec #178: libcec-3.0.1.tar.gz is incorrectly packed

pingram3030, 2016-01-11:
This seems wrong:
```
-rw-rw-r--. 1 build build  1117284 Jan 11 14:54 libcec-3.0.1.tar.gz
drwxrwxr-x. 9 build build      242 Jul  4  2015 libcec-libcec-3.0.1
```

opdenkamp, 2016-01-19:
where did you find this?

pingram3030, 2016-01-21:
link copied from https://github.com/Pulse-Eight/libcec/releases
```
$ wget -q https://github.com/Pulse-Eight/libcec/archive/libcec-3.0.1.tar.gz
$ tar zxf libcec-3.0.1.tar.gz 
$ ls -lh
total 1.1M
-rw-rw-r--.  1 user   user   1.1M Jan 21 20:16 libcec-3.0.1.tar.gz
drwxrwxr-x.  9 user   user    340 Jul  4  2015 libcec-libcec-3.0.1
...
```

pingram3030, 2016-01-21:
aaand with the latest release.
```
$ wget -q https://github.com/Pulse-Eight/libcec/archive/libcec-3.1.0.tar.gz
$ tar zxf libcec-3.1.0.tar.gz 
$ ll
total 2.2M
-rw-rw-r--. 1 user   user   1.1M Jan 21 20:16 libcec-3.0.1.tar.gz
-rw-rw-r--. 1 user   user   1.1M Jan 21 20:19 libcec-3.1.0.tar.gz
drwxrwxr-x. 9 user   user    340 Jul  4  2015 libcec-libcec-3.0.1
drwxrwxr-x. 9 user   user    340 Jan 20 11:22 libcec-libcec-3.1.0

```

opdenkamp, 2016-01-21:
okay, but that's just how github packages these, we have no control over this. it's "<repo name>-<tagname>"

closed 2016-01-21 by opdenkamp

pingram3030, 2016-01-21:
Wow, that is very.. special..  I just did some testing and.. yeah.. very special.. :)

all good, thanks.