Pulse-Eight/libcec #180: Raspberry Pi2 Raspbian Jessie - CMake Never Detects Libraries

christophersen-alpha, 2016-01-16:
Platform:
Raspberry Pi2
Raspbian Jessie (all upgraded as of this moment)

I followed the instructions in the ReadMe

Running the cmake command as described in the readme never gives "Raspberry Pi support: yes"

The libraries are in /opt/vc/lib

ldconfig sees them:
raspberrypi:/opt/vc/lib $ ldconfig -p | grep -e vcos -e bcm_host -e vchiq
	libvcos.so (libc6,hard-float) => /opt/vc/lib/libvcos.so
	libvchiq_arm.so (libc6,hard-float) => /opt/vc/lib/libvchiq_arm.so
	libbcm_host.so (libc6,hard-float) => /opt/vc/lib/libbcm_host.so

I did the fix from #172 

I run the command as:
cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..

It provides no support for Raspberry Pi.
If I comment out the "if" it stills shows no support but will list RPi as a feature for the library. This still doesn't work.
If comment them out the "if" with fix #172 I get this error during make -j4:

In file included from /home/pi/devel/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:37:0:
/home/pi/devel/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h: In constructor ‘CEC::CRPiCECAdapterCommunication::CRPiCECAdapterCommunication(CEC::IAdapterCommunicationCallback*)’:
/home/pi/devel/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:118:35: warning: ‘CEC::CRPiCECAdapterCommunication::m_bDisableCallbacks’ will be initialized after [-Wreorder]
     bool                          m_bDisableCallbacks;
                                   ^
/home/pi/devel/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:105:10: warning:   ‘bool CEC::CRPiCECAdapterCommunication::m_bInitialised’ [-Wreorder]
     bool m_bInitialised;   /**< true when the connection is initialised, false otherwise */
          ^
/home/pi/devel/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:68:1: warning:   when initialized here [-Wreorder]
 CRPiCECAdapterCommunication::CRPiCECAdapterCommunication(IAdapterCommunicationCallback *callback) :
 ^
[ 24%] 

It still finishes though.

When doing sudo make install it complains that it can't find the libraries:

[100%] Building CXX object src/libcec/CMakeFiles/_cec.dir/libcecPYTHON_wrap.cxx.o
Linking CXX shared module _cec.so
/usr/bin/ld: cannot find -lvcos
/usr/bin/ld: cannot find -lvchiq_arm
/usr/bin/ld: cannot find -lbcm_host
collect2: error: ld returned 1 exit status
src/libcec/CMakeFiles/_cec.dir/build.make:98: recipe for target 'src/libcec/_cec.so' failed
make[2]: *** [src/libcec/_cec.so] Error 1
CMakeFiles/Makefile2:192: recipe for target 'src/libcec/CMakeFiles/_cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/_cec.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

If I then make symlinks in /usr/local/lib/ for all the appropriate libraries and them re-run "sudo make install" it finds them and compiles. But cec-client still doesn't work.

If I make the symlinks before the cmake command and start over it still doesn't find them normally and the make -j4 command still produces the same errors as shown above.

I'm not sure where the issue may be with this.

I would appreciate any insight.

Thanks

christophersen-alpha, 2016-01-16:
Lib raspberrypi versions:

$ dpkg -l libraspberrypi*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version          Architecture     Description
+++-======================-================-================-==================================================
ii  libraspberrypi-bin     1.20151118-1     armhf            Miscellaneous Raspberry Pi utilities
ii  libraspberrypi-dev     1.20151118-1     armhf            EGL/GLES/OpenVG/etc. libraries for the Raspberry P
ii  libraspberrypi-doc     1.20151118-1     armhf            EGL/GLES/OpenVG/etc. libraries for the Raspberry P
ii  libraspberrypi0        1.20151118-1     armhf            EGL/GLES/OpenVG/etc. libraries for the Raspberry P

Versions of dependencies installed in first instructions:

$ dpkg -l cmake liblockdev1-dev libudev-dev libxrandr-dev python-dev swig
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version          Architecture     Description
+++-======================-================-================-==================================================
ii  cmake                  3.0.2-1          armhf            cross-platform, open-source make system
ii  liblockdev1-dev        1.0.3-1.6+b1     armhf            Development library for locking devices
ii  libudev-dev:armhf      215-17+deb8u2    armhf            libudev development files
ii  libxrandr-dev:armhf    2:1.4.2-1        armhf            X11 RandR extension library (development headers)
ii  python-dev             2.7.9-1          armhf            header files and a static library for Python (defa
ii  swig                   2.0.12-1         armhf            Generate scripting interfaces to C/C++ code

$ uname -a
Linux raspberrypi 4.1.15-v7+ #830 SMP Tue Dec 15 17:02:45 GMT 2015 armv7l GNU/Lin

Thanks

kingosticks, 2016-01-16:
Do you mean you used the fix in https://github.com/Pulse-Eight/libcec/pull/168?

christophersen-alpha, 2016-01-17:
Yes #168.

My apologies.

bardal, 2016-01-19:
I successfully built using a fresh Rasbian install, the readme, and #168 only last week. Did you apply #168 using git or did you manually apply the edits?  If the latter, have you ensured your changes match #168 exactly?

christophersen-alpha, 2016-01-19:
I did both manually and through git checkout.

Did you compile it on a raspberry pi2? Perhaps an issue with the firmware.

Though I am guessing its something different. I can't figure out why its not picking up the libraries with cmake.

I was hoping there was some obvious system or something issue I was missing.

christophersen-alpha, 2016-01-19:
Also I just tried again based on your success and fixed the not picking up libraries thing. I rebooted the pi (thought I had done that but ...) and it find the libraries and cmake says raspberry pi support properly now.

However I still those errors during compile with make -j4 as listed in the first comment and cec-client still doesn't find any adapters.

Thanks

christophersen-alpha, 2016-01-19:
Even further.

cec-client -l says "Found Devices: NONE"
and cec-client -s says "failed to open vchiq instance"

Thanks

christophersen-alpha, 2016-01-19:
And because I was finally hit with inspiration.

Even with macgyvering the Checkplatform... and at least getting it to compile when it said RPi feature it probably was working. However my user didn't have permission to /dev/vchiq. I fixed that and bam it works properly.

Thank you for your comment bardal, it made me go back and check everything and it all works.

closed 2016-01-19 by christophersen-alpha