Pulse-Eight/libcec #181: simultaneous configuration of hdmi port for tv and avr

sahadum, 2016-01-17:
Hi,
I searched the forums and websites for a solution to configure the hdmi port for the tv and the avr in kodi/libcec. The configuration of libcec inside Kodi only supports to configure the hdmi port for the tv OR the avr but not both at the same time.
My setup:
Samsung TV (with one connect mini) 
--TV HDMI Port 4--> Marantz AVR 
----AVR HDMI Port 1--> VU+ Solo2 (sat) with Vti Image(v9.x)
----AVR HDMI Port 5--> RPi2 with OpenElec 6/Kodi/libcec(v3.0.0 and v3.0.1 tested)

My Problem:
I configured Kodi/libcec with the option “Connected to: AVR” and “HDMI Port Number: 5”. Changing the input source on Samsung tv will change the avr port to 5, that correct, but always changes the tv port to 1, that’s not correct.
Using “Connected to: TV” and “HDMI Port Number: 4” did not help. Changing the source now has no effect on the avr and still change the tv port to 1.
The reason for me to use TV HDMI Port 4 is that Samsung TV only supports ARC (audio return channel) on HDMI Port 4. This is necessary to get the sound ex. from streaming apps or youtube.

Possible solution:
Like the Vti Image for the VU+ Solo2 it would be great to have the possibility to configure tv **AND** avr HDMI port for Kodi/libcec. As example, setting options in Vti Cec Config to TV HDMI = 4 and AVR HDMI = 1 brings up the Vu+ to tv and avr. If theses configuration options are already possible within libcec but not available at the gui of kodi please let me know what I have to do.

Thx.

opdenkamp, 2016-01-19:
If it's the raspberry pi, then you don't need the settings in Kodi because they're read from the EDID. But if your TV switches to the wrong input, then that EDID is somehow invalid.

You can override the physical address in Kodi, in the settings for libCEC. Setting physical address to 4500 should work for you.

Closing this issue since it's not a problem in libCEC

closed 2016-01-19 by opdenkamp

sahadum, 2016-01-19:
Hi,

thx for the help. I will try this.

Closed #181<https://github.com/Pulse-Eight/libcec/issues/181>.

-