Pulse-Eight/libcec #185: libcec3 v3.1.0 does not work with LG TV's

smorar, 2016-01-24:
After upgrading to 3.1.0, libcec no longer detects events from my LG tv's (55LB6500-TH) remote control.
Downgrading bock to 3.0.1 resolved this issue.

DjDiabolik, 2016-05-29:
You have install the 3.1.0 on ? You thinks i can try to upgrade on my OSMC on a RPi2 ??
The current image of OSMC use the 3.0.1 and i have some trouble whit that:
I disabled the "Autoswitch source feature" when i turn on my Pi2.... it's ok but if restart my Pi2 the Simplink stop to work and i need to switch source on tv one time and back to hdmi of pi2 and simplink restart to working.
If i enable ths "Autoswitch source feature" this little troubles disappears and all working whitout any problems........

opdenkamp, 2016-08-02:
Can I get a log from this when it doesn't work please. Also, which model year is your LG panel?

DjDiabolik, 2016-08-11:
My LG it's a 49LF590V-ZA whit latest firmware recently installed but this little trouble remain unresolved :)

opdenkamp, 2016-10-18:
I've just tested the latest build with a webos2.0 LG TV and and old one from 2007.

If this issue remains after the next release (3.2 or 4.0), please open a new ticket and attach a log

closed 2016-10-18 by opdenkamp

kkkkamal, 2017-06-03:
I need to enable simplink on boot automatically. But, currently we have to do it manually every-time from Remote Control using the Settings->Input->Simplink->HDD Recorder.

I have been stuck for almost a week trying to figure out what the issue is. We are using a cec-client to control the LG TV. Is there a way to enable simplink on the TV from the Pi or any methods to enable simplink from the Pi side.

Hardware Used:

Raspberry PI LG Television ( 32LF565B ) Any input is highly appreciated.