Pulse-Eight/libcec #186: Switching to my RPi2 powers off my AVR

Theetjuh, 2016-01-25:
Hi,

I'm using libcec since forever on raspbmc and now on osmc.
Allmost always running perfectly and if not helped by their great community.
The issue I'm running into is since I bought a new AVR a year ago.
Sometimes it doesn't and sometimes it does powers off my AVR when I switch sources from lets say PS4 to RPi2.

I made two debug logs just today, this was on a fresh Kodi Jarvis (v16 RC1) which has the same symptops.
First one: http://paste.osmc.io/irimuronit
Second one: http://paste.osmc.io/inituqeqer

I grabbed the logs when the issue happend, before powering the AVR on again.
This is tested with and without the PlayStations connected to ruleout other activity.

My equipment:
Sony KD-65X8505C
Onkyo TX-NR646
Sony PlayStation 3
Sony PlayStation 4
Raspberry Pi 2 (latest version OSMC)

Thanks in advance,
Mathé

elouet, 2016-01-29:
This looks exactly like the issue I saw (#165). What is odd about the log is that the AVR doesn't respond to repeated <give device power status> sent by libcec, and presumably this causes the attempt to power on the AVR, which is already on unfortunately. Sending a "power key" press (toggle type key) causes the problem.

Theetjuh, 2016-01-29:
Nice to see someone else with the same issues!
I responded to your issue as well, hopefully this help.

Theetjuh, 2016-02-14:
Not really much response...

malard, 2016-02-14:
We have been away at a tradeshow for the past couple of weeks, so will pick this up soon

Cheers

dsebastien, 2016-02-18:
+1, I've got this issue also with an Onkyo TX-NR636

popcornmix, 2016-03-20:
Latest Milhouse nightly build on Pi includes https://github.com/Pulse-Eight/libcec/pull/208.
Would be interesting to hear reports if this improves behaviour.

Theetjuh, 2016-03-31:
Looks like the patch popcornmix is talking about fixed the issue!

dsebastien, 2016-03-31:
I've installed the latest OpenElec RC and the problem seems to be fixed for me. @popcornmix you are my hero! :)

dsebastien, 2016-04-24:
I was a bit optimistic, the issue is still there :(
I'm now using the latest OpenElec beta 2 version on an Rpi 3.

Here's a bit more info about my config and how the problem occurs for me:

Initial state:
* my TV (Samsung UE46 ES6710) is powered up
* my TV is on HDMI 1 source (where the receiver is plugged in)
* my TV is NOT yet connected to Kodi through its "AnyNet"
* my receiver is powered on
* Kodi is powered on

Then
* I switch my receiver to the source where Kodi is (no problem so far)
* I connect my TV to AnyNet (I select Kodi in the list)

At this point the CEC mess starts:
* receiver shuts off
* the TV loses the signal to HDMI 1
* I power on the receiver
* the TV gets the signal back
* the receiver shuts off again
* ...

It happens multiple times (like 5-10 at least) and it feels like that moving the arrows on the TV remote somewhat helps (though this is probably an illusion :p).

At some point, the receiver stays on and I can use my TV remote to control Kodi.

I don't know if this might also be related, but I've configured my receiver so that when I'm on the Stream Box source (i.e., Kodi), the remote acts as my TV remote (allowing me to control Kodi from both the TV remote and the receiver remote).

Here's what I found in the log:
```
20:47:16   7.450990 T:1963040768  NOTICE: Starting Kodi (16.1-RC2 Git:4cf382f). Platform: Linux ARM 32-bit
20:47:16   7.451022 T:1963040768  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
20:47:16   7.451057 T:1963040768  NOTICE: Kodi compiled Apr  7 2016 by GCC 5.3.0 for Linux ARM 32-bit version 4.4.6 (263174)
20:47:16   7.451260 T:1963040768  NOTICE: Running on OpenELEC (official) - Version: 6.95.2 7.0, kernel: Linux ARM 32-bit version 4.4.6
20:47:16   7.451291 T:1963040768  NOTICE: FFmpeg version: 2.8.6
20:47:16   7.451318 T:1963040768  NOTICE: WARNING: unsupported ffmpeg version detected
20:47:16   7.451348 T:1963040768  NOTICE: Host CPU: ARMv7 Processor rev 4 (v7l), 4 cores available
20:47:16   7.451375 T:1963040768  NOTICE: ARM Features: Neon enabled
20:47:16   7.451403 T:1963040768  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
20:47:16   7.451436 T:1963040768  NOTICE: special://xbmcbin/ is mapped to: /usr/lib/kodi
20:47:16   7.451464 T:1963040768  NOTICE: special://masterprofile/ is mapped to: /storage/.kodi/userdata
20:47:16   7.451493 T:1963040768  NOTICE: special://home/ is mapped to: /storage/.kodi
20:47:16   7.451521 T:1963040768  NOTICE: special://temp/ is mapped to: /storage/.kodi/temp
20:47:16   7.451579 T:1963040768  NOTICE: The executable running is: /usr/lib/kodi/kodi.bin
20:47:16   7.451640 T:1963040768  NOTICE: Local hostname: OpenELEC
20:47:16   7.451670 T:1963040768  NOTICE: Log File is located: /storage/.kodi/temp/kodi.log
20:47:16   7.458201 T:1963040768  NOTICE: -----------------------------------------------------------------------
20:47:16   7.559525 T:1963040768   ERROR: DBus: Error org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
20:47:16   7.566553 T:1963040768  NOTICE: load settings...
20:47:16   7.672486 T:1963040768  NOTICE: Found 2 Lists of Devices
20:47:16   7.672567 T:1963040768  NOTICE: Enumerated PI devices:
20:47:16   7.672597 T:1963040768  NOTICE:     Device 1
20:47:16   7.672953 T:1963040768  NOTICE:         m_deviceName      : HDMI
20:47:16   7.672987 T:1963040768  NOTICE:         m_displayName     : HDMI
20:47:16   7.673016 T:1963040768  NOTICE:         m_displayNameExtra:
20:47:16   7.673046 T:1963040768  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
20:47:16   7.673074 T:1963040768  NOTICE:         m_channels        : FL,FR
20:47:16   7.673103 T:1963040768  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,24000,32000,44100,48000,88200,96000,176400,192000
20:47:16   7.673134 T:1963040768  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3
20:47:16   7.673180 T:1963040768  NOTICE:     Device 2
20:47:16   7.673238 T:1963040768  NOTICE:         m_deviceName      : Analogue
20:47:16   7.673276 T:1963040768  NOTICE:         m_displayName     : Analogue
20:47:16   7.673305 T:1963040768  NOTICE:         m_displayNameExtra:
20:47:16   7.673366 T:1963040768  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
20:47:16   7.673396 T:1963040768  NOTICE:         m_channels        : FL,FR
20:47:16   7.673424 T:1963040768  NOTICE:         m_sampleRates     : 48000
20:47:16   7.673454 T:1963040768  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
20:47:16   7.673482 T:1963040768  NOTICE:     Device 3
[... 199 lines cut ...]
20:47:39  30.948624 T:1858048928   ERROR: CecLogMessage - could not open a connection (try 3)
20:47:40  31.949158 T:1858048928   ERROR: CecLogMessage - RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
20:47:40  31.949301 T:1858048928   ERROR: CecLogMessage - Open - vc_cec could not be initialised
20:47:40  31.949379 T:1858048928   ERROR: CecLogMessage - could not open a connection (try 4)
20:47:41  32.949917 T:1858048928   ERROR: CecLogMessage - RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
20:47:41  32.950111 T:1858048928   ERROR: CecLogMessage - Open - vc_cec could not be initialised
[... cut ...]
20:47:46  37.953735 T:1858048928   ERROR: CecLogMessage - RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
20:47:46  37.953884 T:1858048928   ERROR: CecLogMessage - Open - vc_cec could not be initialised
20:47:46  37.953949 T:1858048928   ERROR: CecLogMessage - could not open a connection (try 10)
20:47:47  38.954189 T:1849660320   ERROR: OpenConnection - could not opening a connection to the CEC adapter
20:47:47  38.954372 T:1858048928   ERROR: CecLogMessage - could not start CEC communications
```

opdenkamp, 2016-10-25:
https://github.com/Pulse-Eight/libcec/commit/7ce191f7463c16786ba988009090969100ec874d should work around the issue that's reported by @Theetjuh 

the problem reported by @dsebastien is #191

closed 2016-10-25 by opdenkamp