Pulse-Eight/libcec #187: Send <set system audio mode> message

elouet, 2016-01-29:
I have been looking at the source code in order to understand how a recorder might be able to set an AVR in operation or, alternatively, to disable AVR amplification and let the TV play audio streams.

The only place in the source where I found a reference to <set system audio mode> is in CCECCommandHandler::ActivateSource, but then if I'm not mistaken the command seems to be transmitted with an int flag as argument in spite of a physical address (or the absence thereof).

Furthermore I see no way to disable, for instance in a libcec config entry, the automatic emission of this message, in case one does not wish the recorder to interract with the AVR, is there a way currently ?

ChrisBrandhorst, 2016-01-31:
I would like the same thing. My Sony has an menu option for setting the System Audio Mode. However, in my setup I want to fire this command from my Raspberry Pi. I tried the following things:

- `1F:72:01` (Recording --> Broadcast, no response from the bus)
- `15:72:01` (Recording --> Audio system, response: feature abort from Audio)
- `0F:72:01` (TV --> Broadcast, response: initiator 'TV' is not supported by the CEC adapter)
- `05:72:01` (TV --> Audio system, response: initiator 'TV' is not supported by the CEC adapter)

I can see that the last response follows from `RPiCECAdapterCommunication.h`:
`
bool SupportsSourceLogicalAddress(const cec_logical_address address) { return address > CECDEVICE_TV && address < CECDEVICE_BROADCAST; }
`
But I can imagine that making the RPi send as a TV causes other problems. Is what we want possible?

elouet, 2016-02-01:
Actually I made a mistake : the request I meant to send in order to trigger AVR sound output is called <system audio mode request> (0x70). The AVR responds by confirming audio mode with a broadcast. This works well on me setup.

For instance :
[image]

elouet, 2016-03-30:
I have a proposal for a patch to send the desired message : 
https://github.com/elouet/libcec/commit/9b6ea4e10ef53e89510ef27460654d5378418bc6

ChrisBrandhorst, 2016-05-28:
Thanks for your write-up! This led me to the solutions for my setup. My recorder has address 2.6.0.0.
For switching audio to AVR: 55:70:26:00
For switching audio back to TV: 55:70

No changes in code necessary for my setup (Sony TV, Marantz AVR, Raspberry Pi recorder)

commit f7551592 (2016-08-02): added: EnableAudio() API command

Closes #187. Closes #220
closed 2016-08-02 by opdenkamp