Pulse-Eight/libcec #189: Update binaries available on website

gritstub, 2016-02-02:
This is a feature request for updated binaries, in particular for Windows.

The latest that I can find is v3.0.0 from the blog on May 13th, 2015 [1], while the official website has v2.2.0 [2]

Is this available somewhere else?

[1] http://blog.pulse-eight.com/2015/05/13/libcecv3-released/
[2] http://packages.pulse-eight.net/

Thanks!

wingrunr21, 2016-05-07:
This would be nice given that there were a few bug fixes in 3.0.1 that related to LG tvs.

afalout, 2016-05-15:
+1 for updated binaries. Windows 10 related updates would be of great help - #190 #130 etc. Thanks.

CyberMonk, 2016-06-21:
Throwing in another request for updated Windows binaries to be posted; looks like 3.1.0 should be available, buy only 3.0.0 listed on site.

opdenkamp, 2016-10-25:
the binaries will be updated with the 4.0.0 release, planned for this week

closed 2016-10-25 by opdenkamp

afalout, 2016-11-03:
Cant find 4.0.0 release mentioned - was it updated? Thenks

afalout, 2016-11-15:
Looks like this issue needs to be reopened as binaries are still not available. Thanks.