Pulse-Eight/libcec #190: CEC-USB Adapter loses connection when Windows resume from suspend

vinser, 2016-02-02:
USB-CEC Adapter (firmware v.5) does not properly work on Windows 10 + Kodi 15.2 and 16RC2 too.

It works fine with Kodi 14.2 (libcec 2.2.2). When I switch off my TV with remote Kodi PC goes Suspend. When I turn TV on and select with Samsung remote Anynet+ -> Select device -> Recorder then Kodi PC wakes up and USB-CEC adapter does reconnect.

And on Kodi 15.2 and 16RC2 USB-CEC adapter does not reconnect after PC wakes up with Anynet+ and Kodi pops up info window about lost connection

Something was broken in libcec, because in Kodi 14.2 they use libcec 2.2.0 and in Kodi 15.2 and 16rc2 - libcec 3.0.0

vinser, 2016-02-02:
Tell me please how can I install libsec 2.2.0 insteas of 3.0.0 installed with Kodi 15.2?

opdenkamp, 2016-02-25:
These versions are incompatible. Can you upload a debug log from this problem. See http://kodi.wiki/view/Log_file/Easy for instructions on how to upload a log.

vinser, 2016-02-27:
I test Subj on 16.0 and it hangs too but now there is no message about connection lost

I uploaded my kodi.log here:
[kodi.log.txt](https://github.com/Pulse-Eight/libcec/files/149204/kodi.log.txt)

dchubkov, 2016-05-14:
I can confirm that that this problem definitely exists and I can reproduce it on latest Kodi version and Windows 10. Each time when I suspend my pc and wake from sleep - then Kodi always hangs

CyberMonk, 2016-06-21:
Also confirming this problems exists under Windows 10 and Kodi 16.1. Occurring with internal Pulse Eight CEC adapter in Skylake NUC. On wake from suspend, loses connection every time and causes Kodi to hang. When launching Kodi from desktop, however, CEC adapter is recognized and functions normally.

Stoff777, 2016-06-28:
Another confirmation. Immediate crash on wake. Very annoying. Win 10 and OpenELEC have the same behaviour.

opdenkamp, 2016-06-29:
this is a problem in Kodi, not libCEC.

```
09:15:32 T:5112    INFO: Asking Windows to suspend...
09:15:32 T:4120   DEBUG: PERIPHERALS::CPeripherals::GetMappingForDevice - device (2548:1002) mapped to Pulse-Eight CEC Adapter (type = cec)
09:15:32 T:4320   DEBUG: WM_POWERBROADCAST: PBT_APMSUSPEND event was sent
09:16:08 T:4320   DEBUG: CAnnouncementManager - Announcement: OnSleep from xbmc
09:16:08 T:4320   DEBUG: GOT ANNOUNCEMENT, type: 8, from xbmc, message OnSleep
```

Kodi fires the OnSleep event after Windows resumed instead of before, and libCEC tells Kodi that the connection was closed. Kodi then doesn't re-open the connection as it should. Likely a deadlock in Kodi's implementation.

closed 2016-06-29 by opdenkamp

josh4trunks, 2016-12-15:
@opdenkamp I am experiencing a similar problem and had a question you might be able to help with.

* If I wake my computer up from suspend (for example using a USB remote/keyboard), **my TV and Receiver do not turn on**. It seems like my computer is on but can't initialize a display, maybe an issue with Xorg without a monitor?
* If I instead use my TV remote to wake my TV it successfully uses CEC to wake my Receiver and computer as expected.
* Using my USB remote/keyboard does successfully suspend my computer and use CEC to turn off my TV and Receiver as expected.

**Would this bug in Kodi explain this behavior?** If so I can submit a ticket to Kodi.

Thank you; my operating system is LibreELEC x64 7.0.2.

opdenkamp, 2017-07-13:
Very late response, but no, this problem was Windows only. If it doesn't work on LibreELEC/OpenELEC, then it's something else, but it's very unlikely to be a problem in libCEC. It's an issue in the OS or in Kodi. Please open a report there if you haven't already