Pulse-Eight/libcec #191: cec-client cannot open connection on RPI

spectrokid, 2016-02-03:
On raspberry II , I followed the compile instructions (fresh raspbian copy) and didn't get any error messages during compile/make.
 Version:
   ```
 pi@raspberrypi:~ $ sudo cec-client -i
    libCEC version: 3.1.0, git revision: libcec-3.1.0+5-6d68d21, compiled on Wed  3 Feb 18:53:24 UTC 2016 by pi@raspberrypi on Linux 4.1.13-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
```

**_result_**
```
pi@raspberrypi:~ $ sudo cec-client -m
starting a monitor-only client. use 'mon 0' to switch to normal mode
No device type given. Using 'recording device'
CEC Parser created - libCEC version 3.1.0
no serial port given. trying autodetect:
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [              68]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              68]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [              68]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [              68]     InitHostCEC - vchi_connect succeeded
ERROR:   [              69]     RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
ERROR:   [              69]     Open - vc_cec could not be initialised
ERROR:   [              69]     could not open a connection (try 1)

```

opdenkamp, 2016-02-25:
the firmware is rejecting `vc_cec_set_logical_address(E)` for some reason. perhaps @popcornmix has an idea why? haven't seen that happening for address E before

dsebastien, 2016-04-24:
Same issue for me (...address(E) also shows up in my logs: #186

voltagex, 2016-09-07:
Ping, just hit this same issue. 

```
voltagex@kiwi:/opt/vc/bin$ cec-client -i
libCEC version: 3.1.0, compiled on Linux-4.1.17-v7+ ... , features: P8_USB, DRM, P8_detect, randr, RPi
```

voltagex, 2016-09-07:
Upgraded to 4.4.19-v7+ and it worked. I'm going to guess that the libraries/headers I built against were "too new" for the rest of the system.

pantomico, 2016-09-10:
Nope:```
 sudo cec-client -i
`libCEC version: 3.1.0, git revision: libcec-3.1.0+25-62b8520, compiled on Sat 10 Sep 13:21:22 UTC 2016 by pi@raspberrypi on **Linux 4.4.20-v7**+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi`

Still getting same error:
`ERROR:   [            9060]     RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
`
```

pantomico, 2016-09-12:
@voltagex : would you be willing to dropbox me a complete drive image of your Raspberry?  This issue has been bugging me for 6 months now...

viruzofficial, 2016-10-08:
I'm also running in the same issue ... 

libCEC version: 3.1.0, git revision: libcec-3.1.0+25-62b8520, compiled on Sat  8 Oct 10:25:51 UTC 2016 by pi@raspberrypi on Linux 4.4.13-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi

Can anyone help?

voltagex, 2016-10-08:
Yeah, what does cec-client -s say?

viruzofficial, 2016-10-10:
`cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              81]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              82]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [              82]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [              82]     InitHostCEC - vchi_connect succeeded
ERROR:   [              83]     RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
ERROR:   [              83]     Open - vc_cec could not be initialised
ERROR:   [              83]     could not open a connection (try 1)
DEBUG:   [            1084]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [            1084]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [            1084]     InitHostCEC - vchi_connect succeeded
ERROR:   [            1085]     RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
ERROR:   [            1085]     Open - vc_cec could not be initialised
ERROR:   [            1085]     could not open a connection (try 2)`

Don't know if this is helpful:(

pantomico, 2016-10-10:
yup, same here:

```
pi@raspberrypi:~ $ cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              79]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              79]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [              79]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [              79]     InitHostCEC - vchi_connect succeeded
ERROR:   [              80]     RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
ERROR:   [              80]     Open - vc_cec could not be initialised
ERROR:   [              80]     could not open a connection (try 1)
DEBUG:   [            1081]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [            1081]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [            1081]     InitHostCEC - vchi_connect succeeded
ERROR:   [            1081]     RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
```

petonic, 2016-10-14:
Same here....  Same output as pantomicio.

My output of `cec-client -i` is: 

    pi@pithy:/opt/vc/bin$ cec-client -i
    libCEC version: 3.1.0, git revision: libcec-3.1.0+25-62b8520, compiled on Fri Oct 14 17:58:22 UTC 2016 by pi@pithy on Linux 4.4.21-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
    pi@pithy:/opt/vc/bin$

voltagex, 2016-10-15:
I suspect you all need to do a kernel+firmware upgrade.

pantomico, 2016-10-15:
Nope, this was right after an rpi-update followed by an apt-get upgrade.

opdenkamp, 2016-10-18:
I cannot reproduce this when downloading the latest Raspbian Jessie from https://www.raspberrypi.org/downloads/raspbian/ and following the instructions from https://github.com/Pulse-Eight/libcec/blob/master/README.md to the letter.

opdenkamp, 2016-10-18:
Perhaps you're trying to use cec-client while another program (Kodi?) is using it, which won't work. As this doesn't happen when using a clean installation and following the instructions, I'm closing this issue.

opdenkamp, 2016-10-18:
Right, let's re-open this one. Can one of you who's having this issue try to compile it with commit https://github.com/Pulse-Eight/libcec/commit/9ddb93b91a9e5dc7d89cccd6c593c64cf57a6d92 reverted and report back.

opdenkamp, 2016-10-18:
@MilhouseVH was kind enough to do an openelec build without 9ddb93b in: http://milhouse.libreelec.tv/builds/testing/RPi2/LibreELEC-RPi2.arm-8.0-Milhouse-20161018174530-%231018y-gdc3b078.tar

MilhouseVH, 2016-10-18:
LibreELEC, but close enough... :smile:

pantomico, 2016-10-18:
I'm a complete newby at Git, so I don't know how to do this automatically.  I went through the source and removed the offending lines by hand.  I (believe it or not) got it to compile, but when I ran the client, I got precisely the same eror message as before.

pantomico, 2016-10-20:
Ok, the plot is thickening.  I had to buy a new memory card so I bought one with Raspbian preinstalled.  From a fresh install, I got it to work.  One thing I thought about is that I configured it to start a graphical desktop, where I previously was set up to start in CLI.  Could this make the difference?

I did get these compilation warnings:
`[ 21%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/AdapterFactory.cpp.o
In file included from /home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:37:0:
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h: In constructor ‘CEC::CRPiCECAdapterCommunication::CRPiCECAdapterCommunication(CEC::IAdapterCommunicationCallback*)’:
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:118:35: warning: ‘CEC::CRPiCECAdapterCommunication::m_bDisableCallbacks’ will be initialized after [-Wreorder]
     bool                          m_bDisableCallbacks;
                                   ^
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:105:10: warning:   ‘bool CEC::CRPiCECAdapterCommunication::m_bInitialised’ [-Wreorder]
     bool m_bInitialised;   /**< true when the connection is initialised, false otherwise */
          ^
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:68:1: warning:   when initialized here [-Wreorder]
 CRPiCECAdapterCommunication::CRPiCECAdapterCommunication(IAdapterCommunicationCallback *callback) :
 ^
[ 24%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/Pulse-Eight/USBCECAdapterMessage.cpp.o
`

21bob21, 2016-12-26:
hi

inphinitum, 2017-01-03:
I'm having the same issue here. Vanilla Raspbian cli version, libcec compiled from master.

chrisbward, 2017-01-04:
Having the same issue on Jessie on a Pi B model;
libCEC version: 4.0.1, git revision: libcec-4.0.1, compiled on Wed  4 Jan 12:49:31 UTC 2017 by pi@raspberrypi-tv on Linux 4.4.38+ (armv6l), features: P8_USB, DRM, P8_detect, randr, RPi

inphinitum, 2017-01-04:
@chrisbward have you tried to force HDMI output via `raspi-config`? I did so and the problem went away.

chrisbward, 2017-01-04:
also done that - this is from a blank Jessie install, however my kernel was at 4.1.x before - tried an rpi-update and then make install on-top of the original 
- will I need to clear the system out before I build again?

opdenkamp, 2017-01-04:
I tried this with a clean installation, on a blank SD card, a while ago and couldn't reproduce it.

The symptoms suggest that the kernel checks for HPD of the connected sink and will return an error if it doesn't detect this. I've seen other hardware that does similar things, but when I spoke to the engineer who works on the Pi's firmware yesterday, he told me that he didn't think that it was the case for the Pi, but wasn't sure. He said that he'll take a look at it once he's back from travelling.

It's not a problem with libCEC or something that libCEC can address. Just keeping this ticket open for now, until I've heard back about the firmware.

chrisbward, 2017-01-04:
I will wipe the SD and follow the instructions again;
https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md

chrisbward, 2017-01-04:
okay... seems to be working now...!

biggest differences in this instance;
- avoided running apt-get install cec-utils
- installed and ran rpi-update from 4.4.34 to 4.4.38 (raspbian jessie lite - 25th nov 16)
- turned on audio for HDMI in raspi-config
- followed instructions in link in above comment

thanks guys, will start testing now

sterling, 2017-01-13:
I was able to fix this problem for myself by doing: `tvservice --off`. This will power down the HDMI and open it up for connections again if something you can't identify has taken over.

jsan55, 2017-02-13:
Thanks sterilng,
 "**tvservice - off**" was also my solution. I have summarized the most important here.
Control your LG Smart TV with Linux 
[http://www.pedaltreter.at/jsan/lgtv/LG_Smart_TV_CEC_en.html](url)

opdenkamp, 2017-03-15:
I've added a warning in libCEC, when it detects this. Thanks

closed 2017-03-15 by opdenkamp