Pulse-Eight/libcec #193: ubuntu modemmanager  issue

9000h, 2016-02-05:
the Ubuntu ModemManager is trying to open the serial port /dev/ttyACM0 and as this is no modem an error get logged, to avoid this a udev rule can be used to prevent this
```
# Ubuntu ModemManager should ignore /dev/ttyACM0  
SUBSYSTEM=="usb", ENV{ID_VENDOR_ID}=="2548", ENV{ID_MODEL_ID}=="1002" ENV{ID_MM_DEVICE_IGNORE}="1"
```

opdenkamp, 2016-02-25:
ah nice, I'll see if I can get that hooked up in our .deb

UEckhardt, 2016-04-23:
I had nearly the same problem on a OpenSuSE 13.2, but also mtp-probe tried to access the serial port. To solve this I have extended the above udev rule and installed this rule as /etc/udev/rules.d/20-libcec.rules
```
# ModemManager and mtp probe should ignore /dev/ttyACM0  
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="2548",  ATTRS{idProduct}=="1002", ENV{ID_MM_DEVICE_INORE}="1", ENV{MTP_NO_PROBE}="1"
```

commit 71dbfcdc (2016-08-02): added: udev rule to prevent modemmanager and mtp from probing the CEC adapter, which prevents it from working right after it gets plugged in

Closes #193
closed 2016-08-02 by opdenkamp