Pulse-Eight/libcec #194: Unable to load USB-CEC adapter when running eventghost

mattail, 2016-02-07:
Hi
First, eventghost is new to me so I apologize in advance if I misunderstand the use of it with the pulse-eight USB-CEC adapter.
I have an intel NUC 5i5RYH running Plex in a closed cabinet which is one of the reason why I chose the USB-CEC adapter. It works perfectly once Plex i launched from desktop. But what I want to do is when I press the blue button on my TV remote to launch Plex.
Now for that I use eventghost and I see in the eventghost window the event "CEC.F1 (blue)" which I use as the trigger to open Plex. And Plex does indeed open but the after Plex is loaded, a message saying that the adapter is not loaded properly is displayed making it impossible to control plex. 
Actual it looks like when the libcec pluging is loaded in eventghost plex is not able to load the adapter correctly. Is it because Eventghost uses the HDMI cec connection already?
Here's 2 examples of my eventghost script. I have tried also to use the "CEC.FI (blue)" trigger to emulate "WIN+1" keystroke which also launches plex but same error once Plex is launched. 

`<?xml version="1.0" encoding="UTF-8"?>
-<EventGhost Time="1454793086.48" Guid="{E7B13C0D-9DFE-431B-88E8-2F350F181068}" Version="1722">
-<Autostart Expanded="True" Name="Autostart">
<Plugin Guid="{FD322EEA-C897-470C-BEF7-77BF15C52DB4}" File="libCEC" Identifier="CEC"> gAKISwVYBgAAAGxpYkNFQ3EAh3EBLg== </Plugin>
<Plugin Guid="{59CBD10F-C1D8-4ADB-999B-9B76BA360F1F}" File="Keyboard" Identifier="Keyboard"> gAKIhXEALg== </Plugin>
</Autostart>
-<Folder Expanded="True" Name="plex keyboard">
-<Macro Expanded="True" Name="Emulate Keystrokes: {Win+1}">
<Event Name="CEC.F1 (blue)"/>
<Action> Window.SendKeys(u'{Win+1}', False, 2) </Action>
</Macro>
</Folder>
-<Folder Name="Plex" Enabled="False" id="5">
-<Macro Expanded="True" Name="Start Program: Plex Home Theater.exe">
<Event Name="CEC.F1 (blue)" id="17"/>
<Action> System.Execute(u'C:\\Program Files (x86)\\Plex Home Theater\\Plex Home Theater.exe', u'', 2, False, 5, u'', False, False, u'', False, False, False, False) </Action>
<Action> EventGhost.EnableItem(XmlIdLink(5)) </Action>
</Macro>
</Folder>
</EventGhost>
`

mattail, 2016-02-07:
Maybe I should add that the Pulse-eight adapter is connected to the NUC and Marantz AVR.

linkmk, 2016-02-25:
Hi, is because eventghost is already using COM port where pulse-eight adapter connects. On my setup happens too, now i will try to disable cec plugin when PLEX starts but no success yet. i hope share my findings here

mattail, 2016-02-25:
Hi linkmk
Thanks for your comment, you're the first to reply.
What you describe is also my conclusion. 
Let me know if you find anything.

opdenkamp, 2016-02-25:
Only one application can connect to the adapter indeed, so if EventGhost is using the adapter, then Plex won't be able to use it.

I'm not very familiar with what Eventghost can and can't do really, and if it can close the CEC plugin when opening another application, but that's what it needs to do if you want Plex to use it.

The other option is to create an Eventghost configuration that forwards the events it receives as standard key presses to Plex

mattail, 2016-02-25:
Yes, if eventghost could turn off the CEC plugin when plex is lauched and re-enable it when plex is closed then it could work.

Slion, 2016-08-25:
@mattail 
Not sure exactly what you are trying to do but it sounds like the poorly named "Sharp Display Manager" tool could possibly help you. It could notably disconnect CEC when launching Plex. It could reconnect CEC at some point but it won't be able to tell when Plex is closed.

See a recent screenshot there:
http://forum.team-mediaportal.com/threads/minidisplay-for-sharp-display-manager.127180/page-4#post-1194258

Head there and follow the download link:
http://forum.team-mediaportal.com/threads/minidisplay-for-sharp-display-manager.127180/

Note:
I never tried EventGhost myself.
I'm the developer of Sharp Display Manager.

opdenkamp, 2016-10-25:
This is something that will have to be addressed by EventGhost. Please ask on their forum for help, as we can't solve this in the library without massive changes that would require another Plex implementation too (and any other client)

closed 2016-10-25 by opdenkamp