Pulse-Eight/libcec #195: set volume to specific value

knallio, 2016-02-09:
I am trying to set my AVR to a specific volume with libcec using C code on a raspberry pi. At the moment I can get the current volume and now I am trying to increase/decrease the volume until the specified volume is reached. But I seem to have timing problems when sending volume changes repeatedly. Either I need a large delay (1 second) between each volume change or the AVR misses some of the changes.

Is there an optimal way to either change the volume repeatedly in a robust and fast way or another way to achieve the original aim of setting a specific volume?

opdenkamp, 2016-02-25:
unfortunately this is not supported by the current version of the CEC specification. Only volume up, down and mute are in there, and there's no uniform way in which volume up and down are interpreted. The change in level is different for most brands and some brands use incremental values when repeating volume commands.

If you want to change the volume quickly, pass `false` as parameter to `VolumeUp()` until the last key that you're sending.

closed 2016-02-25 by opdenkamp

knallio, 2016-02-25:
Thanks for your reply. I already assumed that it is not in the specification, that is why I try to send multiple volumeup commands. What works as expected is the following code snippet: 
```
for (int i=0;i<10;i++)
{
                libcec_volume_up(g_cec_iface.connection,1);
                sleep(2);
}
```
With this code my AVR gets ten volume up commands and raises the volume accordingly. But that is too slow for me and if I remove the sleep (or shorten the time to about 0.9 s)  some of the commands are lost.

I tried your last suggestion and changed the code to the following:
```
for (int i=0;i<9;i++)
{
                libcec_volume_up(g_cec_iface.connection,0);
 }
 libcec_volume_up(g_cec_iface.connection,1);
```
But again the AVR did not raise the volume ten times as expected. Is there anything else I can try?