Pulse-Eight/libcec #197: Build on Raspberry Pi 2

GeoHolz, 2016-02-11:
Hello,

I try to install it on my raspberry : 
```
pi@screenly ~/libcec/build $ cat /etc/debian_version
7.8
pi@screenly ~/libcec/build $ uname -a
Linux screenly 3.18.7-v7+ #755 SMP PREEMPT Thu Feb 12 17:20:48 GMT 2015 armv7l GNU/Linux
```
I get error when i compiled libcec : 
```

pi@screenly ~/libcec/build $ make -j4
Scanning dependencies of target cec
[  2%] [  2%] [  2%] [  2%] Building CXX object src/libcec/CMakeFiles/cec.dir/CECClient.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/CECProcessor.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/LibCEC.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/LibCECC.cpp.o
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:35:0:
/home/pi/libcec/src/libcec/CECClient.h:49:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
In file included from /home/pi/libcec/src/libcec/LibCEC.cpp:35:0:
/home/pi/libcec/src/libcec/LibCEC.h:49:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
In file included from /home/pi/libcec/src/libcec/LibCEC.cpp:35:0:
/home/pi/libcec/src/libcec/LibCEC.h:134:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/LibCEC.h:135:19: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.h:135:31: error: template argument 1 is invalid
/home/pi/libcec/src/libcec/LibCEC.h:135:31: error: template argument 2 is invalid
/home/pi/libcec/src/libcec/LibCEC.h:153:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/LibCEC.h:154:19: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.h:154:31: error: template argument 1 is invalid
/home/pi/libcec/src/libcec/LibCEC.h:154:31: error: template argument 2 is invalid
In file included from /home/pi/libcec/src/libcec/CECProcessor.cpp:35:0:
/home/pi/libcec/src/libcec/CECProcessor.h:59:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:63:5: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:71:58: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:76:5: error: ‘CECClientPtr’ does not name a type
In file included from /home/pi/libcec/src/libcec/CECProcessor.cpp:35:0:
/home/pi/libcec/src/libcec/CECProcessor.h:90:27: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:92:29: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:95:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:96:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:126:29: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:163:37: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:191:37: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/CECProcessor.h:191:49: error: template argument 2 is invalid
/home/pi/libcec/src/libcec/CECProcessor.h:191:49: error: template argument 4 is invalid
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:37:0:
/home/pi/libcec/src/libcec/CECProcessor.h:59:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:63:5: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:71:58: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:76:5: error: ‘CECClientPtr’ does not name a type
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:37:0:
/home/pi/libcec/src/libcec/CECProcessor.h:90:27: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:92:29: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:95:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:96:7: error: ‘CECClientPtr’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:126:29: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:163:37: error: ‘CECClientPtr’ has not been declared
/home/pi/libcec/src/libcec/CECProcessor.h:191:37: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/CECProcessor.h:191:49: error: template argument 2 is invalid
/home/pi/libcec/src/libcec/CECProcessor.h:191:49: error: template argument 4 is invalid
In file included from /home/pi/libcec/src/libcec/CECProcessor.cpp:38:0:
/home/pi/libcec/src/libcec/devices/CECBusDevice.h:52:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
In file included from /home/pi/libcec/src/libcec/CECProcessor.cpp:38:0:
/home/pi/libcec/src/libcec/devices/CECBusDevice.h:193:5: error: ‘CECClientPtr’ does not name a type
In file included from /home/pi/libcec/src/libcec/CECClient.cpp:38:0:
/home/pi/libcec/src/libcec/LibCEC.h:49:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
In file included from /home/pi/libcec/src/libcec/LibCEC.cpp:39:0:
/home/pi/libcec/src/libcec/CECProcessor.h:59:11: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
/home/pi/libcec/src/libcec/CECProcessor.h:63:5: error: ‘CECClientPtr’ does not name a type
[... 58 lines cut ...]
/home/pi/libcec/src/libcec/LibCEC.cpp:67:13: error: request for member ‘clear’ in ‘((CEC::CLibCEC*)this)->CEC::CLibCEC::m_clients’, which is of non-class type ‘int’
/home/pi/libcec/src/libcec/LibCEC.cpp:73:3: error: ‘m_client’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.cpp: In member function ‘virtual bool CEC::CLibCEC::Open(const char*, uint32_t)’:
/home/pi/libcec/src/libcec/LibCEC.cpp:89:20: error: ‘CECClientPtr’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.cpp:89:32: error: template argument 1 is invalid
/home/pi/libcec/src/libcec/LibCEC.cpp:89:32: error: template argument 2 is invalid
/home/pi/libcec/src/libcec/LibCEC.cpp:89:44: error: invalid type in declaration before ‘it’
/home/pi/libcec/src/libcec/LibCEC.cpp:89:44: error: expected ‘;’ before ‘it’
/home/pi/libcec/src/libcec/LibCEC.cpp:89:44: error: ‘it’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.cpp:89:59: error: request for member ‘begin’ in ‘((CEC::CLibCEC*)this)->CEC::CLibCEC::m_clients’, which is of non-class type ‘int’
/home/pi/libcec/src/libcec/LibCEC.cpp:89:84: error: request for member ‘end’ in ‘((CEC::CLibCEC*)this)->CEC::CLibCEC::m_clients’, which is of non-class type ‘int’
/home/pi/libcec/src/libcec/LibCEC.cpp:89:89: error: expected ‘)’ before ‘;’ token
/home/pi/libcec/src/libcec/LibCEC.cpp:89:35: warning: unused variable ‘iterator’ [-Wunused-variable]
/home/pi/libcec/src/libcec/LibCEC.cpp:89:91: error: ‘it’ was not declared in this scope
/home/pi/libcec/src/libcec/LibCEC.cpp:89:95: error: expected ‘;’ before ‘)’ token
/home/pi/libcec/src/libcec/LibCEC.cpp:623:1: error: expected ‘}’ at end of input
/home/pi/libcec/src/libcec/LibCEC.cpp:623:1: warning: control reaches end of non-void function [-Wreturn-type]
make[2]: *** [src/libcec/CMakeFiles/cec.dir/CECProcessor.cpp.o] Erreur 1
make[2]: *** Attente des tâches non terminées....
make[2]: *** [src/libcec/CMakeFiles/cec.dir/LibCEC.cpp.o] Erreur 1
make[2]: *** [src/libcec/CMakeFiles/cec.dir/CECClient.cpp.o] Erreur 1
make[1]: *** [src/libcec/CMakeFiles/cec.dir/all] Erreur 2
make: *** [all] Erreur 2

```

GeoHolz, 2016-02-11:
I found, i needed gcc and g++ 4.8

mrgreywater, 2016-02-22:
Would you mind closing this issue?

closed 2016-02-23 by GeoHolz