Pulse-Eight/libcec #202: Possible to switch HDMI input using CEC?

sofakng, 2016-03-07:
Is it possible to use HDMI-CEC (i.e. libcec) to switch inputs on the TV?

For example, I am hoping to be able to send a simple command like "input 1" to have the TV switch to HDMI input #1.

I've done some research and it looks like this is only possible through vendor specific commands?  If so, is there a list for Vizio and/or Panasonic?

RedsGT, 2016-05-11:
Hi sofakng,

I use raw 15:44:69 to toggle through the AV sources on my Denon receiver.  I also just tried 15:44:69:01, 15:44:69:02, etc and that lets me pick a specific AV(HDMI) input.

According to http://www.cec-o-matic.com, 10:44:69 should work with TV's.

Let us know how it goes.

santi6291, 2016-05-21:
@sofakng try `echo "tx 4F:82:30:00" | cec-client -s`
for more details copy `4F:82:30:00` on http://www.cec-o-matic.com/

work when changing between HDMI source I have no idea how to change it between non-HDMI sources

Slion, 2016-07-18:
The other I was sending key presses to my Android TV Philips 55PUK7150/12 and I'm pretty sure some of those keys controlled the source input. Maybe that would work for you too.

opdenkamp, 2016-10-25:
Unfortunately we can't add this to the library, because we aim to be compatible with the HDMI-CEC specification, which doesn't allow this.

So you'll have to use @santi6291's way indeed.

closed 2016-10-25 by opdenkamp

nikhilbhardwaj, 2016-12-04:
@santi6291 Does this require the device that is plugged into the other HDMI ports to be CEC compatible as well?

`echo "tx 4F:82:30:00"` takes me to `HDMI3` on my TV which has a Chromecast plugged in.
`echo "tx 4F:82:20:00"` takes me to `HDMI2` for an instant before Segfaulting and taking me back to the source that I was on earlier. I see similar behavior for `echo "tx 4F:82:40:00"` where my Xbox one is plugged in.

santi6291, 2016-12-04:
@nikhilbhardwaj the other device don't have to be to change hdmi, thats could be a built in funtion of your tv I recommend leaving `cec-client` running and changing inputs to see whats happening.

PAV1THRA, 2017-02-03:
Hi 
I am using the above commands for switching the HDMI source. Is there any command to switch ON the TV with specifying a particular HDMI port say HDMI 2. By default when I use 'as' command to switch ON, the TV is always set to HDMI 1.

santi6291, 2017-02-22:
@PAV1THRA you can use 
turn on: `echo 'on 0' | cec-client -s` and 
turn off: `echo 'standby 0' | cec-client -s`

julespaget, 2017-07-12:
Hi everyone, i would like to switch source (HMDI 1 with TV for example).
I have a TV Toshiba regza using CEC and a Raspberry Pi 3. I thought I can set the TV on an active source,  with this command : tx 10 82 but it doesn't work....
I look on here : http://www.cec-o-matic.com/ but I found nothing

PS : Sorry for my bad english :)

opdenkamp, 2017-07-13:
The CEC command take a physical address as parameter.
e.g. physical address 3.0.0.0 (HDMI 3 of your TV most likely) goes like this `4F:82:30:00`

julespaget, 2017-07-13:
This is my config :
device #0: TV
address:		0.0.0.0
active source:	no
vendor:			Toshiba
osd string:		TV
CEC version: 	1.3a
power status:	on
language: 		fre

device #1: 		Recorder 1
address:		1.0.0.0
active source:	no
vendor:			Toshiba
osd string:		CECTester
CEC version: 	1.4
power status:	on
language: 		eng

and when i try this :  tx 1f:82:00:00
TRAFFIC: [                     14555] 		<< 1f:82:00:00
waiting for input

why waiting for input ??? after that, the config is the same

opdenkamp, 2017-07-13:
This is why the command isn't included in libCEC and why you need to use raw TX commands to do this.

What the command actually does is telling other devices that it became the active source. So usuaully a source device, broadcasting a message saying "I am active". Because of how CEC is structured, the switches between your screen and source will do the switching, even if the message actually came from another device (it can't easily tell the difference). That's why this will work in most situations when trying to make another source active, but won't work when trying to switch the TV to it's internal input. It will just think "nope, I didn't send this message" and ignore it.

julespaget, 2017-07-13:
So how can i do to pass the tv which is in hdmi 1 in tv without this command?

opdenkamp, 2017-07-13:
I'm not sure what you're asking. Do you want to have the TV switch to it's internal input, so not to a device like a bluray player that's connected?
If so, then you can try the <inactive source> command, but if that doesn't work, you're out of luck. This is simply not included in the current version of the HDMI-CEC specification.

julespaget, 2017-07-13:
Yes i want to switch tv it's internal input...

opdenkamp, 2017-07-13:
Then read the next sentence of my previous comment please (just added the missing command from that sentence) If `inactive source` doesn't work, then it can't be done

julespaget, 2017-07-13:
do you mean this :

```
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.2
no serial port given. trying autodetect:
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [             167]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             167]     Open - vc_cec initialised
DEBUG:   [             167]     logical address changed to Free use (e)
NOTICE:  [             167]     connection opened
DEBUG:   [             167]     processor thread started
DEBUG:   [             167]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [             167]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             167]     << e0
DEBUG:   [             228]     >> POLL sent
DEBUG:   [             228]     TV (0): device status changed into 'present'
DEBUG:   [             228]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [             228]     << e0:8c
TRAFFIC: [             445]     >> 0f:87:00:00:39
DEBUG:   [             445]     TV (0): vendor = Toshiba (000039)
DEBUG:   [             445]     expected response received (87: device vendor id)
DEBUG:   [             445]     replacing the command handler for device 'TV' (0)
NOTICE:  [             445]     registering new CEC client - v4.0.2
DEBUG:   [             445]     detecting logical address for type 'recording device'
DEBUG:   [             445]     trying logical address 'Recorder 1'
DEBUG:   [             445]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             445]     << 11
DEBUG:   [             445]     >> TV (0) -> Broadcast (F): device vendor id (87)
TRAFFIC: [             865]     << 11
DEBUG:   [            1140]     >> POLL not sent
DEBUG:   [            1140]     using logical address 'Recorder 1'
DEBUG:   [            1140]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            1140]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            1140]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            1140]     Recorder 1 (1): CEC version 1.4
DEBUG:   [            1140]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [            1140]     logical address changed to Recorder 1 (1)
DEBUG:   [            1140]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            1140]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            1141]     GetPhysicalAddress - physical address = 1000
DEBUG:   [            1141]     AutodetectPhysicalAddress - autodetected physical address '1000'
DEBUG:   [            1141]     Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [            1141]     << Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: [            1141]     << 1f:84:10:00:01
NOTICE:  [            1295]     CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, git revision: 8.0.3+92-22c1002~dirty, compiled on Mon May 29 08:28:38 UTC 2017 by stephan@buildserver.openelec.tv on Linux 3.19.8-100.fc20.x86_64 (x86_64), features: P8_USB, DRM, P8_detect, RPi
DEBUG:   [            1295]     Recorder 1 (1): vendor = Toshiba (000039)
DEBUG:   [            1295]     replacing the command handler for device 'Recorder 1' (1)
DEBUG:   [            1295]     << Recorder 1 (1) -> Broadcast (F): vendor id Toshiba (39)
TRAFFIC: [            1295]     << 1f:87:00:00:39
DEBUG:   [            1447]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [            1447]     << 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            1782]     << requesting power status of 'TV' (0)
TRAFFIC: [            1782]     >> 01:8c
TRAFFIC: [            1782]     << 10:8f
DEBUG:   [            1782]     >> TV (0) -> Recorder 1 (1): give device vendor id (8C)
DEBUG:   [            1842]     << Recorder 1 (1) -> TV (0): vendor id Toshiba (39)
TRAFFIC: [            1842]     << 1f:87:00:00:39
WARNING: [            2143]     unhandled response received: opcode=87 initiator=1 destination=f response=0
WARNING: [            2206]     unhandled response received: opcode=4 initiator=1 destination=0 response=0
WARNING: [            2480]     unhandled response received: opcode=82 initiator=1 destination=f response=0
TRAFFIC: [            2481]     >> 01:a0:00:00:39:02
DEBUG:   [            2481]     sending abort with opcode a0 and reason 'invalid operand' to TV
DEBUG:   [            2481]     << transmitting abort message
TRAFFIC: [            2481]     << 10:00:a0:03
DEBUG:   [            2481]     >> TV (0) -> Recorder 1 (1): vendor command with id (A0)
WARNING: [            2571]     unhandled response received: opcode=8e initiator=1 destination=0 response=0
[... 197 lines cut ...]
TRAFFIC: [           59340]     << 1f:85
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Toshiba
osd string:    TV
CEC version:   1.3a
power status:  on
language:      fre

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Toshiba
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
DEBUG:   [           60402]     expected response not received (82: active source)
```
when recorder is active, and put in inactive source with "is" ?

opdenkamp, 2017-07-13:
Correct. If this doesn't make the TV switch back to it's internal input (or the previously selected input), then it's not going to work

julespaget, 2017-07-13:
.... Bad news...

opdenkamp, 2017-07-13:
That's unfortunate, but it's simply not supported by the current version of the CEC specification, which only allows devices to mark themselves as active and switches making connected device to them active.

GreihMurray, 2018-06-12:
The best way I have found is to type the code echo tx 4F:82:10:00 | cec-client -s -d 1 directly into the command prompt. If your pi is connected to a port other than HDMI 1 you will need to change the 1 in 10 accordingly

andyjjones28, 2019-11-19:
> That's unfortunate, but it's simply not supported by the current version of the CEC specification, which only allows devices to mark themselves as active and switches making connected device to them active.

Hi opdenkamp, if I understand your post correctly, does this mean that it's _up to CEC device_ to respond to 1f:85 (discover active source) messages with opcode 82?  If they don't, cec-client will respond to scans with:

`currently active source: unknown (-1)`

Is my interpretation correct?

cakechris, 2019-11-21:
Hello @opdenkamp, I too am interested in the question asked by @andyjjones28 .

Thanks,
Chris

surfzoid, 2022-03-13:
hi 
i got the same problem from my philips Tv
here it is my solution
i have an PI P400 connected to hdmi2
cec-client -S
return 0.0.0.0 my tv and 1.0.0.0 freebox and 2.0.0.0 P400 but don't see my computer connected to 5.0.0.0
so, the magic command is
cec-client -m -d 8
then use the physical buton of my tv to switch directly to the good hdmi and then press ctr+c before come back to P400 hdmi
i first try but without success
echo "tx 0F:86:50:00" | cec-client -s -d 2
but the two following command work
cec-ctl --tv #need to init and config each time
cec-ctl --to 0 --active-source phys-addr=5.0.0.0

inf0rmix, 2024-04-05:
cec-ctl --to 0 --active-source phys-addr=X.0.0.0

works for me too