Pulse-Eight/libcec #204: CEC stops working on Cubox-i4x4 after a couple of days on Openelec 6.0.3

phunkyfish, 2016-03-11:
Hi,

Prior to this version CEC would stop working after a couple of hours. With the latest version it lasts much longer maybe a day or two and then stops working.

Happy to provide any logs etc. required if you can tell me what to send.

phunkyfish, 2016-03-11:
I have it connected to an LG 49UB850V with soundbar. FYI I have a small PC with an external Pulse Eight adapter which works perfectly.

opdenkamp, 2016-10-26:
I'm going to need a log to see if there are any clues in there, but it's probably something that @warped-rudi will need to look at as the author of that code. I don't have the hardware afaik.

warped-rudi, 2016-10-26:
Actually, I only did the code for the first generation Cubox (TDA995X). The Cubox-i (IMX6) stuff was authored by @wolfgar, but seems to re-use some of my code. I don't know why he didn't upstream it. I remember having asked him to do so... However, he stopped working on that and virtually disappeared. When I saw the latest commits here adding support for the AML SoCs, I was actually thinking to prepare a pull request for @opdenkamp, so that we finally get an "official" home for that code. Currently, the situation is bad, as there are different versions of the library code as well as the kernel driver floating around. I cannot even tell which version OpenELEC is using!

wolfgar, 2016-10-26:
Hi @warped-rudi,  
You assume right : your TDA995x support in libcec was clearly a source of inspiration when I ported iMX6 and I likely share some fragments of code...  
As you explain, the current state is too convoluted : as there is no official driver nor even stable interface with userspace.
If you remember I really tried  in several occasion to progress (firstly openly shared my work, then tried to avoid entropy commenting -and being ignored- another modified libcec/driver for imx6, then contacted jon with a new driver and some comments/questions about the interface between kernel/userspace  ...).
So I really wanted to upstream (I contacted you @opdenkamp if you remember both to know about the license to sign and also with a few questions about libcec) but this fragmentation (and in some occasion hostile behavior -not from the mentioned people here-) and a big change in my life (I launched a startup and I try to design a product) killed this initial motivation...  
Sorry for this, my current free time is so sparse I can't tell I am able to revive this at this time but I have not totally disappeared and if I can give an hand to have constructive work, you can ping me...

opdenkamp, 2016-10-26:
Hi guys,

The license is about licensing the code to others. You can find the details on this page http://www.pulse-eight.net/contributors

If there's no stable driver or interface, then it's indeed not going to work properly any time soon.

Shall I close this ticket?

Good luck with the new company @wolfgar :)

warped-rudi, 2016-10-27:
@opdenkamp : I think you should close this ticket. It's rather old and there's probably not much you can do about it. I might have a look at the code on weekend and talk to @wolfgar on how to proceed. About the kernel interface I would not worry too much: I'm quite sure that once a version of libCEC lands here, Jon will include the interface it requires in SolidRun's kernel. At least if it isn't completely awkward...

And @wolfgar :+1:

opdenkamp, 2016-10-27:
Ok, I'll close this one. Thanks for looking into it.

closed 2016-10-27 by opdenkamp