Pulse-Eight/libcec #205: Kodi on RPi2 constantly sending CEC signals

kartana, 2016-03-12:
Here is a log where I have two problems:

1. Sometimes AVR powers off instead of being selected
2. The RPi2/Kodi goes nuts when I turn off the TV+AVR. it constantly sends some signals that turn on the AVR, turn off the AVR and the Standby LED from my TV goes crazy. There is some fighting going on over something. 

Setup: Philips TV connected to Sony AVR that has Chromecast as HDMI1 and RPi2 running latest stable OpenELEC. CEC is enabled in Kodi with everything on OFF or IGNORE (like auto power on/off).

http://pastebin.com/di8vB7sQ

Problem started a week ago or so. I had the OSMC Jarvis build running when it came out without a problem and suddenly problem 2 it started happening (I had problem 1 before). It's the same with the latest OpenELEC now.

kartana, 2016-03-22:
Has no one a RPi2 with OSMC or OpenELEC and a Chromecast Gen1 and can test this? For some reason those two conflict now with CEC. This is driving me crazy!

enricostano, 2016-04-03:
Yep, similar setup, behavior and timing here. Using ArchLinux ARM on Rpi1. No media player installed I just use `omxplayer -o hdmi` from ssh.

Some debugging:
https://gist.github.com/enricostano/0c34a7ff7d1c4c9e378beefebc8d0ebf
https://gist.github.com/enricostano/f96aaa605a4384625c091e2039cc2a46

In my case I cannot switch to another TV channel or HDMI input (Chromecast, or whatever), when I try to switch channel I get `HDMI cable is unplugged` in the `$ /opt/vc/bin/tvservice -M` output and then the TV switches back to the Rpi outputting `HDMI is attached` (please look at the [first debug link](https://gist.github.com/enricostano/0c34a7ff7d1c4c9e378beefebc8d0ebf)).

Removing the other CEC enabled device from the TV HDMI inputs the issue disappears.

OhmEye, 2016-04-13:
I have this same symptom, if a chromecast is plugged into my amp, Kodi sends cec events every few seconds. My amp LCD shows the Kodi source being activated like clockwork several times a minute. If I turn off the TV, Kodi turns it back on almost immediately with the next event. If I unplug the chromecast, the events stop, the amp stops showing them on the LCD, and the TV stays off. I've been trying every possible combination of cec settings in openelec and the amp to no avail, everything works as expected except when chromecast and the rpi2 with openelec are both plugged in.

I've also been using this setup for a couple years with no issues until just recently.

kartana, 2016-05-26:
Seems fixed since the last Chromecast update. Can anyone confirm?

Slion, 2016-07-18:
Logs show 'routing change' messages repeatedly.
You could try to fix suggested in issue https://github.com/Pulse-Eight/libcec/issues/233

opdenkamp, 2016-10-25:
forgot to commit this last week, this prevents it from sending out active source messages all the time https://github.com/Pulse-Eight/libcec/commit/5926317200ff78a1c03269c377c25f77f0a0f4f3

closed 2016-10-25 by opdenkamp

enricostano, 2016-10-25:
Thanks @opdenkamp ! Will test and report back ASAP

kartana, 2016-10-28:
The issue described by me on Mar 12 is back in full force. As soon as I turn off the TV, Kodi on the RPi3 tries to grab the HDMI port from all the other to my AVR connected devices turning the AVR and TV on or at least tries to, turning the standby LED from TV to a disco light. Probably the TV doesn't know how to process the signals at this point.

opdenkamp, 2016-10-28:
That can't be related to this commit and sounds like a bug in Kodi. Please post a log.

kartana, 2016-10-28:
> That can't be related to this commit and sounds like a bug in Kodi. Please post a log.

My SO just told me that this problem returned last week so you are probably right. I'll post a log if I find the time this weekend.