Pulse-Eight/libcec #210: Cannot import cec in python

wlingke, 2016-03-29:
Hi, I followed the instructions to install libcec onto a clean raspberry pi 3. The installation was successful, and I am able to send commands to my television using the command line. However, I am unable to get the python bindings to work.

When I run the example file: https://github.com/Pulse-Eight/libcec/blob/master/src/pyCecClient/pyCecClient.py

I just get an error saying that "no module named 'cec'". Do I need to install some other package?

Thanks

PAV1THRA, 2016-06-15:
Hi wlingke
Even I am facing the same issue. Please let me know if u got any solution for the same . Thanks

prussiap, 2016-07-13:
I have the same issue with pi 3.. following the instructions. can use the cec-client but cant import cec from ipython.

opdenkamp, 2016-08-02:
must be some local problem as it's working fine when I install it from the .deb: `sudo apt-get get install python-libcec`

if it's not available on your distribution, ask the maintainer to package it.

the most likely cause here is that you need to set python's search path to find the _parent_ directory of the `cec` directory that contains the python module.

closed 2016-08-02 by opdenkamp