Pulse-Eight/libcec #217: Not detecting the USB port and the logical address

appup93, 2016-05-06:
Hi 
When I run the application using VS 2010 the auto detection of the usb port is failing. I got the application to work by hardcoding the usb port value. But the application is not detecting the address of the tv so as a result I am not able to send commands to the tv like "on" and putting it on "standby".
Please provide me a solution

malard, 2016-05-06:
what happens when you run the compiled version from our website? also vs2010 is not supported

appup93, 2016-05-06:
I ran it using VS 2012 . The number of devices found, I am getting it as zero.
  uint8_t iDevicesFound = g_parser->FindAdapters(devices, 10, NULL);
  if (iDevicesFound <= 0)
The iDevicesFound  is 0 when I run it, so as a result its exiting.
I set the cec-config file as the startup project.
By the version downloaded from your website do you mean the Cec-tray.exe?

opdenkamp, 2016-05-09:
solved per email support

closed 2016-05-09 by opdenkamp