Pulse-Eight/libcec #219: Compilation Error Windows( VS 2013)

appup93, 2016-05-09:
Does anyone know why this error is occuring? I have been stuck with this issue for a couple of days.
I am trying to debug the project in vs 2013 but i am getting this error while trying ti run the visual-studio.cmd file

CMake Error at src/cec-client/CMakeLists.txt:21 (find_package):
  By not providing "Findplatform.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "platform",
  but CMake did not find one.

  Could not find a package configuration file provided by "platform" with any
  of the following names:

    platformConfig.cmake
    platform-config.cmake

  Add the installation prefix of "platform" to CMAKE_PREFIX_PATH or set
  "platform_DIR" to a directory containing one of the above files.  If
  "platform" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

opdenkamp, 2016-05-09:
try cleaning your build dir (git clean -xfd) now that visual studio's cl.exe can be found and try to start from scratch.

opdenkamp, 2016-05-09:
problem found per email. the script requires a 64 bits Windows and Visual Studio or it will fail.

closed 2016-05-09 by opdenkamp