Pulse-Eight/libcec #225: autodetect FAILED

reydelleon, 2016-06-01:
Hi there,

After following the instruction in the README to install the library in my Pi 2 Model B, I keep getting the following error, regardless of the command:

`autodetect FAILED`

Here is what the output was when building:

```
-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   yes
-- TDA995x support:                        no
-- Exynos support:                         no
-- DRM support:                            yes
-- Python support:                         version 3.4.2 (3.4)
-- lib info: git revision: libcec-3.1.0+5-6d68d21, compiled on Wed  1 Jun 02:40:51 UTC 2016 by pi@EDWIN-ENTERTAINMENT on Linux 4.4.9-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/libcec/build
```

Everything seems fine there. It says that it supports the Pi and everything, but in the end it fails. What can this be?

opdenkamp, 2016-08-02:
in the end _what_ fails. can you show me the full output that leads to the problem

opdenkamp, 2016-10-25:
No response. The README files have been updated. Please check again after using the new instructions

closed 2016-10-25 by opdenkamp