Pulse-Eight/libcec #230: SourceActivatedCallback with a VIZIO D58u-D3

aanderse, 2016-07-03:
Hello,

I do not believe that the "SourceActivatedCallback" in libcec is working properly for my television. I own a VIZIO D58u-D3 television.

I slightly modified the python example program that came with libcec to print out whenever a source was activated but nothing is ever printed out when I turn the television on.

I did notice that the television is notifying my pulse-eight cec adaptor because the "LogCallback" lets me know when the television is turned on:
NOTICE:  [31052]     << powering on 'TV' (0)

Here are the details of my system:
aaron@moya ~ $ uname -a
Linux moya 4.4.0-28-generic #47-Ubuntu SMP Fri Jun 24 10:09:13 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

aaron@moya ~ $ apt list --installed | grep libcec
libcec-dev/xenial,now 3.0.3~xenial amd64 [installed]
libcec3/xenial,now 3.0.3~xenial amd64 [installed,automatic]
python-libcec/xenial,now 3.0.3~xenial amd64 [installed]

Any advice/help on this?

Thank you,
Aaron

Slion, 2016-07-17:
Not sure why you are expecting a source activated notification when powering on your TV.

opdenkamp, 2016-10-25:
Powering on the TV won't generate that callback indeed. It only gets generated when you switch to the input that has the PC connected.

closed 2016-10-25 by opdenkamp