Pulse-Eight/libcec #233: Android TV turns back on after standby

Slion, 2016-07-17:
Hardware:
- Philips TV 55PUK7150/12 (2015)
- Windows 10 PC
- libcec 3.1+ custom build

I have a this utility I've been using to set my TV on standby when Windows gives me "monitor off" notification and back on when the monitor is turned back on.

It used to be quite reliable on a Philips TV 55PFL7606 (2011) however with that Android TV it is working only every second attempt. Looking at the logs off failed attempts  the TV is sending a routing change command which results in my PC (Recorder) source being turned on which somehow turn the TV back on. Why does the Android TV broadcast a routing change command every second attempt is quite puzzling.

```
NOTICE:  1828630 << putting 'TV' (0) in standby mode
NOTICE:  1828695 >> source deactivated: Recorder 1 (1)
CEC source activated: Recorder 1/False
NOTICE:  1828695 << Recorder 1 (1) -> broadcast (F): inactive source
CEC command Src:TV Dst:Broadcast Ack: True Eom: True OpcodeSet: True Opcode: routing change Timeout: 1000
NOTICE:  1829641 >> source activated: Recorder 1 (1)
CEC source activated: Recorder 1/True
NOTICE:  1829641 << powering on 'TV' (0)
NOTICE:  1829699 << Recorder 1 (1) -> broadcast (F): active source (1000)
CEC command Src:TV Dst:Broadcast Ack: True Eom: True OpcodeSet: True Opcode: request active source Timeout: 1000
NOTICE:  1829875 << Recorder 1 (1) -> broadcast (F): active source (1000)
CEC command Src:TV Dst:Broadcast Ack: True Eom: True OpcodeSet: True Opcode: report physical address Timeout: 1000
CEC command Src:TV Dst:Broadcast Ack: True Eom: True OpcodeSet: True Opcode: device vendor id Timeout: 1000
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give physical address Timeout: 1000
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give osd name Timeout: 1000
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give device vendor id Timeout: 1000
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give physical address Timeout: 1000
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give device power status Timeout: 1000
CEC command Src:TV Dst:Broadcast Ack: True Eom: True OpcodeSet: True Opcode: set stream path Timeout: 1000
NOTICE:  1833680 >> source activated: Recorder 1 (1)
CEC source activated: Recorder 1/True
NOTICE:  1833680 << Recorder 1 (1) -> broadcast (F): active source (1000)
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give physical address Timeout: 1000
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give osd name Timeout: 1000
CEC command Src:TV Dst:Recorder 1 Ack: False Eom: True OpcodeSet: True Opcode: give device vendor id Timeout: 1000
```

Slion, 2016-07-17:
After enabling all logs and looking at the libcec code it seems like libcec ends up turning the TV back on in response to that 'routing change'.

My logs look like that then :
(was now reporting as playback device instead of recording as I was trying things out to solve that issue) 
```
CEC command Src:TV Dst:Broadcast Ack: True Eom: True OpcodeSet: True Opcode: routing change Timeout: 1000
DEBUG:  94639 Playback 1 (4) was already marked as active source
NOTICE:  94639 >> source activated: Playback 1 (4)
CEC source activated: Playback 1/True
DEBUG:  94639 sending active source message for 'Playback 1'
NOTICE:  94639 << powering on 'TV' (0)
DEBUG:  94639 >> TV (0) -> Broadcast (F): routing change (80)
NOTICE:  94701 << Playback 1 (4) -> broadcast (F): active source (1000)
DEBUG:  94810 << Playback 1 (4) -> TV (0): menu state 'activated'
```
Activating our adapter source runs the following function that's turning the TV back on:
`CCECCommandHandler::ActivateSource`

I could certainly workaround that issue but I'm not sure what's the proper fix.
- Should we not activate a source that's already activated?
- Is the Android TV to blame for sending that 'route change' command?
- Is libcec response to that 'route change' wrong?
- Should activating a source turn on the TV at all?

Slion, 2016-07-17:
According to the specs 'routing change' message should only be used when a CEC switch is manually switched. Therefore it's fair to say that Android TV should not send that when it's being turned off. 

However it probably does not make sense for libcec to perform a source activation when receiving 'routing change'.

It would be nice to see a fix for this issue merged to master.
I've modified the code as follow to fix my issue but it's possibly not the way it should be fixed on master.

```
int CCECCommandHandler::HandleRoutingChange(const cec_command &command)
{
  if (command.parameters.size == 4)
  {
    CCECBusDevice *device = GetDevice(command.initiator);
    if (device)
    {
      uint16_t iOriginalAddress = ((uint16_t)command.parameters[0] << 8) | ((uint16_t)command.parameters[1]);
      uint16_t iNewAddress = ((uint16_t)command.parameters[2] << 8) | ((uint16_t)command.parameters[3]);
      LIB_CEC->AddLog(CEC_LOG_DEBUG, "Routing change - original address: %x - new address: %x", iOriginalAddress, iNewAddress);
      // SL: We are not a switch so we should not do anything here
      // Set active route simply activate source if our USB CEC adapter matches the new address.
      // Some Android TV 5.1 devices are sending a 'route change' message after the TV was set to standby. 
      // That resulted in the TV being turn back on instantly as the activate source process rightly turns the TV on.
      // Philips 55PUK7150/12 is known for exhibiting such behaviour.
      //device->SetActiveRoute(iNewAddress);

      return COMMAND_HANDLED;
    }
  }
```

opdenkamp, 2016-08-11:
I'll add a check for the source of the command and ignore it if it's not sent by the TV. Lots of TVs use routing changes to activate a source, even though it's not a switch, that's why libCEC is doing this.

Slion, 2016-08-11:
> I'll add a check for the source of the command and ignore it if it's not sent by the TV. 
I'm confused. In my case it is sent by the TV so what you are proposing would not help me.

opdenkamp, 2016-08-11:
Oh it's an android tv, was assuming that it was an android box

> However it probably does not make sense for libcec to perform a source activation when receiving 'routing change'.

This does make sense, because lots of older TVs send this out, if they're using 1.3a or an earlier version. TVs on newer versions *should* not do this, but it's not implemented in the way it should have been implemented on all TVs. Adding a version check would help in your case, but may break things for others.

opdenkamp, 2016-10-25:
https://github.com/Pulse-Eight/libcec/commit/5926317200ff78a1c03269c377c25f77f0a0f4f3 should work around your issue and still keep old stuff working. let me know if it doesn't fix it, then I'll reopen the issue

closed 2016-10-25 by opdenkamp

Slion, 2017-05-21:
@opdenkamp Thanks for those fixes.
I'm using my custom build so it could be a while until I get to test them.