Pulse-Eight/libcec #235: Listen to remote control when libcec isn't the active source [TV]

AndreaSpan, 2016-07-26:
Hi everyone! 

I'm using libcec (the python library) on a Raspberry Pi connected to a TV. 
CEC commands are working very nicely but I have some limits with the remote control listener. 

Actually, remote control listener are working very well as the Raspberry Pi is the active source but when the active source is another one, I'm not able to listen to the key pressed on the remote control. 

Is there any particular configuration to get this working? I would like my cec client to listen everytime to the key pressed on the remote control, is it possible ?

Thank you for your help! :)

Andrea

Slion, 2016-07-28:
I believe this should be possible though you may have to code it yourself.
It seems to me the adapter can get the CEC commands even when it's not the recipient.

opdenkamp, 2016-08-02:
This is not supported by the Raspberry Pi, which only allows you to listen to 1 address. Our own CEC adapters do support this: https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter

closed 2016-08-02 by opdenkamp