Pulse-Eight/libcec #237: CEC Partial loss Pi2B

scrag, 2016-08-01:
OSMC.

As of the February release of OSMC I have had issues with CEC being partially lost with a Panasonic Viera V10A and Panasonic SC-BT200 surround system.

Watching TV with the speakers set to the surround system I select the source to OSMC and the speakers revert back to the TV and I lose the surround system as the speaker source. The only way to get the surround speakers back is to reselect the surround speakers via VieraLink.

This only happens when the source is selected to OSMC (RPI). If I select the other inputs such as the Apple TV and Chromecast the surround system stays selected as the speaker source..

Sam Nazarko was kind enough to provide a patch to revert libCEC to the pre February release. This fixes the issue. Here is the link to that thread:
https://discourse.osmc.tv/t/2016-02-4-cec-partial-fail-on-input-change-to-osmc/15031

The change that brought about this "glitch" is quite annoying..... Once a source is selected the surround system should stay enabled as the speaker source. Sam's patch is still working but he has advised me that as of libCEC v17 it will most likely no longer work.

If this could be looked at it would be greatly appreciated.

Thanks in advance.

scrag, 2016-08-04:
This pull is what is apparently causing the issue:
https://github.com/Pulse-Eight/libcec/pull/208

opdenkamp, 2016-08-04:
That PR was never included in libCEC. If it was included in osmc's build of it, then it's Sam's own patching that caused the problem.

closed 2016-08-04 by opdenkamp