Pulse-Eight/libcec #239: AVR/TV turn back on after standby + Kodi steals source when another CEC device is powered on

spocky, 2016-08-04:
Since I've switched to LibreElec Krypton builds (was previously on Openelec 7 beta), whenever I power off my Tv Box (which is CEC enabled), it turns off my AVR and Tv, as it ever did, but a few seconds later, my AVR and Tv turn back on and switch the source to Kodi.
Also when my AVR+Tv+TvBox are off and I power on my Tv box, everything turns on as expected, and a few seconds later, the AVR switches the source to Kodi.
So it seems that almost whenever a change is happening on CEC, Kodi tries to get activated. Disabling CEC in Kodi fixes the issue.

This issue is very similar to #233 and I believe they're connected, but my Tv is not an Android Tv and I'm not seeing any "routing change" message in log, hence the new issue.

My system :
* Tv : Panasonic TX-P42ST60E
* AVR : Yamaha RX-V773
* Kodi : Cubox-i (also referenced below as "Playback2") with latest imx6 Krypton alpha3
* Tv box : Freebox Player (also referenced below as "Playback1")

Here are the details concerning my log file [here](http://pastebin.com/tvCL9W1j) :
* 16:12 : Kodi is booted with AVR and TV already on, and the source already set to kodi. Everything works as expected.
* 16:13 : I'm powering on my Tv box. Two seconds later, my AVR automatically switches source to "Tv box". CecLog shows that it detected that "Playback1" was taking control, and marks Playback2 as inactive. For some reason, less than 10 seconds later, it makes itself as the active source again, making my AVR switch source to Kodi.
* 16:14 : I use the "AV" button of my Tv Box, to force it to be active and take control over Kodi. The AVR switches back to "Tv box" source. This time, it works fine, as Kodi doesn't try to take control back.
* 16:15 : While still on "Tv Box" source, I use the "Power off" button of my Tv Box, which usually switches everything off. This works as expected... but only for a few seconds. Then Tv+AVR automatically turn on again and switch the source to Kodi. CecLog shows that after everything has been set to standby, less than 5 seconds later, "Playback2" is set as the active source, forcing everything to power on.

I've tested a build containing the #233 fix proposed by @Slion compiled by Misek [here](http://forum.solid-run.com/kodi-xbmc-player-on-cubox-i-and-hummingboard-f7/-devel-2016-05-06-libreelec-master-v8-0-kodi-17-0--t2964-s120.html) but this doesn't seem to fix the issue.

spocky, 2016-08-05:
For the sake of completeness, [here's](http://pastebin.com/EFALX7Ju) a log of my previous working system (OpenElec jarvis).

* 15:18 : Kodi is booted with AVR and TV already on, and the source already set to kodi. Everything works as expected.
* 15:19 : I'm powering on my Tv box. Two seconds later, my AVR automatically switches source to "Tv box". CecLog shows that it detected that "Playback1" was taking control, and marks Playback2 as inactive. 
* 15:20 : While still on "Tv Box" source, I use the "Power off" button of my Tv Box, which switches everything off, as expected.

One of the most visible difference between them is the reference to "OnDeviceChanged" method in LibreElec Krypton : and most particularly this line 

> OnDeviceChanged - state connected over disconnected, timer is already running

which seems to be the one forcing source acquisition. Now, seeing the logs, it looks like it's happening in the Kodi code rather than libcec. Now, is that function responding to a unlogged libcec event I'm not expert enough to tell.

Slion, 2016-08-09:
It could be that that libcec decides that receiving a change of physical address means it needs to set itself active. I've seen it doing similar funnies before as you mentioned above. Someone would need to take a look in the code.
Meanwhile, if possible, you could try setting your Kodi box as a recording device rather than a playback device and see if it helps. Somehow I have a feeling things like multiple device of the same type on an HDMI network have received very little testing.
I use libcec on a Windows 10 with MediaPortal so there is little I can test with your Kodi linux setup.

Slion, 2016-08-10:
Your problem at 16:13 is caused by the fact that Kodi gets `OnScreensaverDeactivated` notification in response to which it sets itself as active source.
You need to find out why Kodi receives this notification.

Slion, 2016-08-10:
Same issue at 16:15 you are getting `OnScreensaverDeactivated`. So the issue could be with the user (yes, I'm looking at you), Kodi or some Kodi libcec plug-in or your linux. It's unlikely to be an issue with libcec itself.

spocky, 2016-08-10:
Thanks @Slion for your messages. Concerning the OnScreensaverDeactivated call, I've suspected it, but seeing that :
* I've not changed any parameter related to screensaver in the settings (it's a fresh install)
* I have no external peripheral that could wake it up (no keyboard connected and wireless mouse is switched off)
so I just supposed it was a consequence and not a cause of this behavior.

Meanwhile, I crossed a commit that could probably fix that issue : https://github.com/mk01/xbmc/commit/2b22ad17b51a5a9883dd4c91a8b34688572c0614
I'll close this issue when I can test a new build including this commit (I have no system setup to compile it by myself).

vpeter4, 2016-08-10:
@spocky: Should I build one with this commit included?

spocky, 2016-08-10:
@vpeter4: that would be great!

vpeter4, 2016-08-11:
Image available on sr forum.

spocky, 2016-08-11:
:+1:  Thank you, this fixed my issue !
Time to close this issue, sorry for opening it in the wrong project :)

closed 2016-08-11 by spocky