Pulse-Eight/libcec #240: USB HDMI-CEC adapter does not work in Windows 10 Anniversary Update

gritstub, 2016-08-04:
The USB HDMI-CEC adapter does not work after my machine has updated to Windows 10 Anniversary Update (Version 1607 (OS Build 14393.10)), with EventGhost 0.4.1.r1710, and libCEC version 3.0.0.

Output from EventGhost:
Plugin: Pulse-Eight CEC adapter
found a CEC adapter: COM3
CEC ERROR: [#] the adapter did not respond correctly to a ping (try ...)

This setup was working prior to the automatic update to Windows 10 Anniversary Update.

Slion, 2016-08-04:
Mine is working just fine on W10AU. However I'm using a custom libcec build and I'm not using EventGhost. Could it be an EventGhost issue?

gritstub, 2016-08-05:
I realized that the issue was caused by another device taking over the same COM port as the USB HDMI-CEC adapter!

To remedy the issue:

1. Open the Device Manager
2. Ports (COM & LPT)
3. Pulse-Eight USB to HDMI CEC Adapter (rev.2) (COM3) Properties
4. Port Settings
5. Advanced...
6. Change COM Port Number

Device is working again!  Thanks for your response @Slion .

closed 2016-08-05 by gritstub