Pulse-Eight/libcec #241: AVR power on when changing TV input source

mdodig, 2016-08-06:
I have clean install of LibreElec Krypton on rPI3 connected to TV (Panasonic) trough AVR (Marantz).
Problem is, when I want to change source from TV to HDMI (to LibreElec), AVR is automatically powered on. This behaviour doesn't occur when using OpenElec with libcec older than v3.

Now I tried the following:
1. enabled logging only for CEC, TV is ON, AVR is ON
2. deleted log, and reboot machine
3. at time 11:55 I powered OFF AVR
4. at time 11:56 I switched TV input source to TV (AVR is OFF)
5. at time 11:57 I switched TV input source to HDMI2 (Kodi/rPI3), then AVR powers ON automatically

I hope this log will help someone to help me find what is turning ON AVR. I know it is CEC but why, and what has changed in newer LibCEC?

My system :
Tv : Panasonic TX-P42ST60E
AVR : Marantz NR-1506
Kodi : rPI3

[kodi.log.txt](https://github.com/Pulse-Eight/libcec/files/404864/kodi.log.txt)

Slion, 2016-08-09:
I believe I was staring at the code responsible for that the other day. Then I thought:  "why not but that's funny". Then again I don't have an AVR on my HDMI bus, only TV and Computer.

mdodig, 2016-08-09:
Do you have an idea how to fix it or workaround it?

Slion, 2016-08-10:
So you have an AVR on your HDMI network but you don't want to use it with Kodi?

Maybe your problems comes from the implementation of `CCECBusDevice::ActivateSource` in the file `CECBusDevice.cpp`. I would try commenting out the call to `SystemAudioModeRequest()` and see if that helps.

You would need someone to make such a custom libcec build for you.

mdodig, 2016-08-10:
Partly correct :)
I have AVR on HDMI network, and sometimes I want it to play audio from Kodi.
For that I have setup AVR to passthrough signal from Kodi to TV when AVR is powered off (this way I have audio playing from TV speakers).
When I want audio on "big speakers" I simply power on AVR.
This is working excellent except when I for some reason switch TV source "HDMI->TV->HDMI" then AVR is automatically powered on.
Sometimes I just can't power it off, because Kodi repeatedly power it on (but this is probably another issue).

Anyway, thank you for trying to help.

Slion, 2016-08-10:
That's a rather unusual use case, just try the fix suggested above I guess.
For your other issue you could also try the fix suggested on #233.

mdodig, 2016-08-28:
Hi Slion,

thanks for your sugestion. I found that file and have looked it's history.
If I read it correctly in line 1031 and bellow is check to see AVR's appearance and power it if found.
So now is normal behaviour to power on AVR, although all power on/off options are turned off?
Is there any chance that we can get option to select this as unwanted feature and somehow disable it?

Thanks for your answers!

Slion, 2016-08-30:
> So now is normal behaviour to power on AVR, although all power on/off options are turned off?

It sure looks like that.

Slion, 2016-08-30:
> Is there any chance that we can get option to select this as unwanted feature and somehow disable it?

You'll need somehow that can implement, test and contribute that for you with the blessing of @opdenkamp .

mdodig, 2016-09-01:
But we have ability to power on AVR with Kodi boot-up, I think there is no need to power it with changing source on TV.
We can think this way: if one is watching TV with audio from TV speakers, and switch input to Kodi which result in powering AVR, going back to TV continues routing sound trough AVR, not TV speakers.
If that person wanted TV audio from AVR it would powered it in the start of watching TV.
I'm saying that this kind of use isn't logical, and it should be changed.
What kind of AVR power status check was in libcec v2? In that version, you could start AVR automatically with Kodi, also TV input switching didn't power it on (which was a good thing).

If someone is willing to compile it, I will contribute as a tester, I have spare microsd card and use Kodi on daily basis so most errors I will find in a day or two...

mdodig, 2016-09-01:
I found similar problem with powering on, and I think when the old one is fixed mine will be too. #156 
https://github.com/Pulse-Eight/libcec/issues/156

If someone needs tester, ping me :)

starcroft, 2016-10-06:
This "feature" is causing me grief too. Before upgrading everything worked fine; with the HT turned off kodi played 2.0 through the TV speakers, turn the HT on and it plays pass through DTS or whatever. Could also control kodi with the TV remote. Perfect.

Now as long as it detects the HT it turns it on when switching to kodi, and sets the TV speaker mode to Hometheatre, when changing the TV source to TV this remains and because the TV does not support ARC I get no sound from the TV until I turn off the HT or change the TV speaker mode back to TV. Rinse and repeat each time the source is changed back to kodi. 

It would be slightly better if it turned off the HT when switching away from kodi, but really this should all be optional.

Also happy to test :)

opdenkamp, 2016-10-25:
Closing this as dupe of #156 which will be addressed in the next release.

closed 2016-10-25 by opdenkamp

mdodig, 2016-10-25:
Thx Lars. Hope it will fix issues :)

mdodig, 2016-10-27:
Wanted to inform that issue has been resolved :)
Latest libcec can be found in Milhouse build.

Thank you all!

opdenkamp, 2016-10-27:
Thanks for reporting back.

I've added a setting in Kodi who do still want the previous behaviour (someone needed it, don't remember who exactly ;-))

mdodig, 2016-10-27:
I didn't see that option, is it in Input menu?

mdodig, 2016-10-27:
PS feel free to ping me if you need test something

opdenkamp, 2016-10-27:
It's the last entry of system -> input devices -> peripherals -> cec adapter, but it's possible that the last version of the PR didn't get included in that last build. Try tomorrow's build if you don't see it with the current one

mdodig, 2016-10-27:
Today's build is missing that option, will check tomorrow. Don't need it, but I'll report here.