Pulse-Eight/libcec #242: Unmapped Vendorcommand for LG

Vollstrecker, 2016-08-06:
Hi,

I replaced my old TV with a LG 55LF6309-ZA. The old one worked with CEC and mythtv like a charme, but the new one makes some problems.

Starting mythtv works. CEC connection is established, source is changed etc.
Switching back to liveTV and back to HDMI-Port works also, mythtv is controllable.
Switching TV off and later back on, gives my no control and the unmapped code.

CEC command: >> 01:89:0b
[cec.txt](https://github.com/Pulse-Eight/libcec/files/405039/cec.txt)

Vollstrecker, 2016-08-10:
I played around a bit, reconnecting from the TV works with all functions (that I use) works, so I guess it's just an "Are you still there" request.

Unfortunately noone knows what answer is expected, as there are SL Initialzed messages after that, that isn't enough.

I connected this to HandleVendorCommandPowerOn, and it seems to work.

opdenkamp, 2016-10-25:
We're ignoring this vendor command on our matrix switches, and I haven't heard reports about this not working.

If this doesn't work for you and `HandleVendorCommandPowerOn()` works, please let me know and I'll re-open this ticket.

closed 2016-10-25 by opdenkamp

Vollstrecker, 2016-10-25:
It isn't enough to ignore it.
See Pull Request #256