Pulse-Eight/libcec #243: Windows Away-Mode not supported - breaks CEC functionality

tomillr, 2016-08-07:
Latest libcec still doesn't support Windows Away-Mode, which has been a part of Windows since XP MCE. This breaks CEC functionality and makes CEC adapter basically useless on Windows machines:

When Windows tries to wake up system in Away-Mode (to perform a schedules Windows Update or tv recording for example), libcec doesn't recognize away-mode call and overwrites it.

Instead of system remaining "silent", libcec sends out CEC command and wakes up connected tv and avr.

---

Problem applies to official libcec tray application as well as to 3rd party application like Kodi that rely on libcec library. 

Unofficial CEC Tray application by CIR21 properly supports Windows Away-Mode, so implementation is possible.

---

OpenPHT needs to register to receive away-mode status:

`RegisterPowerSettingNotification(m_hWnd, &GUID_SYSTEM_AWAYMODE, DEVICE_NOTIFY_WINDOW_HANDLE);`

The powermanager then needs to check for the Away-Mode call like this:
```
case PBT_POWERSETTINGCHANGE:
{
POWERBROADCAST_SETTING* s = (POWERBROADCAST_SETTING*)nEventData;
if( s != NULL && IsEqualGUID( s->PowerSetting, GUID_SYSTEM_AWAYMODE ) )
// The computer is entering "away" mode.
```

Further technical documentation is available in the attached documents or over on MSDN:
[Microsoft documentation](https://msdn.microsoft.com/de-de/library/windows/desktop/aa373208%28v=vs.85%29.aspx?f=255&MSPPError=-2147217396)

[awaymode.pdf](https://github.com/Pulse-Eight/libcec/files/405454/awaymode.pdf)
[AvailabilityRequests.pdf](https://github.com/Pulse-Eight/libcec/files/405455/AvailabilityRequests.pdf)

Slion, 2016-08-10:
Interesting stuff. Do you have links to that unofficial CEC Tray application?

tomillr, 2016-08-10:
The program used to be available for download over at the developer's site:

[Chirsoft](https://sites.google.com/site/chir21/home/CEC-Tray)

You can probably still get it from the dev (chir21-at-gmail-com).

opdenkamp, 2016-08-11:
The tray app has support for away mode, but it sounds like there's a bug that prevents it from doing (or not doing) what it should be doing.

This is not a problem with libCEC, it does not break CEC functionality in any way and it doesn't make the adapter useless as you claim, so I've changed the title. Away mode is simply a non-standard modification of S3. We'll look into implementing it properly in our tray application.

Kodi does not support away mode in it's powermanager code, so unless some Kodi dev hooks that up, Kodi will always request libCEC to wake up the TV when the PC wakes up.

opdenkamp, 2016-10-26:
Moved to https://github.com/Pulse-Eight/cec-dotnet/issues/2

closed 2016-10-26 by opdenkamp