Pulse-Eight/libcec #244: CEC adapter not working when using a DisplayPort to HDMI adapter

uqs, 2016-08-11:
Hey, I was using and loving a CEC adapter with Kodi on my FreeBSD box till a fateful lightning strike killed the HDMI port of that computer.

I'm now forced to use a DisplayPort to HDMI adapter, connect that to the CEC adapter to my TV ... and CEC hasn't worked ever since. I checked with a RPi and Simplink (what LG calls CEC) still works fine there, it also works when I use the RPi as the HDMI port, then I can receive the USB signals from the TV remote using cec-client on my FreeBSD box just fine.

So everything works, the adapter, the TV, the software ... but not on the DP port. Is this a physical limitation? Is the DP-HDMI adapter too cheap?

What I fail to understand is that CEC should terminate in the adapter, so why does it even matter what's connected on the PC side of the adapter?

Also, as soon as I plug it in via the DP adapter, all other connected devices to the TV (PS3, Chromecast) will disable Simplink/CEC as well. I understand that CEC is a bus and 1 broken device can ruin it for all.

This is the output when connected to the DP port of that machine:

# cec-client -l
libCEC version: 3.1.0, git revision: 3.1.0, compiled on Thu May  5 20:04:49 UTC 2016 by root@freebsd:10:x86:64-default-job-01 on FreeBSD 10.1-RELEASE-p32 (amd64), features: P8_USB, randr
WARNING: [              16]     TRANSMIT_FAILED_TIMEOUT_LINE
WARNING: [              57]     the adapter did not respond with a correct firmware version (try 1, size = 1)
WARNING: [             117]     TRANSMIT_FAILED_TIMEOUT_LINE
WARNING: [            1606]     the adapter did not respond with a correct firmware version (try 2, size = 0)
WARNING: [            3159]     the adapter did not respond with a correct firmware version (try 3, size = 0)
Found devices: 1

device:              1
com port:            /dev/ttyU1
vendor id:           2548
product id:          1002
firmware version:    1

This is the output when connected to the HDMI port of a RPi (but USB is still on the FreeBSD box):

# cec-client -l
libCEC version: 3.1.0, git revision: 3.1.0, compiled on Thu May  5 20:04:49 UTC 2016 by root@freebsd:10:x86:64-default-job-01 on FreeBSD 10.1-RELEASE-p32 (amd64), features: P8_USB, randr
WARNING: [              13]     TRANSMIT_FAILED_TIMEOUT_LINE
WARNING: [              54]     the adapter did not respond with a correct firmware version (try 1, size = 1)
WARNING: [              94]     TRANSMIT_FAILED_TIMEOUT_LINE
WARNING: [             173]     COMMAND_REJECTED: CEC transmission
WARNING: [             193]     COMMAND_REJECTED: TRANSMIT_EOM
Found devices: 1

device:              1
com port:            /dev/ttyU1
vendor id:           2548
product id:          1002
firmware version:    4
firmware build date: Thu Dec  6 11:15:20 2012 +0000
type:                Pulse-Eight USB-CEC Adapter

opdenkamp, 2016-08-12:
your DP->HDMI adapter is pulling the CEC line low, which is breaking CEC for all connected devices.

You need to buy one that doesn't do this to fix the problem, or replace your video card, or use a CEC-less HDMI cable that we sell in our webshop, connected between the CEC adapter and DP->HDMI adapter.

closed 2016-08-12 by opdenkamp

uqs, 2016-08-20:
The CEC-less cable has now arrived and I can confirm that CEC is back to fully working, even with the DP->HDMI adapter (I tried several different ones I had lying around). So thanks!

I wonder though, why this CEC-less cable is not shipped with the Pulse-Eight adapter by default ...

malard, 2016-08-20:
they cost more money to make as we have to find a company that will modify their machines to produce cables that "dont work" so to speak. They are also more expensive than normal cables and we equally get complaints from people saying oh cec doesnt work etc (basically we can't win either way!)

uqs, 2016-08-23:
Can't the adapter itself "terminate" the (possibly) bogus CEC signal?