Pulse-Eight/libcec #245: Can't open connection with RPI

AragurDEV, 2016-08-22:
I have a Phillips Device. EasyLink is enabled and HDMI-Cable can handle CEC (tested with Chromecast)
Here my Console Output:
```
starting a monitor-only client. use 'mon 0' to switch to normal mode
No device type given. Using 'recording device'
CEC Parser created - libCEC version 3.1.0
no serial port given. trying autodetect:
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [             107]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             112]     InitHostCEC - vchiq_initialise succeeded
DEBUG:   [             114]     InitHostCEC - vchi_initialise succeeded
DEBUG:   [             116]     InitHostCEC - vchi_connect succeeded
ERROR:   [             118]     RegisterLogicalAddress - vc_cec_set_logical_address(E) returned invalid arg (8)
ERROR:   [             121]     Open - vc_cec could not be initialised
ERROR:   [             122]     could not open a connection (try 1)
```

Slion, 2016-08-31:
I'm not even sure it finds the adaptor.

opdenkamp, 2016-10-18:
this is a duplicate of https://github.com/Pulse-Eight/libcec/issues/191

please check the responses in that ticket

closed 2016-10-18 by opdenkamp