Pulse-Eight/libcec #247: Automatically change active source with Python script

PiRune, 2016-09-05:
Hi All :)

Being a noob, I have modified the pyCecClient.py-script to control my TV, PS4, Apple TV and receiver with help from www.cec-o-matic.com, but I can't figure out how to _listen_ for cec-commands with Libcec and python. 

**Ideally I would like to make a script that runs when the TV is turned on and active source is unknown.** (a problem I often encounter)

In pseudo code it would be something like this:
```
When TV turns on:  #event function?
    sleep 10 seconds
    check currently active source
    If currently active source is unknown and Apple TV on:
        set Apple TV as active source
    else:
        set TV as active source
```
**Does anybody know how to do this with python and Libcec?**

'I know that you can use [echo "scan" | cec-client -s | grep "currently active"] to get identify the source if you are not using python, but it seems like a vert slow and wastefull way of doing it. I would like the script to run 24/7, so a full scan every 20 seconds is too much.

PiRune, 2016-09-08:
it appears that 

```
if message == ">> 0F:85":
    self.ProcessCommandTx('1F:82:41:00')
```
Works :)

opdenkamp, 2016-10-25:
Thanks for answering your own question :)

closed 2016-10-25 by opdenkamp