Pulse-Eight/libcec #248: Installing HDMI CEC with Windows Embedded Standard

PAV1THRA, 2016-09-21:
Hi,

I am unable to communicate with the TV using HDMI CEC connected with the Windows Embedded Standard OS.

Could you please tell whether HDMI CEC supports the Windows Embedded Standard OS.

opdenkamp, 2016-10-25:
Yes, but the driver to communicate with the adapter is not installed on Windows Embedded by default, as it tries to keep it's footprint as small as possible.

If you add this driver when installing Windows, then it'll install the required files:
https://dl.dropboxusercontent.com/u/59212404/driver.iso or
https://dl.dropboxusercontent.com/u/59212404/driver.zip

closed 2016-10-25 by opdenkamp

bcoady, 2019-07-15:
The link is dead. Can you re-post where to get the driver for Windows Embedded.