Pulse-Eight/libcec #250: Cannot build on raspberrypi/osmc dist - failing on step 'make -j4'

jkiddo, 2016-10-08:
osmc@osmc:~/libcec/build$ uname -a
Linux osmc 4.4.16-6-osmc #1 SMP PREEMPT Sun Oct 2 13:04:26 UTC 2016 armv7l GNU/Linux
osmc@osmc:~/libcec/build$ make -j4
[  2%] [  5%] [  8%] [ 10%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterMessageQueue.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterCommunication.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o
Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/AdapterFactory.cpp.o
In file included from /home/osmc/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp:37:0:
/home/osmc/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:45:47: fatal error: interface/vmcs_host/vc_cecservice.h: No such file or directory
 #include <interface/vmcs_host/vc_cecservice.h>
                                               ^
compilation terminated.
/home/osmc/libcec/src/libcec/adapter/RPi/RPiCECAdapterDetection.cpp:40:47: fatal error: interface/vmcs_host/vc_cecservice.h: No such file or directory
 #include <interface/vmcs_host/vc_cecservice.h>
                                               ^
compilation terminated.
In file included from /home/osmc/libcec/src/libcec/adapter/AdapterFactory.cpp:48:0:
/home/osmc/libcec/src/libcec/adapter/RPi/RPiCECAdapterCommunication.h:45:47: fatal error: interface/vmcs_host/vc_cecservice.h: No such file or directory
 #include <interface/vmcs_host/vc_cecservice.h>
                                               ^
compilation terminated.
In file included from /home/osmc/libcec/src/libcec/adapter/RPi/RPiCECAdapterMessageQueue.cpp:37:0:
/home/osmc/libcec/src/libcec/adapter/RPi/RPiCECAdapterMessageQueue.h:41:47: fatal error: interface/vmcs_host/vc_cecservice.h: No such file or directory
 #include <interface/vmcs_host/vc_cecservice.h>
                                               ^
compilation terminated.
src/libcec/CMakeFiles/cec.dir/build.make:169: recipe for target 'src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterCommunication.cpp.o' failed
make[2]: *** [src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterCommunication.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
src/libcec/CMakeFiles/cec.dir/build.make:146: recipe for target 'src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o' failed
make[2]: *** [src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o] Error 1
src/libcec/CMakeFiles/cec.dir/build.make:215: recipe for target 'src/libcec/CMakeFiles/cec.dir/adapter/AdapterFactory.cpp.o' failed
make[2]: *** [src/libcec/CMakeFiles/cec.dir/adapter/AdapterFactory.cpp.o] Error 1
src/libcec/CMakeFiles/cec.dir/build.make:192: recipe for target 'src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterMessageQueue.cpp.o' failed
make[2]: *** [src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterMessageQueue.cpp.o] Error 1
CMakeFiles/Makefile2:227: recipe for target 'src/libcec/CMakeFiles/cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/cec.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

opdenkamp, 2016-10-18:
The header path is set incorrectly, or you didn't install them.

The instructions from the README.md file work when using the official Raspbian distribution. If they don't work for another distribution, then you need to ask for help on it's forums or ask the maintainer(s).

closed 2016-10-18 by opdenkamp

jkiddo, 2017-07-14:
Problem was solved by building https://github.com/raspberrypi/userland on target

garbear, 2022-08-12:
I solved this via `sudo apt install libraspberrypi-dev`.

In my bash script I only wanted this to happen on the RPi. Here's the check I used:

```bash
# Needed for libcec on the Raspberry Pi
if [ "$(cat /proc/cpuinfo | grep "^Model" | awk '{print $3}')" = "Raspberry" ]; then
  sudo apt install -y libraspberrypi-dev
fi
```

No warranties.