Pulse-Eight/libcec #251: Cannot get building on Windows 10

codegecko, 2016-10-09:
Am trying to get started with libCEC, and I can't get this project building at all on Windows 10.
I've tried:
VS2015 with C++ 14 build tools
VS2012 with C++ 12 build tools
Both ways I've tried in IDE and command line, neither work.

Have downloaded P8 platform and copied into the "include" folder which has resolved missing include errors.

IDE throws errors stating 'identifier "uint8_t" is undefined' through the project and in reference files, which makes me believe that I may not have the right tools and/or libs installed?

I'm not naturally fluent in C++ so some pointers (or ideally, if someone could just build LibCecSharp for me and provide a download) it would be of tremendous help.

Thanks in advance,

Benjamin

xdissent, 2016-10-09:
I just got it to build today, here is what I learned:

* The p8-platform is included as a git submodule for windows builds. Run `git submodule update --init` before building.
* You must indeed have VS 2008, 2010 *and* either 2013 or 2015.
* You don't need Windows DDK unless you want to build an installer (haven't tried).
* Edit `support/build.cmd` and change `VSVERSION` to `14` for VS 2015. Leave it as `12` for VS 2013.
* VS 2008 *must* be **PRO** edition. Express will not work, because p8-platform requires `<msclr/marshal.h>`, which doesn't ship with the express edition for some reason. Also, the stdint stuff seems to be incompatible.
* For x64 support:
  * You must enable "Languages -> C++ -> x64 compilers" in the install options for VS 2008 (disabled by default).
  * VS 2010 *must* be **PRO** edition. Otherwise the x64 MSBuild platform targets for 2008/v90 will not be installed by VS 2010.
  * I had trouble building x64 LibCecSharp with VS 2013 Express, although it's supposed to do x64 apparently. Works fine in 2015 community though.
* You *must* install all of these VS versions **in ascending order**. It makes a huge mess of things otherwise.
* Make sure you open each version of VS at least once after installing to initialize configurations. I rebooted between each install for good measure.
* It is unbelievably difficult to track down all these old VS versions for dl, but they're all free and unlocked now - even the pro trials.
* Windows 10 "Reset this PC" is my favorite feature of all time.

The deal is, VS 2008 ships with the v90 platform that libcec is targeting, but it uses an old build system. The VS 2010 installer migrates the existing 2008 platform to the newer build system. Without it, VS 2013/15 can't find the v90 platform. You can tell you've done things correctly when `C:\Program Files (x86)\Microsoft Visual Studio 9.0\VC\include\msclr\marshal.h` (VS 2008) and `C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\Platforms\Win32\PlatformToolsets\v90` (VS 2010) exist. For x64 support, make sure `C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\Platforms\x64\PlatformToolsets\v90` exists as well. If not, either your VS 2008 or VS 2010 isn't pro, or you didn't elect to install the x64 compilers in VS 2008.

codegecko, 2016-10-11:
Brilliant - thankyou! I will try this tomorrow.

Would the libcec team be open to publishing a Nuget package for LibCecSharp? I'm happy building & publishing it (and keeping it updated using a CI server) if that's OK.

xdissent, 2016-10-11:
I'm not sure how obvious this would be to normal windows devs, but it might also be worth mentioning a couple more things:

* You have to install the Microsoft Visual C++ 2008 Redistributable Runtime package in order to **run** your LibCecSharp app if you don't have VC 2008 on the target machine. (x64 version comes with x86 too).
* To use LibCecSharp in a .NET 4 app, you have to add `useLegacyV2RuntimeActivationPolicy` to `App.config`:

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <startup useLegacyV2RuntimeActivationPolicy="true">
    <supportedRuntime version="v4.0" sku=".NETFramework,Version=v4.5.2" />
  </startup>
</configuration>
```
* The "Any CPU" configuration seems to be impossible to use: you have to add separate x64 and x86 configurations, and link to the appropriate LibCecSharp builds. I had to edit my csproj and add platform conditionals to the LibCecSharp reference:
```xml
  <ItemGroup Condition="'$(Configuration)|$(Platform)' == 'Debug|x64'">
    <Reference Include="LibCecSharp">
      <HintPath>..\libcec\Debug\x64\LibCecSharp.dll</HintPath>
    </Reference>
  </ItemGroup>
  <ItemGroup Condition="'$(Configuration)|$(Platform)' == 'Release|x64'">
    <Reference Include="LibCecSharp">
      <HintPath>..\libcec\Release\x64\LibCecSharp.dll</HintPath>
    </Reference>
  </ItemGroup>
  <ItemGroup Condition="'$(Configuration)|$(Platform)' == 'Debug|x86'">
    <Reference Include="LibCecSharp">
      <HintPath>..\libcec\Debug\x86\LibCecSharp.dll</HintPath>
    </Reference>
  </ItemGroup>
  <ItemGroup Condition="'$(Configuration)|$(Platform)' == 'Release|x86'">
    <Reference Include="LibCecSharp">
      <HintPath>..\libcec\Release\x86\LibCecSharp.dll</HintPath>
    </Reference>
  </ItemGroup>
```
* `cec.dll` has to be in the same directory as `LibCecSharp.dll`. Adding a reference to LibCecSharp will automatically copy `LibCecSharp.dll` to your output dir, but I had to add a post-build event for `cec.dll`:
```
xcopy /y /d "$(SolutionDir)libcec\$(ConfigurationName)\$(PlatformName)\cec.dll" "$(TargetDir)"
```

A Nuget pkg would have saved me an entire weekend probably, so I'm all for that suggestion! But it's nice to know how it all fits together under the hood I guess. =)

At least maybe some of this information can be added to the README?

opdenkamp, 2016-10-18:
thanks for this detailed description, I'll add it to the README.md file.

I was just reading up on nuget earlier today, to see if I can get this hooked up easily, but no promises ;)

codegecko, 2016-10-23:
The AnyCPU is only available for "traditional" .NET-compiled apps, not C++ ones with .NET entry points.
I'll see if I can port LibCecSharp to C# 2.0 and then you should be able to compile this into a Nuget package that works across platforms.

codegecko, 2016-10-23:
I finally managed to compile this on a Win10 VM with VS08 Pro, VS10 Pro, and VS15 Pro. If you hit me up directly I can supply the VM image via a download.

The LibCecSharp project definitely needs converting to a .NET lang (rather than C++ native with .NET entry points) to make this realistic for a Nuget package, Nuget doesn't support conditional referencing based on platform IIRC.

I'll look to sketch in some personal time in the next 7 days to attempt a rewrite, but if someone could reach out to me who has some intimate knowledge of C++ and libcec that'd be great, in case I have any specific questions that I can't address here.

opdenkamp, 2016-10-23:
The idea of the libCEC library is to provide a cross platform CEC library, and .Net is simply one of the supported types, so "converting" LibCecSharp into a full .Net library is not what we want to do with this library.

I've spent some time on trying to get nuget hooked up, but it lacks the required functionality to get this compiled automatically indeed.

It's also not needed if you simply want to develop an application that uses libCEC or LibCecSharp. You can simply install the latest binary build and reference LibCecSharp.dll. No need to install all the required VS versions to get it compiled from source. That's only needed if you want to change libCEC itself and compile the .Net wrapper.

opdenkamp, 2016-10-25:
I've split up the library part and the .Net part, so you only need vs 2013+ now to compile the .Net example applications. They can be found here now: https://github.com/Pulse-Eight/cec-dotnet

I've updated all the README files and included your input. The new cec-dotnet repos will get a README asap

closed 2016-10-25 by opdenkamp