Pulse-Eight/libcec #252: Sony KD-55W895 not seeing RPi3 as CEC Device

DaveTBlake, 2016-10-17:
Unfortunately I'm still having problems with CEC from Sony TV (KD-55W895), as initially discussed in #126 last year. Same TV but I now have a new pi. I had given up on CEC, yet randomly after a power cycle a few nights ago I briefly had CEC working! TV out of standby next day, and CEC was gone again, but this gives me hope that CEC could work, and so worth pursuing.

I have tried power cycling again, all kinds of combinations, and no CEC. I have to wonder if there is some timing issue between devices?

Details: Librelec 7.90.005 on RPi3 (hence libcec 3.1.0)
Log file here http://pastebin.com/uzike67h
Getting error:
"CecLogMessage - RegisterLogicalAddress - vc_cec_set_logical_address(1) returned device is busy (3)"

@opdenkamp could you take a look at this please.

As before have tried:
Connecting to a different HDMI port makes no difference.
New CEC enabled cable makes no difference.
Power cycling makes no difference, neither does trying different HDMI ports.
RPi as only HDMI devive makes no difference, no devices shown in list.
The TV does detect the presence of other possible CEC devices e.g. when a PVR is plugged in.
Second Pi to have this issue see

DaveTBlake, 2016-10-18:
Have upgraded to LE 7.90.007
Immediately after upgrade CEC worked (TV was on during install), but next day after TV had been in standby and RPi left on it stopped. Tried power cycling devices again in various combinations, but it didn't help.
Log file here 
http://pastebin.com/S0viWiKy

Installed a Millhouse build (#1015) that is +15 posts on libcec 3.1.0, but could not get CEC to work no matter the power-up combination I tried, and installing didn't make it work this time either.

A timing issue of some kind?

opdenkamp, 2016-10-18:
This is a duplicate of https://github.com/Pulse-Eight/libcec/issues/191 which I cannot reproduce when using current master with the current version of Raspian.

Something must have changed in the Pi's firmware that causes this as nothing has changed in libCEC afaict that could cause this since the first version that supported the Pi

popcornmix, 2016-10-18:
Last change to firmware CEC code was July 2014. I can't imagine a firmware change causing this issue.

opdenkamp, 2016-10-18:
This is the only code change that I can imagine that could be causing this: https://github.com/Pulse-Eight/libcec/commit/9ddb93b91a9e5dc7d89cccd6c593c64cf57a6d92

It got introduced in May 2015, but didn't get included in a release before 3.1.0.

But I can't reproduce any issues like this, tested on multiple TVs and with multiple devices connected, using a fresh Raspbian installation that I downloaded and apt-get upgraded this morning

popcornmix, 2016-10-18:
I don't believe @DaveTBlake 's TV has ever worked with libCEC.
He [reported problems](http://forum.kodi.tv/showthread.php?tid=232332) with OpenELEC 5.95.2 (over a year old) which I suspect wouldn't have had that commit.

opdenkamp, 2016-10-18:
Could you try to revert commit https://github.com/Pulse-Eight/libcec/commit/9ddb93b91a9e5dc7d89cccd6c593c64cf57a6d92 and upload a new log without that commit in @DaveTBlake

DaveTBlake, 2016-10-18:
I can confirm that my TV has did not work with libCEC and OpenElec 5.95.2. Having tried all the hardware changes I could never prove it was not the RPi itself because I didn't have any other TV to test it with so I gve up.

With LibreElec on a new RPi3 the CEC randomly worked for an evening - the first time of CEC and Kodi, although other devices were OK with CEC on the TV.  Hence I have become hopeful that a solution can be found.

Sorry @opdenkamp not sure how to build own LE  version to test the reversion of that commit. I dev Kodi on Windows, but a bit lost on other platforms. If you can guide me more then I am happy to try and work through this.

 it is just over a year old so not an old set. OpenElec

opdenkamp, 2016-10-18:
Hm right, you can't replace libcec.so easily on openelec because of the read only filesystem.
I can build the lib for you, and you could stick it in /tmp and use LD_LIBRARY_PATH to point to it when starting Kodi, but I'm not sure how to stop the running Kodi on OpenElec and run your own.

Is a Kodi build available for Raspbian? Then it's easy to do if you have a spare SD card.

popcornmix, 2016-10-18:
Ping @MilhouseVH - could you revert this commit in tonight's build LibreELEC build?
(assuming there aren't conflicts from later commits)

MilhouseVH, 2016-10-18:
I'm not able to revert https://github.com/Pulse-Eight/libcec/commit/9ddb93b91a9e5dc7d89cccd6c593c64cf57a6d92 cleanly, but I've reverted the changes with https://gist.github.com/MilhouseVH/530f703f13b762f73422e0209345c4b2 instead.

Rather than include this in tonight's build (I've no idea if CEC still works) I've uploaded a special build #1018x for RPi2 [here](http://milhouse.libreelec.tv/builds/testing/RPi2/LibreELEC-RPi2.arm-8.0-Milhouse-20161018172302-%231018x-gdc3b078.tar) - @DaveTBlake can you give it a spin?

DaveTBlake, 2016-10-18:
Thanks @MilhouseVH will do.
But before that I did one more power cycle test (running 1015), sadly with debug off as wanted to watch video, and CEC came to life. I have been trying most of the afternoon, I'm sure I did this sequence before just this time CEC worked. Go figure?

The sequence - power every thing off at plug; Start TV on other HDMI source; power up Kodi; Switch TV to the HDMI source Kodi on.

DaveTBlake, 2016-10-18:
@opdenkamp what power cycle sequence(s) would you like me to get debug logs for?

opdenkamp, 2016-10-18:
from whatever doesn't work for you ;)
so did reverting this PR work for you?

DaveTBlake, 2016-10-18:
Works briefly.... Then again CEC had started to work before I updated with the revertion (see previous post). Not sure I can say that the revert helped or not. It is like out of 20 times power cycle I might get CEC once.

What can you see from this
Log part 1 http://pastebin.com/SBe2T3gZ
part 2 http://pastebin.com/gkz67seE
had to split for pastebin limits.

What I did:
Install 1018y (while TV on and CEC randonly working). CEC continues to work.
So from working CEC (with revert) I did my night routine - TV in standby leave RPi as is and Kodi to sleep, wait 15 mins, then bring TV out of standby. It says it has CEC to "recorder 1", that is Kodi, but screen remains blank. Touch keyboard to wake up Kodi, but remote keys don't do anything.
CEC broken.
Then thinking test over, and family wanting TV, I started my PVR, also CEC enabled on 2nd HDMI port. TV stayed on Kodi screen an put up CEC message. CEC then works on Kodi.

Sorry if this is not the cleanest of testing. Off for meal now so will be back tomrrow. Your assistence with this much appreciated.

opdenkamp, 2016-10-18:
So this log shows that the TV is working fine, until the point where it says `reconnecting to the CEC adapter after standby mode`. The TV then no longer responds to anything (transmission are not acked).

If you can reproduce this by making it work after a power cycle of the Pi, then it's probably this standby/resume that's killing things for you.

@popcornmix could calling https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp#L261 a second time after the resume be causing trouble? https://dl.dropboxusercontent.com/u/59212404/dave.log line 251 (filtered log)

popcornmix, 2016-10-18:
I wouldn't recommend calling InitHostCEC a second time.
It does reinitialise the vchiq communication channel that the cec callbacks are transmitted through, without stopping/starting the cec service.

In fact I'm not sure that function needs to be called at all.
You call bcm_host_init() which does:
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c#L95
so should be handling that for you.

The standby you refer to - is this just TV standby, or is this the Pi suspending?

The thing is Pi doesn't really support suspend. There is no suspend option on stable OE/LE builds (the Pi can't wake up again). Milhouse builds do include a "virtual suspend" feature which just stops driving the HDMI output, but Kodi continues running until a keypress is received.
But this won't have been in @DaveTBlake 's tests with non-Milhouse builds.

Personally I'd have thought calling bcm_host_init (and the stuff in CRPiCECAdapterCommunication::Open to get the cec callbacks) at start, and bcm_host_deinit at exit would be enough. Don't do anything for suspend/resume - it doesn't really exist.

opdenkamp, 2016-10-18:
The PC's standby, and it's Kodi that signals this: https://github.com/xbmc/xbmc/blob/master/xbmc/peripherals/devices/PeripheralCecAdapter.cpp#L194

It simply closes libCEC and opens it again.

@DaveTBlake you can test if this fixes it by #ifdef'ing that part out for the Pi

opdenkamp, 2016-10-18:
I'll change the code to use `bcm_host_init()` only btw

popcornmix, 2016-10-18:
I don't have a CEC TV here, but I have a CEC capable receiver.
I can confirm that cec-client still seems to work with InitHostCEC removed.

popcornmix, 2016-10-18:
From the logs I could see "POLL not sent" was the start of the differences.
I couldn't tell exactly what failed.
Do you know if cecservice_send_command returned an error?
Did it send okay but no callbacks occurred?
Did sending and receiving appear to work, but for some reason TV wasn't happy?

opdenkamp, 2016-10-18:
Then we need a log from this with `#define CEC_DEBUGGING 1` set :)

opdenkamp, 2016-10-18:
I'll do some more testing with this tomorrow. I should be able to reproduce this with cec-client's `r` command, which (pretty much) does the same thing as Kodi does here.

MilhouseVH, 2016-10-18:
libcec won't build with `-DCEC_DEBUGGING=1` - maybe the debugging code has rotted?

x86: http://sprunge.us/BOMa
RPi2: http://sprunge.us/TGLH

If you would like anything tested and need me to provide a build, I prefer patches rather than instructions otherwise you'll never know for sure what is being tested (I have an amazing ability to cockup the most basic instructions)... The patch doesn't have to be a github commit - any sort of pastebin-type link will work for me, as long as it applies cleanly...

DaveTBlake, 2016-10-19:
In an attempt to show if the revert of 9ddb93b helped at all I have gone back to LE 7.90.007

I can now get some consistent behaviour, well I have repeated it at least once, and I believe I get the same with 007 and with the revertion.

From power up (at plug), I need to power TV first then the Pi with the TV with a different HDMI port selected. When I switch to the Pi HDMI CEC then identifies OK and works. But if TV goes into standby and Kodi  sleeps (not the Pi just Kodi screen saving mode), then on bringing TV out of standby although it lists the Pi as a CEC device it does not work from the remote. However if I then start my Humax PVR, another CEC device, the TV suddenly sees the Pi CEC too.

Log of this here http://pastebin.com/rFGSbrqb
- getting CEC to work on RPi by starting up another CEC device.

This is a change from my experiences with OE 5.95.2 last year, and yes it does seem to be  standby/resume related. I hope that the above sequence is revealing in some way.

All your help much appreciated. Ready to test anything that @MilhouseVH can build for me :)

opdenkamp, 2016-10-19:
Right, so it's this (fake) standby/resume that's causing this indeed then, thanks :)

Could you capture a log in which it breaks after resuming and then start that Humax, so I can see what it's doing when it starts to work

DaveTBlake, 2016-10-19:
That behavior is in the above log from 9:14 onwards when I brought TV out of standby this morning.

opdenkamp, 2016-10-19:
Sorry, I meant the Pi, not the TV

DaveTBlake, 2016-10-19:
Sorry, bit confused.
In log you have CEC working (watching video last night), then TV into standby at 22:26 with Pi/Kodi left on over night. At 9:14 TV is brought out of standby and Kodi resumes, but remote not do anything. At 9:15:20 I turned on the Humax, then I navigated Kodi menu using CEC.

I think that is what you are asking about, or something else?

DaveTBlake, 2016-10-19:
Ah just had to commit that pastebin again to confirm not robot, maybe you didn't see it?

http://pastebin.com/rFGSbrqb shows what you ask for I think

DaveTBlake, 2016-10-19:
I never put Pi into standby, I don't think it has one. I leave it on with Kodi in screen saver mode as no convenient way to stop and start - Kodi can power down, but then I have to faff with plug for Pi to get it to restart. Groping for plug is too much effort!

MilhouseVH, 2016-10-19:
> Right, so it's this (fake) standby/resume that's causing this indeed then, thanks :)

In case there's any confusion 7.90.007 has no support for the fake RPi standby/resume feature that disables power to the HDMI socket.

opdenkamp, 2016-10-19:
right, then that is another issue :)

Kodi isn't doing the standby/resume thing that broke things in yesterday's log. The TV wasn't responding at all in there. It wasn't acking transmissions sent to it according to the log, but that may have been a problem with the `bcm_host_init()` stuff.

What's happening here is that the TV powers up, triggers HPD and broadcasts it's physical address. libCEC isn't picking that up as the TV being powered on, and only notices it when the TV routes to HDMI 1 (`0f:80:00:00:10:00`).

In this situation, it's also going to work again if you select the device in the TV's sources list

DaveTBlake, 2016-10-19:
> In this situation, it's also going to work again if you select the device in the TV's sources list

No it doesn't, I tried that. Only starting Humax kicks it into working.

opdenkamp, 2016-10-19:
I'll make libCEC send a power status request to the TV when the TV broadcasts it's address, that should fix the power on thing for you.

| No it doesn't, I tried that. Only starting Humax kicks it into working.

Did you try that with this last build, without standby/resume from Kodi?

DaveTBlake, 2016-10-19:
I think I did, but I will go do it again.

opdenkamp, 2016-10-19:
I'm pretty sure that it'll work ;) If not, route to the TV's internal input or another HDMI input and back, and check if that works. Then pb another log please.

DaveTBlake, 2016-10-19:
http://pastebin.com/KR2QUTeY

At 10:47 TV brought out of standby onto HDMI with Pi, "recorder 1 " came up on screen as the selected device and Kodi screensaver stopped (screen brightens), but remote keys do nothing.
Try TV sync menu to select CEC device, but none are listed.
Put TV back in standby and back, nothing detected on HDMI.
Turn on Humax, and CEC starts to work.
Turn off Humax
TV into standby.
 --- Figure I may have misunderstood what you want me to test!
So start again from "morning" state e.g.  CEC working last time, TV in standby, Kodi in screensave
At 11:10 TV brought out of standby (same as 10:47)
Change TV source to different HDMI
Change TV source to own tuner
Change TV source to pi, still id as "recorder 1", but remote not work (Kodi in screen save)
Try TV select CEC device as "recorder 1", it is in list, but get message "communication failed"

DaveTBlake, 2016-10-19:
@opdenkamp let me know if I managed to produce a useful log, or you need something else. 
The family will be running LE 7.90.007 this evening, I can try to test around them.

opdenkamp, 2016-10-19:
I couldn't repro it with cec-client/libCEC only. checking kodi's implementation atm

DaveTBlake, 2016-10-19:
Thanks. At least I have a repeatable situation now, or I hope it is.

DaveTBlake, 2016-10-20:
Tried to turn debug on for CEC without being obtrusive on screen adding
```
<loglevel>1</loglevel>
  <debug>
      <extralogging>true</extralogging>
      <setextraloglevel>32768</setextraloglevel>
      <showloginfo>false</showloginfo> 
  </debug>
```
to advancedsettings.xml
But not getting the same message output in the log. Is there a way to do this or do I need to keep using the on screen settings?

opdenkamp, 2016-10-20:
I've never looked into the possibility of getting debug output in a release build without the popup, sorry :)

MilhouseVH, 2016-10-20:
`libCEC` component logging has a value of 16384.

This should work in advancedsettings.xml:
```
  <loglevel>1</loglevel>
  <debug>
      <extralogging>true</extralogging>
      <setextraloglevel>16384</setextraloglevel>
      <showloginfo>false</showloginfo> 
  </debug>
```

DaveTBlake, 2016-10-20:
After retesting with LE7.90.007 I find that if I don't turn on my Humax as well then I can't get CEC to work. I had thought I had managed without that 2nd device at some point, but I can't repeat it now. 

Sequence (with no Humax):
Every thing off at the plug.
TV power on, wake out of standby.
Switch to HDMI of Humax even though Humax is in standby
OR
Leave TV on HDMI port of Pi
Power up Pi at plug
Switch TV to HDMI port of Pi (if not on it already)
Kodi on screen, Pulse Eight message pops up.
TV has no CEC devices Sony in "Sync" menu
Remote does not effect Kodi.

Log of all this here http://pastebin.com/B6GaXN8w

Is there anything else I can try?

opdenkamp, 2016-10-20:
Not at this time, need to get some stuff committed first, thanks :)

DaveTBlake, 2016-10-21:
Sorry for being over enthusiastic :)

Wierd one last night: got CEC working by having Humax on, then turned Humax to standby. CEC to Pi keeps working while watch a few video files, but at the end of the 2nd it nolonger worked. So turn Humax back on, CEC works. Turn Humax to standby and keep using CEC, but again after 1hr or less it stops. Some kind of timeout??? I have a log for this, but quite big so not sure you wil be able to spot anything, or even if you will want to try?

opdenkamp, 2016-10-21:
Sure, upload it please and tell me around which time I need to look, can be useful. Thanks

DaveTBlake, 2016-10-21:
Log large, video with sync issues,  so in dropbox https://www.dropbox.com/s/pzvblmpfh4oylwd/kodi21_04_36.log?dl=0

Having used CEC to select what to play (episodes of "Bones") at 20:26 and 21:09, discovered that CEC had stopped working at the end of playback at 21:55. So 21:56 we woke the Humax and CEC works again. End of playback e19 at 22:45 CEC has stopped again. Put TV in standby and go to bed.
Notice at 2:57 have some CEC activity, not sure why.
Same at 4:36 "Tuner 1" is what the TV calls the Humax. I guess the Humax does something to update EPG etc. and TV sees activity?

opdenkamp, 2016-10-26:
I've addressed the `bcm_host_init()` calls after standby, but something else is wrong here.

Everything is working fine and keys are received until 21:29:45. Then the video player does a `GENERAL_RESYNC` and starts spamming calls to adjust the HDMI clock according to this log
I remember @popcornmix telling me that CEC shares the same clock, and I can see that the TV is treating the Pi as a new source device after the video stopped, indicating that CEC wasn't working while the video was playing. Once the clock adjustments stopped, libCEC sends an active source message, and CEC starts working again.

@popcornmix correct me if I'm wrong, but this appears something that's caused by these clock adjustments and CEC simply can't work correctly because of how the hw works. Stopping these adjustments (presumably sync to display setting in Kodi) would fix this.

popcornmix, 2016-10-26:
There are two hdmi clocks on Pi. HDMI state machine clock and pixel clock.
The audio and CEC runs off state machine clock and the video off the pixel clock.
The clock adjustment only affects the pixel clock, so I couldn't imagine it affecting CEC.
Also max adjustment is < 0.1% which is within the HDMI clock tolerance.

With out of the box settings, the pixel clock isn't adjusted.
With omxplayer enabled it will only be adjusted when sync playback to display is enabled.
With omxplayer disabled (default) it will only be adjusted when "PLL adjustment to maintain audio/video" is enabled (disabled by default).

@DaveTBlake Can you confirm you have the issue with "PLL adjustment" set to "off" and omxplayer disabled?

popcornmix, 2016-10-26:
Okay, I've double checked and the omxplayer case is as described, but the mmal case can adjust PLL when sync playback to display is disabled which wasn't what I'd thought. I'll change that for the next test build.

@DaveTBlake For now can you enable omxplayer and disable "adjust display refresh rate" and test CEC and see if that makes any difference?
With tomorrows build, you can test with omxplayer disabled and "sync playback to display" disabled which shouldn't have any "CRBP::AdjustHDMIClock" messages in log.

opdenkamp, 2016-10-26:
Thanks for looking into this :)

opdenkamp, 2016-10-26:
And for the explanation :)

DaveTBlake, 2016-10-28:
All my logs so far have been with "PLL adjustment" set to "off" and omxplayer disabled.
Initial CEC tests also had "adjust display refresh rate" off too, but mostly it is enabled. Maybe that has contributed to varying CEC behaviour, I'm unsure so will test.

opdenkamp, 2016-10-28:
Please upload another log if this still happens with Milhouse's last build

DaveTBlake, 2016-10-28:
OK, I'll see if I can get CEC working without the Humax with latest build.

DaveTBlake, 2016-10-28:
Tried Millhouse build 1027, just as before CEC only works when I have my Humax on too. Once CEC is working I can turn the Humax off, but after time or TV in standby then wake up CEC does not work. 

Would you like more logs? And do you want me to try the various combinations of video settings?

opdenkamp, 2016-10-28:
No idea what version of the PR and lib @MilhouseVH included in that build :) libCEC has been tagged for 4.0.0 now, so that should be the one to use, with the current version of the PR.

Please post a new log in which it breaks, and leave the Humax disconnected please.

DaveTBlake, 2016-10-28:
The build has LibCEC 4.0.0. Current logs have me turn on Humax at some point to check that behaviour, I will reboot without. But no Humax, I see Pulse Eight messages on screen, but TV does not see Pi as a CC device.

opdenkamp, 2016-10-28:
It may say 4.0.0, but that bump happened a couple of days ago, before it was tagged and the PR for Kodi got updated too. So you may be testing an incomplete version.

[30 more replies]

closed 2016-10-29 by opdenkamp