Pulse-Eight/libcec #253: cec-client won't transmit custom bits

DoucheBaguette, 2016-10-21:
I am running Raspbian Jessie on a Raspberry Pi 2. I installed cec-utils with apt.

If I run: `echo "on 0" | cec-client -s`, the TV turns on. I can similarly turn the TV to standby.

However, I can't get any "custom" commands to transmit. All of the documentation I can find online says to use the syntax of: `echo "tx 12 34 56 78" | cec-client -s`, but if I do that, cec-client only shows the first byte as "TRAFFIC", `<< 12`.

If I change the command syntax to `echo "tx 12:34:56:78" | cec-client -s`, I see the whole command listed as "TRAFFIC", just like the other commands (`<< 10:82:40:00`), but nothing happens, and there is no return messages or traffic.

It seems like it's not actually transmitting the command. By watching the other traffic, I can see that my device is address 1, and the TV is 0. But even if I do a simple `echo "tx 10:36" | cec-client -s` which should put the TV into standby, nothing happens. But `echo "standby 0" | cec-client -s` works fine.

opdenkamp, 2016-10-21:
can you give me the output of `cec-client -i` please. I've just tested it on the pi with the current code from master, and it works as expected.

DoucheBaguette, 2016-10-21:
`libCEC version: 3.0.1,  compiled on Fri Jul  3 19:20:43 UTC 2015 by pbuilder@mgorven-pc.thefacebook.com on Linux 3.16.0-41-generic (armv7l), features: P8_USB, P8_detect, 'RPi'`

opdenkamp, 2016-10-21:
just tried with current master, and it's working correctly when I try something like
`echo "tx 10:36" | cec-client -s`

closed 2016-10-21 by opdenkamp

DoucheBaguette, 2016-10-21:
I just compiled a fresh binary from source, and it works, as you said. I guess the version in apt is buggy. Thanks

DoucheBaguette, 2016-10-21:
@opdenkamp , sorry to bother, but could you test a few other commands for me? This sounds stupid, but, the command you used as a test works for me, but I honestly can't get a single other command working. And even though "10:36" WORKS, it doesn't seem to provide any feedback in the console. This is on the latest source build.

I know not all devices will support all commands, but "10:91" should report menu language, "10:82:20:00" should switch the active input to "2", etc - but none of them seem to work - and I have used the input change CEC message in the past in a prior project (on an older version of raspbian) with the same TV.

The full output when I run 10:91 is:

> echo "tx 10:91" | cec-client -s
> opening a connection to the CEC adapter...
> DEBUG:   [              68]	Broadcast (F): osd name set to 'Broadcast'
> DEBUG:   [              68]	InitHostCEC - vchiq_initialise succeeded
> DEBUG:   [              68]	InitHostCEC - vchi_initialise succeeded
> DEBUG:   [              69]	InitHostCEC - vchi_connect succeeded
> DEBUG:   [              69]	Open - vc_cec initialised
> DEBUG:   [              70]	logical address changed to Free use (e)
> NOTICE:  [              70]	connection opened
> DEBUG:   [              71]	processor thread started
> DEBUG:   [              71]	<< Broadcast (F) -> TV (0): POLL
> DEBUG:   [              71]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
> TRAFFIC: [              71]	<< e0
> DEBUG:   [             132]	>> POLL sent
> DEBUG:   [             132]	TV (0): device status changed into 'present'
> DEBUG:   [             132]	<< requesting vendor ID of 'TV' (0)
> TRAFFIC: [             132]	<< e0:8c
> TRAFFIC: [             369]	>> 0f:87:6b:74:6d
> DEBUG:   [             370]	>> TV (0) -> Broadcast (F): device vendor id (87)
> DEBUG:   [             370]	TV (0): vendor = Vizio (6b746d)
> DEBUG:   [             370]	expected response received (87: device vendor id)
> NOTICE:  [             370]	registering new CEC client - v3.1.0
> DEBUG:   [             371]	detecting logical address for type 'recording device'
> DEBUG:   [             371]	trying logical address 'Recorder 1'
> DEBUG:   [             371]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
> TRAFFIC: [             371]	<< 11
> TRAFFIC: [             641]	<< 11
> DEBUG:   [             912]	>> POLL not sent
> DEBUG:   [             912]	using logical address 'Recorder 1'
> DEBUG:   [             912]	Recorder 1 (1): device status changed into 'handled by libCEC'
> DEBUG:   [             912]	Recorder 1 (1): power status changed from 'unknown' to 'on'
> DEBUG:   [             912]	Recorder 1 (1): vendor = Pulse Eight (001582)
> DEBUG:   [             912]	Recorder 1 (1): CEC version 1.4
> DEBUG:   [             912]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
> DEBUG:   [             913]	logical address changed to Recorder 1 (1)
> DEBUG:   [             913]	Recorder 1 (1): osd name set to 'CECTester'
> DEBUG:   [             913]	Recorder 1 (1): menu language set to 'eng'
> DEBUG:   [             913]	GetPhysicalAddress - physical address = 2000
> DEBUG:   [             913]	AutodetectPhysicalAddress - autodetected physical address '2000'
> DEBUG:   [             913]	Recorder 1 (1): physical address changed from ffff to 2000
> DEBUG:   [             913]	<< Recorder 1 (1) -> broadcast (F): physical address 2000
> TRAFFIC: [             913]	<< 1f:84:20:00:01
> NOTICE:  [            1064]	CEC client registered: libCEC version = 3.1.0, client version = 3.1.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 2.0.0.0, git revision: libcec-3.1.0+28-c559955, compiled on Fri 21 Oct 19:58:51 UTC 2016 by pi@raspberrypi on Linux 4.4.21-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
> DEBUG:   [            1064]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
> TRAFFIC: [            1064]	<< 10:47:43:45:43:54:65:73:74:65:72
> DEBUG:   [            1364]	<< requesting power status of 'TV' (0)
> TRAFFIC: [            1364]	<< 10:8f
> TRAFFIC: [            1485]	>> 01:46
> DEBUG:   [            1485]	>> TV (0) -> Recorder 1 (1): give osd name (46)
> DEBUG:   [            1485]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
> TRAFFIC: [            1486]	<< 10:47:43:45:43:54:65:73:74:65:72
> **TRAFFIC: [            1850]	>> 01:90:00**
> DEBUG:   [            1850]	>> TV (0) -> Recorder 1 (1): report power status (90)
> DEBUG:   [            1850]	TV (0): power status changed from 'unknown' to 'on'
> DEBUG:   [            1851]	expected response received (90: report power status)
> TRAFFIC: [            1851]	<< 10:91
> DEBUG:   [            1942]	unregistering all CEC clients
> NOTICE:  [            1942]	unregistering client: libCEC version = 3.1.0, client version = 3.1.0, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 2.0.0.0, git revision: libcec-3.1.0+28-c559955, compiled on Fri 21 Oct 19:58:51 UTC 2016 by pi@raspberrypi on Linux 4.4.21-v7+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
> DEBUG:   [            1943]	Recorder 1 (1): power status changed from 'on' to 'unknown'
> DEBUG:   [            1943]	Recorder 1 (1): vendor = Unknown (000000)
> DEBUG:   [            1943]	Recorder 1 (1): CEC version unknown
> DEBUG:   [            1943]	Recorder 1 (1): osd name set to 'Recorder 1'
> DEBUG:   [            1943]	Recorder 1 (1): device status changed into 'unknown'
> DEBUG:   [            1943]	unregistering all CEC clients
> DEBUG:   [            2071]	UnregisterLogicalAddress - releasing previous logical address
> DEBUG:   [            2071]	logical address changed to Broadcast (f)

opdenkamp, 2016-10-24:
You won't get (much) feedback on the console, because of the `-s` parameter that you provided. It simply sends the command and exits.