Pulse-Eight/libcec #254: Raspberry pi build wont make

scoobyd00, 2016-10-21:
I can get this far then it crashes....

[ 13%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o
/home/pi/libcec/src/libcec/adapter/RPi/RPiCECAdapterDetection.cpp:40:47: fatal error: interface/vmcs_host/vc_cecservice.h: No such file or directory
 #include <interface/vmcs_host/vc_cecservice.h>
                                               ^
compilation terminated.
src/libcec/CMakeFiles/cec.dir/build.make:146: recipe for target 'src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o' failed
make[2]: *** [src/libcec/CMakeFiles/cec.dir/adapter/RPi/RPiCECAdapterDetection.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:227: recipe for target 'src/libcec/CMakeFiles/cec.dir/all' failed
make[1]: *** [src/libcec/CMakeFiles/cec.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

Any ideas why?

opdenkamp, 2016-10-21:
because you didn't follow the instructions from the readme.md file to the letter..

closed 2016-10-21 by opdenkamp

scoobyd00, 2016-10-21:
I followed these to the letter...

sudo apt-get update
sudo apt-get install cmake liblockdev1-dev libudev-dev libxrandr-dev python-dev swig
cd
git clone https://github.com/Pulse-Eight/platform.git
mkdir platform/build
cd platform/build
cmake ..
make
sudo make install
cd
git clone https://github.com/Pulse-Eight/libcec.git
mkdir libcec/build
cd libcec/build
cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
make -j4
sudo make install
sudo ldconfig

Could you help me whats wrong?

opdenkamp, 2016-10-21:
The instructions apply to Raspbian, which install headers and libs in /opt/vc
If you distribution does something else, please ask it's maintainers

scoobyd00, 2016-10-22:
Thankyou for the information, installed everything into /opt/vc on my build and it successfully compiled and is working great.