Pulse-Eight/libcec #255: Cannot turn off Samsung via CEC, turn on working

adrianmihalko, 2016-10-23:
I am using cec-client library on Raspbain to communicate with my Samsung TV.

I can get the TV to turn on just fine using

`echo "on 0" | cec-client -s -d 0`

But I cannot get it to go into standby using

`echo "standby 0" | cec-client -s -d 0`

Here is the output:

```
pi@raspberrypi:~ $ echo "standby 0" | cec-client -s -d 0
== skipped log-level parameter: invalid level '0' ==
opening a connection to the CEC adapter...
DEBUG:   [             131]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             136]	InitHostCEC - vchiq_initialise succeeded
DEBUG:   [             141]	InitHostCEC - vchi_initialise succeeded
DEBUG:   [             142]	InitHostCEC - vchi_connect succeeded
DEBUG:   [             149]	logical address changed to Free use (e)
DEBUG:   [             153]	Open - vc_cec initialised
NOTICE:  [             155]	connection opened
DEBUG:   [             158]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [             161]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             163]	<< e0
DEBUG:   [             166]	processor thread started
DEBUG:   [             197]	>> POLL sent
DEBUG:   [             200]	TV (0): device status changed into 'present'
DEBUG:   [             202]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [             204]	<< e0:8c
TRAFFIC: [             402]	>> 0f:87:00:00:f0
DEBUG:   [             402]	TV (0): vendor = Samsung (0000f0)
DEBUG:   [             405]	>> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             407]	expected response received (87: device vendor id)
DEBUG:   [             407]	replacing the command handler for device 'TV' (0)
NOTICE:  [             408]	registering new CEC client - v3.0.1
DEBUG:   [             408]	detecting logical address for type 'recording device'
DEBUG:   [             408]	trying logical address 'Recorder 1'
DEBUG:   [             408]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             409]	<< 11
TRAFFIC: [             682]	<< 11
DEBUG:   [             955]	>> POLL not sent
DEBUG:   [             955]	using logical address 'Recorder 1'
DEBUG:   [             955]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             955]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             955]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             955]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             956]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             956]	logical address changed to Recorder 1 (1)
DEBUG:   [             959]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             959]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             961]	GetPhysicalAddress - physical address = 4000
DEBUG:   [             961]	AutodetectPhysicalAddress - autodetected physical address '4000'
DEBUG:   [             961]	Recorder 1 (1): physical address changed from ffff to 4000
DEBUG:   [             961]	<< Recorder 1 (1) -> broadcast (F): physical adddress 4000
TRAFFIC: [             962]	<< 1f:84:40:00:01
NOTICE:  [            1112]	CEC client registered: libCEC version = 3.0.1, client version = 3.0.1, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 4.0.0.0,  compiled on Fri Jul  3 19:20:43 UTC 2015 by pbuilder@mgorven-pc.thefacebook.com on Linux 3.16.0-41-generic (armv7l), features: P8_USB, P8_detect, 'RPi'
DEBUG:   [            1113]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [            1113]	<< 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            1414]	<< requesting power status of 'TV' (0)
TRAFFIC: [            1414]	<< 10:8f
TRAFFIC: [            1558]	>> 01:90:00
DEBUG:   [            1558]	TV (0): power status changed from 'unknown' to 'on'
DEBUG:   [            1559]	expected response received (90: report power status)
NOTICE:  [            1560]	<< putting 'TV' (0) in standby mode
TRAFFIC: [            1560]	<< 10:36
DEBUG:   [            1564]	>> TV (0) -> Recorder 1 (1): report power status (90)
DEBUG:   [            1651]	unregistering all CEC clients
NOTICE:  [            1651]	unregistering client: libCEC version = 3.0.1, client version = 3.0.1, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 4.0.0.0,  compiled on Fri Jul  3 19:20:43 UTC 2015 by pbuilder@mgorven-pc.thefacebook.com on Linux 3.16.0-41-generic (armv7l), features: P8_USB, P8_detect, 'RPi'
DEBUG:   [            1651]	Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [            1651]	Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [            1651]	Recorder 1 (1): CEC version unknown
DEBUG:   [            1651]	Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [            1651]	Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [            1652]	unregistering all CEC clients
DEBUG:   [            2172]	UnregisterLogicalAddress - releasing previous logical address
```

opdenkamp, 2016-10-24:
The command is sent to the TV correctly and Samsung supports the command. It's probably disabled in the TV's settings, usually called "auto standby" or something similar.

If it doesn't work, please send an email to support@pulse-eight.com as this is not a problem with the library

closed 2016-10-24 by opdenkamp

Vanuan, 2017-11-07:
Was having the same problem. Found a corresponding setting:

System -> Anynet+ (HDMI-CEC) -> Auto Turn Off
[image]
[image]
[image]

segaura, 2019-08-31:
> But I cannot get it to go into standby using
> 
> `echo "standby 0" | cec-client -s -d 0`
> 
> Here is the output:

I've gone through the same issue and the solution is to send the `standby` command from the active source, e.g. this way

`echo "as" | cec-client -s -d 0`
(some timely wait)
`echo "standby 0" | cec-client -s -d 0`

don't know if this is a Samsung interpretation of CEC.
My TV is a UE65NU7000, libCEC 4.0.4 from OSMC distro and one single option in TV settings related to CEC, the one to turn it on or off.