Pulse-Eight/libcec #257: Main changelog should not be in the debian directory

marillat, 2016-10-27:
Hi,

Please don't use debian/changelog.in as main changelog.
debian/changelog is here to document only debian changes.

opdenkamp, 2016-10-27:
This is what shows up as main changelog everywhere. If Debian/Ubuntu has some other way to display this as main changelog on things like Launchpad (not strictly Debian ofc), then I'm happy to change it.

marillat, 2016-10-27:
dh in debian/rules call dh_installchangelogs who install the main ChangeLog file in /usr/share/doc/<package>/changelog and also install debian/changelog in /usr/share/doc/<package>/changelog.Debian so user can see changes in the main source code and changes in the debian package.

opdenkamp, 2016-10-27:
So I need to override `dh_installchangelogs` in `debian/rules`? That's not going to change what's being displayed as changelog on pages like https://launchpad.net/~pulse-eight/+archive/ubuntu/testing/+packages , is it? Users don't really care about what changes specific to Debian are in the binary, it's one binary to them with 1 changelog :)

If Debian wants to add it's own changes to the package that's not in our upstream /debian dir, why not just add it to the changelog? That'll clearly show who changed what and for which reason?

opdenkamp, 2017-01-02:
Closing this ticket. If there's any other way to get the lib's changelog displayed and that satisfies Debian policy, then I'm happy to change it.

closed 2017-01-02 by opdenkamp