Pulse-Eight/libcec #258: LG TV - Stand-by to ON not enough to wake Kodi up

camelreef, 2016-10-27:
Hello,

I am using the latest LibreELEC nightly build provided by Milhouse on a Raspberry Pi 1.
[Kodi 17 - LibreELEC Testbuilds for RaspberryPi (Kodi 17.0)](http://forum.kodi.tv/showthread.php?tid=269814)

I have an old-ish LG TV (32LS3590). I am experiencing intermittent CEC drops and other weird/changed behaviours.

The main one is that when I turn the TV ON from Stand-by, I do not recover the Kodi interface without having to send 2 commands from the remote first, which appears to restart the whole CEC thing from scratch on the Kodi side.

Here are logs with CEC debug turned on.

[kodi.log.txt](https://github.com/Pulse-Eight/libcec/files/555711/kodi.log.txt)

1. The Pi is rebooted with the TV in the ON state.
2. I turn the TV in Stand-by at **18:18:25 140.499390**
3. I turn the TV ON from Stand-by at **18:18:57 172.485062**
4. I press a first button (OK on remote) at **18:19:16 190.785095**
5. I press a second button (still OK on remote) and finally get the interface back, with the last command accepted at **18:19:33 208.202209**

Is there anything you would need in addition to this?

I am available to provide more logs and manipulations on demand.

I can test code changes preferably through the Milhouse builds, I'd like to avoid patching and building myself if I can help it, but I can still do it if provided with clear instructions, I am capable enough.

Thanks!

Nico

opdenkamp, 2016-10-27:
The old LG TVs do all kinds of weird things that they aren't supposed to do according to the CEC specification, and libCEC works around these issues by emulating an LG playback device. That's what it's doing here too, and it looks like LG's simplink stuff simply isn't fully initialised yet when you press the button for the first time, and the TV then sends it's vendor ID (another LG weirdness). Then it starts working.

I've tried to speed it up before, by actively sending the deck status commands (`10:1b:20`) before the tv asks for it, but it bugged out some other old LGs.

Have you tried waiting longer? What happens when you press the button about a minute after it came out of standby?

camelreef, 2016-10-27:
Thanks for the quick reply!

So, here is what I did, following your message:

Go from Stand-by to ON only, and wait for a 

1. long time without doing anything or pressing any remote key: I get the no signal screensaver forever
2. Then, after at least a minute, I press a remote button a single time: I get a TV OSD block, but quickly return to the no signal screensaver forever
3. Then I press a remote button a second time: Kodi returns, the command is immediately taken into consideration, and I get all the Kodi OSD no

tifications about CEC turning on/registering or whatnots

Waiting does not appear to introduce a change in behaviour.

Would you like the log files for all this?

Nico

opdenkamp, 2016-10-27:
Yes please, but I can't promise that I can fix it without breaking some other model again, so a potential fix is going to have to wait for 4.0.1. It'll need testing.

camelreef, 2016-10-27:
Here they are:

[kodi.log.txt](https://github.com/Pulse-Eight/libcec/files/555916/kodi.log.txt)

Process and time stamps:

1. TV is ON, reboot of the RPi, wait for PVR stuff to sync, first Kodi log at **13:02:13  18.935728**
2. TV from ON to Stand-by at **14:04:46 171.967850**
3. TV from stand-by to ON, before waiting for more than a minute, at **14:05:50 235.891495**
4. First remote button press (OK button), before waiting again for more than a minute, at **14:07:13 319.122345**
5. Second remote button press (OK button) at **14:08:51 417.271332**

I'm fine with your caveats (darn vendors and their ways of interpretting/implementing documented standards!) and need of testing!

Nico

propergol, 2016-12-01:
This issue is not LG specific.
I have a lot of issues with latest OSMC krypton and LibCEC 4.0 on my Samsung +Yamaha AVR.
One of those issues is that, If I set the "when TV is turn off" to "stop playing" in order to stop live PVR then on the next wake up of the TV with TV remote, I have a black screen.
I need to do multiple things blindly with the remote and then the kody's interface suddenly come alive.

narspt, 2016-12-07:
I can reproduce similar behavior with an LG 22MT55D tv (2015 model), but if I can understand it correctly my tv apparently just doesn't send any cec commands at all when in standby... the 1st key press just turns the tv on, then we need to send a 2nd key press quickly (before the tv goes to standby again due to no signal from kodi device) and that will be sent to kodi, waking it up and activating video signal output... If I'm correct then maybe the only way would be if there is a way to detect that tv got out of standby and if yes then wake up kodi automatically?

JFYI not exactly same problem but maybe good you know as maybe somehow related... Another issue I have with this same tv is that I wanted to prevent kodi to always wake up my tv at kodi startup (to avoid tv always waking up after an ac power loss for eg.), to do it I need to:
- disable "switch source to this device on startup"
- and set "devices to power on during startup" to none
(yes, both required, if I leave one of them then it still wakes up)
but if disable that then I have no cec working at all, apparently the tv needs this to initialize cec? again... no way to detect if the tv is turned on or detect when it gets turned on (or out of standby) and only then send activate source? (to avoid waking it up unnecessarily)

kkkkamal, 2017-06-03:
I need to enable simplink on boot automatically. But, currently we have to do it manually every-time from Remote Control using the Settings->Input->Simplink->HDD Recorder.

I have been stuck for almost a week trying to figure out what the issue is. We are using a cec-client to control the LG TV. Is there a way to enable simplink on the TV from the Pi or any methods to enable simplink from the Pi side.

Hardware Used:

Raspberry PI LG Television ( 32LF565B ) Any input is highly appreciated.

opdenkamp, 2017-07-12:
@camelreef I was just looking at this ticket, but the log is either no longer available, or there's an issue with github preventing it from being opened after their changes to the site earlier this year (browser reports `The request signature we calculated does not match the signature you provided. Check your key and signing method.`). Could you re-upload a recent log somewhere please

camelreef, 2017-07-12:
Hello!
I do not have those logs anymore, so I cannot repost them. (can't generate them anymore, it now works for me).
The error you are getting is weird, I can click and get the above logs myself...

opdenkamp, 2017-07-12:
Ok, if it works, then it's all good and I can close this ticket in that case. Looks like it's just NoScript is blocking it as possible xss scripting attempt.

closed 2017-07-12 by opdenkamp