Pulse-Eight/libcec #259: Build failure with   std::string strError(m_strError);   return strError;

asavah, 2016-10-28:
Build fails with
```
[ 17%] Building CXX object src/libcec/CMakeFiles/cec.dir/adapter/AdapterFactory.cpp.o
/home/asavah/kross/src/libcec/src/libcec/adapter/AOCEC/AOCECAdapterCommunication.cpp: In member function ‘std::__cxx11::string CEC::CAOCECAdapterCommunication::GetError() const’:
/home/asavah/kross/src/libcec/src/libcec/adapter/AOCEC/AOCECAdapterCommunication.cpp:118:20: error: binding ‘const P8PLATFORM::CMutex’ to reference of type ‘P8PLATFORM::CMutex&’ discards qualifiers
   CLockObject lock(m_mutex);
                    ^~~~~~~
In file included from /home/asavah/kross/src/libcec/src/libcec/adapter/AOCEC/AOCECAdapterCommunication.h:41:0,
                 from /home/asavah/kross/src/libcec/src/libcec/adapter/AOCEC/AOCECAdapterCommunication.cpp:42:
/home/asavah/kross/build/odroidc2/rootfs/usr/include/p8-platform/threads/mutex.h:130:12: note:   initializing argument 1 of ‘P8PLATFORM::CLockObject::CLockObject(P8PLATFORM::CMutex&, bool)’
     inline CLockObject(CMutex &mutex, bool bClearOnExit = false) :
            ^~~~~~~~~~~
src/libcec/CMakeFiles/cec.dir/build.make:182: recipe for target 'src/libcec/CMakeFiles/cec.dir/adapter/AOCEC/AOCECAdapterCommunication.cpp.o' failed

```

It seems to be due to an unneeded copy-paste
https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/adapter/AOCEC/AOCECAdapterCommunication.cpp#L116
has ``` CLockObject lock(m_mutex);```

While the ::GetError() implementation for all the other adapters does not contain the lock.

Eg:
https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/adapter/TDA995x/TDA995xCECAdapterCommunication.cpp#L142

opdenkamp, 2016-10-28:
all the others are wrong :)
thanks, will fix it

opdenkamp, 2016-11-07:
I'm going to have to change this in the platform lib first

commit 0cf3f0fe (2017-03-15): changed: const IAdapterCommunication::GetLogicalAddresses(), making the mutex mutable for now without changing the platform lib. closes #259
closed 2017-03-15 by opdenkamp